home-assistant/operating-system #1821: USB Drive won't mount

iWriteThingsDown, 2022-03-31:
### Describe the issue you are experiencing

I am running HA OS on a Pi4b

When I plug in a USB drive, it doesn't show up as a usable drive anywhere. I am trying to access it via Samba, which is all working, and I can access and use the media directory through that, but the drive never shows up.

I have definitely had this working with a 5TB drive in the past, and have tried another 5TB drive, a 4TB drive and a 64GB thumb drive, all with the same result.

Drives are formatted to NTFS.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Plug in external disk drive to USB port

### Anything in the Supervisor logs that might be useful for us?

```txt
22-03-31 12:14:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/002/003 - None
22-03-31 12:14:21 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bsg/0:0:0:0 - None
22-03-31 12:14:21 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bsg/0:0:0:1 - None
22-03-31 12:14:27 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/sda - /dev/disk/by-id/usb-WD_Elements_SE_2622_57583732443831434A533853-0:0
22-03-31 12:14:27 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/sda1 - /dev/disk/by-id/usb-WD_Elements_SE_2622_57583732443831434A533853-0:0-part1
22-03-31 12:14:27 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/sda2 - /dev/disk/by-id/usb-WD_Elements_SE_2622_57583732443831434A533853-0:0-part2
```

### Anything in the Host logs that might be useful for us?

```txt
[77699.258804] usb 2-2: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[77699.279912] usb 2-2: New USB device found, idVendor=1058, idProduct=2622, bcdDevice=10.23
[77699.279930] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[77699.279942] usb 2-2: Product: Elements SE 2622
[77699.279954] usb 2-2: Manufacturer: Western Digital
[77699.279965] usb 2-2: SerialNumber: 57583732443831434A533853
[77699.286516] usb-storage 2-2:1.0: USB Mass Storage device detected
[77699.289300] scsi host0: usb-storage 2-2:1.0
[77700.319193] scsi 0:0:0:0: Direct-Access     WD       Elements SE 2622 1023 PQ: 0 ANSI: 6
[77700.324470] sd 0:0:0:0: [sda] Spinning up disk...
[77700.324893] scsi 0:0:0:1: Enclosure         WD       SES Device       1023 PQ: 0 ANSI: 6
[77701.342375] .....ready
[77705.439068] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[77705.439356] sd 0:0:0:0: [sda] 9767475200 512-byte logical blocks: (5.00 TB/4.55 TiB)
[77705.439373] sd 0:0:0:0: [sda] 4096-byte physical blocks
[77705.440128] sd 0:0:0:0: [sda] Write Protect is off
[77705.440146] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[77705.441752] sd 0:0:0:0: [sda] No Caching mode page found
[77705.441793] sd 0:0:0:0: [sda] Assuming drive cache: write through
[77705.565754]  sda: sda1 sda2
[77705.570072] sd 0:0:0:0: [sda] Attached SCSI disk
```

### System Health information

## System Health

version | core-2022.3.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Australia/Sydney

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.24.0
Stage | running
Available Repositories | 1073
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 29.0 GB
disk_used | 4.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.14.0), AirCast (3.3.0), Home Assistant Google Drive Backup (0.106.2), Let's Encrypt (4.12.0), Terminal & SSH (9.3.0), Samba share (9.5.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

### Additional information

I reformatted the Pi4 with regular Linux and the drive mounts without any issue. Only when using HA OS does this seem to occur.

agners, 2022-03-31:
> I have definitely had this working with a 5TB drive in the past, and have tried another 5TB drive, a 4TB drive and a 64GB thumb drive, all with the same result.

HAOS has no automatic mount logic for generic drives currently. You must have used some additional scripts to make this happen. There is the idea of a Mount Manager (see https://github.com/home-assistant/supervisor/issues/2564), but that is not implemented yet.

iWriteThingsDown, 2022-04-01:
If there's no automatic mount logic, is there a manual way to mount?

agners, 2022-04-01:
No officially supported way at this point.

But there is [this](https://gist.github.com/eklex/c5fac345de5be9d9bc420510617c86b5)...

closed 2022-04-01 by agners