home-assistant/operating-system #1828: Bluetooth devices unusable in hassos ova for x86_64

fredrik-eriksson, 2022-04-01:
### Describe the issue you are experiencing

I'm trying to give my homeassistant installation support for bluetooth. Partly for the bluetooth le tracking, and partly to try the [hassio-plejd](https://github.com/icanos/hassio-plejd) integration.

I've tried two different dongles:

- Asus BT-400 (Broadcom chipset, BCM20702A0)
- Asus BT-500 (Realtek chipset, 8761BU I think)

I've successfully configured libvirt/qemu to pass through the USB devices to the OVA, and the USB IDs are listed with lsusb in the guest, but it seems that bluez does not detect them properly. Searching around it seems that the most likely cause is missing firmware. On the other hand I don't see any errors about firmware in the dmesg log, so maybe there is something more/else missing?

For the log below I had both devices connected at boot, but I've also tried them one at a time with the same result.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Attach bluetooth dongle to hypervisor hardware
2. Configure USB passthrough for bluetooth dongle to HassOS VM
3. Ensure passthrough worked by listing usb devices with lsusb
4. Run "bluetoothctl list", no controller is detected

### Anything in the Supervisor logs that might be useful for us?

```txt
not really...
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.133385] Bluetooth: Core ver 2.22
[    0.133385] NET: Registered protocol family 31
[    0.133385] Bluetooth: HCI device and connection manager initialized
[    0.133385] Bluetooth: HCI socket layer initialized
[    0.133385] Bluetooth: L2CAP socket layer initialized
[    0.133385] Bluetooth: SCO socket layer initialized
...
[    2.723025] usb 2-2: new full-speed USB device number 3 using uhci_hcd
...
[    2.891045] usb 2-2: New USB device found, idVendor=0b05, idProduct=17cb, bcdDevice= 1.12
[    2.892660] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.892662] usb 2-2: Product: BCM20702A0
[    2.892664] usb 2-2: Manufacturer: Broadcom Corp
[    2.892665] usb 2-2: SerialNumber: <serial>
...
[    3.815122] usb 3-2: new full-speed USB device number 2 using uhci_hcd
[    3.971677] usb 3-2: New USB device found, idVendor=0b05, idProduct=190e, bcdDevice= 2.00
[    3.972712] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.973588] usb 3-2: Product: ASUS USB-BT500
[    3.974415] usb 3-2: Manufacturer: Realtek
[    3.975215] usb 3-2: SerialNumber: <serial>
...
[    5.324103] Bluetooth: hci0: command 0x0c03 tx timeout
...
[   13.324202] Bluetooth: hci0: BCM: Reset failed (-110)
```

### System Health information

## System Health

version | core-2022.3.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/Stockholm

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | failed to load: unreachable
can_reach_cloud_auth | failed to load: unreachable
can_reach_cloud | failed to load: unreachable

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 30.8 GB
disk_used | 6.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | TellStick (2.1.0), Check Home Assistant configuration (3.10.0), Terminal & SSH (9.3.0), Mosquitto broker (6.0.1), Plejd (0.8.0-beta)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 0
views | 2
mode | storage

</details>

### Additional information

Just in case legacy stuff can be related: originally I ran my hass installation on a raspberry pi 2, but eventually I migrated it (through backup and restore) to a OVA instance (version 5.13). Since then I've dutifully updated and is currently on latest release. The OVA is currently running in qemu under libvirtd on a Gentoo hypervisor.

weidi, 2022-04-16:
Seeing the similar thing with my Onboard Bluetooth device with HassOS

mine is a Realtek 802.11ac with embedded bl rom

agners, 2022-04-21:
> Asus BT-400 (Broadcom chipset, BCM20702A0)

This model has USB PID:VID [0b05:17bc and should be supported](https://linux-hardware.org/?id=usb:0b05-17cb) since Linux 3.2. However, they seem to require firmwares which are not available from the [official Linux firmware repository](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git). It also seems that Ubuntu doesn't distribute firmware files for this device. This could be because nobody cared to add those firmware files, or for license reasons, I am not sure.

This [GitHub gist](https://gist.github.com/ssledz/69b7f7b0438e653c08c155e244fdf7d8) explains how-to extract the firmware files from the Windows drivers. However, I don't think we can safely distribute those for license reasons.

> Asus BT-500 (Realtek chipset, 8761BU I think)

This model has USB PID:VID [0b05:190e and should be supported](https://linux-hardware.org/?id=usb:0b05-190e) with Linux 5.14 and newer. HAOS 7.x comes with Linux 5.10, HAOS 8.x with Linux 5.15. The required firmware `rtl_bt/rtl8761a_fw.bin` should be present as well. Can you try HAOS 8.0.rc2 (currently on beta channel) to see if that indeed works?

In general, I recommend not buy hardware from Vendors which don't ship firmware through the Linux firmware repositories. Broadcom is known to be notoriously bad at providing firmware files to the community. Intel is well known to provide firmware files and pro-actively add them for new devices.

fredrik-eriksson, 2022-04-21:
Actually... A few days ago I got a RPI4 to spare and migrated from the OVA (back) to a RPI. The onboard bluetooth was not powerful enough, so I plugged in the BT-500; and it worked. So I'm guessing the OVA is actually missing bluetooth support?

For the record this is the bluetooth messages in the host logs (as compared to the logs from the OVA):
```
[    2.227645] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[    2.363388] usb 1-1.3: New USB device found, idVendor=0b05, idProduct=190e, bcdDevice= 2.00
[    2.363414] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.363431] usb 1-1.3: Product: ASUS USB-BT500
[    2.363444] usb 1-1.3: Manufacturer: Realtek
[    2.363457] usb 1-1.3: SerialNumber: <serial>
...
[    4.060931] Bluetooth: Core ver 2.22
[    4.061015] NET: Registered protocol family 31
[    4.061029] Bluetooth: HCI device and connection manager initialized
[    4.061051] Bluetooth: HCI socket layer initialized
[    4.061068] Bluetooth: L2CAP socket layer initialized
[    4.061102] Bluetooth: SCO socket layer initialized
...
[    4.095404] usbcore: registered new interface driver btusb
...
[    6.957091] Bluetooth: HCI UART driver ver 2.3
[    6.957102] Bluetooth: HCI UART protocol H4 registered
[    6.957160] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    6.957315] Bluetooth: HCI UART protocol Broadcom registered
[    7.090067] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    7.090434] bcmgenet fd580000.ethernet eth0: Link is Down
[    7.128408] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.128420] Bluetooth: BNEP filters: protocol multicast
[    7.128434] Bluetooth: BNEP socket layer initialized
...
[   61.274206] Bluetooth: RFCOMM TTY layer initialized
[   61.274224] Bluetooth: RFCOMM socket layer initialized
[   61.274246] Bluetooth: RFCOMM ver 1.11
```

So... My problem is solved; but the issue with bluetooth not working in OVA is probably not :)

agners, 2022-04-22:
@fredrik-eriksson which version of the OS are you using on the Pi? Could be that Raspberry Pi comes with some extra drivers. Altough, usually those are Broadcom drivers (since the Pi uses Broadcom chips), and the BT-500 uses ASUS-BT500.

Anyways, I am quite sure HAOS 8.0 should work with Asus BT-500 as well on OVA, so I think we can close this for now.  In case it isn't working for someone interested, please open a new issue.

As for Asus BT-400, I don't intend to add support for it.

closed 2022-04-22 by agners