home-assistant/operating-system #1837: 8.rc1 fails to connect to wifi on rpi4

regevbr, 2022-04-08:
### Describe the issue you are experiencing

After upgrading to 8.rc1 on my rpi4, I couldn't connect to my HA instance via wifi. I managed to ssh using ethernet and discovered that it can't find any wifi devices to connect to:
```sh
# nmcli radio
WIFI-HW  WIFI     WWAN-HW  WWAN    
enabled  enabled  enabled  enabled 
# nmcli device wifi rescan
Error: No Wi-Fi device found.
# nmcli device wifi
# cat /etc/*-release
NAME="Home Assistant OS"
VERSION="8.0.rc1 (RaspberryPi 4 64bit)"
ID=haos
VERSION_ID=8.0.rc1
PRETTY_NAME="Home Assistant OS 8.0.rc1"
CPE_NAME=cpe:2.3:o:home-assistant:haos:8.0.rc1:*:development:*:*:*:rpi4-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS RaspberryPi 4 64bit"
VARIANT_ID=rpi4-64
SUPERVISOR_MACHINE=raspberrypi4-64
SUPERVISOR_ARCH=aarch64
```
After downgrading back to 7.6 it worked again

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.0.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to 8.0.rc1
2. Fail to connect to device using wifi

### Anything in the Supervisor logs that might be useful for us?

```txt
I can't access them while using the faulty os version
```

### Anything in the Host logs that might be useful for us?

```txt
I can't access them while using the faulty os version
```

### System Health information

## System Health

version | core-2022.4.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | Asia/Jerusalem

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4567
Installed Version | 1.24.3
Stage | running
Available Repositories | 1009
Downloaded Repositories | 36

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | beta
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 57.8 GB
disk_used | 33.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.14.0), Home Assistant Google Drive Backup (0.106.2), Plex Media Server (2.8.0), Transmission (dev), Studio Code Server (4.2.0), Node-RED (11.1.1), NGINX Home Assistant SSL proxy (3.1.1), MariaDB (2.4.0), AdGuard Home (4.5.0), Mosquitto broker (6.0.1), Zigbee2mqtt (1.25.0-1), Node-RED Dashboard (1.2.0), SharpTools.io (1.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 28
views | 8
mode | storage

</details>

Sheyla1996, 2022-04-08:
Same problem

Magurele, 2022-04-08:
Also affected.

dealerch, 2022-04-08:
same here , Wifi setting disappeared

agners, 2022-04-08:
Hm, this is likely connected with the latest kernel update of Raspberry Pi. I'll investigate.

pascalkrielen, 2022-04-09:
Same issue, Rpi4

Gorole, 2022-04-09:
Same problem :-/
Rpi4

samauro1, 2022-04-09:
Same issue, Rpi4 64

cipriancu, 2022-04-10:
Same issue, Rpi4

Narselon, 2022-04-11:
Same issue on the Rpi 3b.

wolfgangpurrer, 2022-04-11:
Same problem here, thanks for the post - with this i was able to identify the issue quickly,
 I connected via cable and  downgraded with ha os update --version 7.6

Gorole, 2022-04-13:
> Same problem here, thanks for the post - with this i was able to identify the issue quickly, I connected via cable and downgraded with ha os update --version 7.6 and now i'm back online :-)

Can you tell me please how i can downgrade? :-)
THX

regevbr, 2022-04-13:
ssh to the machine (using ethernet) and run `ha os update --version 7.6` if I remember correctly

agners, 2022-04-15:
WiFi is working again on 8.0.rc2.

closed 2022-04-15 by agners

bschatzow, 2022-04-15:
8.0.rc2 left my system unable to boot.  Created an issue on this.

agners, 2022-04-16:
> 8.0.rc2 left my system unable to boot

"Unable to boot" means for me not seeing anything at all (e.g. bootloader broken etc.). What you describe in #1852 boots fine, but somehow Supervisor has a problem starting up the system properly. I'll respond to that issue.

bschatzow, 2022-04-16:
Poor choice of words. It boots but HA doesn't start. Left in a condition that I could not restore from cli.

regevbr, 2022-04-17:
I can confirm it works on rc2, thanks @agners :-)