home-assistant/operating-system #1838: Odroid N2 installation image at boot gets IP from dhcp but then gives 'serverip' not set error

alijani1, 2022-04-08:
### Describe the issue you are experiencing

When installing for first time on an odroid n2+ ssd,  using V7.6 image for system tries to boot and gets an IP address you can see in the cnsule but then it errors out with serverip not set error and it does not continue with the install.  See attached image.

Got image from https://github.com/home-assistant/operating-system/releases/download/7.6/haos_odroid-n2-7.6.img.xz per instructions.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. download Odroid home assistance image file from https://github.com/home-assistant/operating-system/releases/download/7.6/haos_odroid-n2-7.6.img.xz
2. flash in SD card
3. insert in Odroid n2+
4. attach network cable on n2+ and power on
5. system boots, even gets IP address, but then errors with serverip not set error

See attached screenshot.

...
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
na
```

### Anything in the Host logs that might be useful for us?

```txt
na
```

### System Health information

na

alijani1, 2022-04-10:
BTW, this same process works with MMC.  It only fails when using SD Card

agners, 2022-04-11:
What you are seeing is U-Boot trying to boot through network since it wasn't able to find the OS on the SD card. So the root problem is that U-Boot can't seem to read from the SD-card.

n2aws, 2022-05-06:
If it helps,  I have an odroid-n2+,  I wrote the 7.6 image to an sd-card,  and was able to boot successfully from it.  (I do not have an emmc)

This is most likely a hardware issue with the SD Card.

Tijgerd, 2022-08-10:
I fixed/bypassed this issue by installing an older version on the SD card and updating it within Home Assistant itself.

closed 2022-11-15 by github-actions[bot]