home-assistant/operating-system #1839: Home Assistant 2022.3.8 with Home Assistant OS 7.6 Requirements for sighthound not found: ['pillow==7.2.0']

majkers, 2022-04-09:
### Describe the issue you are experiencing

After updating to HASS 7.6 I started to get:

```
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
WARNING: You are using pip version 22.0.3; however, version 22.0.4 is available.
You should consider upgrading via the '/usr/local/bin/python3 -m pip install --upgrade pip' command.
2022-04-09 14:15:44 ERROR (stream_worker) [homeassistant.components.stream.stream.camera.kamera_dome1] Error from stream worker: Error opening stream (ERRORTYPE_113, Host is unreachable) rtsp://192.168.1.7/ch0_0.h264
2022-04-09 14:15:45 ERROR (SyncWorker_7) [homeassistant.util.package] Unable to install package pillow==7.2.0: ERROR: Cannot install pillow==7.2.0 because these package versions have conflicting dependencies.
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
WARNING: You are using pip version 22.0.3; however, version 22.0.4 is available.
You should consider upgrading via the '/usr/local/bin/python3 -m pip install --upgrade pip' command.
```

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing special
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing special
```

### System Health information

Nothing special

### Additional information

Nothing special

agners, 2022-04-11:
Reopen since you claim its caused by the OS update.

What OS version have you been using before? What version of Core are you using? Are you using custom components? If so, can you try disabling them to see if the problem still persists?

majkers, 2022-04-12:
Yes I am using Sighthound integration but the official one but from the same author but https://github.com/robmarkcole/HASS-Sighthound I didsubmited an issue as well there and author has produced a new version. I will check if it helps and let know. BUT.  Sighthound  is also offical integration https://www.home-assistant.io/integrations/sighthound/ so that could be an issue there as well

agners, 2022-04-12:
It seems that has been fixed with https://github.com/robmarkcole/HASS-Sighthound/commit/ff7486671bc2363b3a0e818716014463258bd528 and release https://github.com/robmarkcole/HASS-Sighthound/releases/tag/0.12.2.

In any case, if it's related to a custom integration (even if its a derivative of a core integration), the issue belongs to the repository of that custom integration. And in any case, since the error is ins the Core logs, it rather belongs to the Core repository (https://github.com/home-assistant/core/issues). Definitely not OS related, that this showed up after OS update is pure coincidence. I recommend updating one thing at a time, it makes it easier to attribute issues to the right component.