home-assistant/operating-system #1840: Latest release breaks TPU (Coral M.2)

thundergreen, 2022-04-09:
### Describe the issue you are experiencing

I entered beta channel and updated to latest beta release 8.0 rc1 and it breaks M.2 coral support with error:

Open issue on Frigate github:

https://github.com/blakeblackshear/frigate/issues/3072

Error:

### Relevant log output

```
[2022-04-09 14:55:17] frigate.edgetpu                ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
Process detector:coral:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/frigate/frigate/edgetpu.py", line 136, in run_detector
    object_detector = LocalObjectDetector(
  File "/opt/frigate/frigate/edgetpu.py", line 44, in __init__
    edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1.0
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.0 rc1 (beta)

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update to beta release 8.0 rc1
2. start Frigate Addon
3. TPU coral recongnition fails with error message above

### Anything in the Supervisor logs that might be useful for us?

```txt
[2022-04-09 14:55:17] frigate.edgetpu                ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
Process detector:coral:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/frigate/frigate/edgetpu.py", line 136, in run_detector
    object_detector = LocalObjectDetector(
  File "/opt/frigate/frigate/edgetpu.py", line 44, in __init__
    edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1.0
```
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System Health information

## System Health

version | core-2022.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/Paris

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 3622
Installed Version | 1.24.3
Stage | running
Available Repositories | 1109
Downloaded Repositories | 115

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 228.5 GB
disk_used | 160.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), MariaDB (2.4.0), AppDaemon (0.8.2), Mosquitto broker (6.0.1), Frigate NVR Beta (0.10.0) (3.8), SSH & Web Terminal (10.1.1), MaryTTS (1.4.0), Samba Backup (5.0.0), chrony (2.3.0), ICantBelieveItsNotValetudo (3.6.1), Valetudo Mapper (1.10.0), ESPHome (2022.3.2), Rhasspy Assistant (2.5.11), TasmoAdmin (0.17.0), Studio Code Server (4.2.0), Log Viewer (0.13.0), Z-Wave JS (0.1.55), Zigbee2mqtt (1.25.0-1), Valetudo Mapper (1.10.0), Let's Encrypt (4.12.0), DahuaVTO2MQTT (1.0.2), RTSPtoWeb - WebRTC (1.2.1), AdGuard Home (4.5.0), Hass Calendar Addon (0.200), Teamspeak server (3.13.6), Unifi wifi ticket (1.0.0), hyperion (0.7.0), Asterisk (2.1.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 56
views | 92
mode | storage

</details>

agners, 2022-04-11:
Hm, it seems that the driver got removed from staging in the v5.12 release. `8.0.rc1` is using v5.15 now. It should be fairly straight forward to include the out-of-tree driver.

agners, 2022-04-11:
Related to #1698.

agners, 2022-04-15:
Google Coral out-of-tree drivers are now part of 8.0.rc2.

closed 2022-04-15 by agners