home-assistant/operating-system #1843: Static routing broken after 2022.4.2 update

andrewbeetee, 2022-04-12:
### Describe the issue you are experiencing

Hi

Apologies in advance - I'm not entirely sure what is handling the networking portions of HAOS. I would gather it is the OS, but docker may have it's own network namespaces that I'm not seeing separate routing table?

After the 2022.4.2 update - Static routes that I configured via nmcli, ceased to work.

I added them again, and it _half_ fixes the problem, and I'm completely stumped at this point.

Traffic initated from HAOS routes correctly, traffic initated from another system (say, my monitoring server, or the reverse proxy), results in HAOS default routing the traffic.  It appears like there is two routing tables?

I rebooted, and I found that, despite telling nmcli to make the routes permanent, they were not. 

HAOS is running inside a esxi VM.  

_ip_ gives me nothing when doing an _"ip r(oute)"_ 

Happy to provide further information, but my  go-to network tools are providing me with little - and what documentation I can find doesn't really describe what's going on under the hood with HAOS & docker 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

5.10.108

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Add Static routes
2. Upgrade Core to 2022.4.2
3. Watch your static routes not work anymore
4. Add them back. 
5. Watch them only observed if the system initates the connection.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System Health information

No issues

VikeDragon, 2022-04-12:
The same problem. The wlan0 connection has been renamed Supervisor wlan0 and it is impossible to specify routing, since the connection is re-created when rebooting. So I can't lower the metric for wlan0 from 600 to 203, and previously I did routing and everything worked.

andrewbeetee, 2022-04-12:
@VikeDragon - I've got "Supervisor enp2s1" and "HassOS default" - with the latter not attached to a Device

Regarding internet access - one of the things the routes have, are my DNS servers - which became unreachable. I do notice the box reaching out to the 1.1.1.1 DoH server (and failing)

agners, 2022-04-12:
@andrewbeetee what happens if you add the routes to the new `Supervisor enp2s1` profile?

This is most likely due to Supervisor 2022.4.0 update, specifically https://github.com/home-assistant/supervisor/pull/3528. I am transferring the issue to Supervisor.