home-assistant/operating-system #1845: move datadisk popup closes at selection

capstan1, 2022-04-14:
### Describe the issue you are experiencing

When trying to move the data partition of Home Assistant via UI ( “Move datadisk” at System > Host), the popup closes before I can accept the change. The drive is detected, but as soon as I choose the entry in the drop-down, the popup closes.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

core-2022.4.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Go to System. In the part "Host" choose the 3 dots menu. 
2. Open "move data disk"
3. Choose your disk from the drop-down. In my case the popup just disappears.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
Unfortunately not
```

### System Health information

## System Health

version | core-2022.4.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.24.0
Stage | running
Available Repositories | 1030
Downloaded Repositories | 38

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 5. Mai 2022, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 29.1 GB
disk_used | 12.8 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.102.0), ADB - Android Debug Bridge (0.8.0), ESPHome (1.15.3), File editor (5.3.3), Google Assistant Webserver (0.0.4), SSH & Web Terminal (10.1.1), Samba share (9.5.1), deCONZ (6.12.0), Node-RED (11.1.1), Glances (0.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 34
views | 15
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

I had the data partition laying on that SSD before (via datactl move command). Had to reinstall Home assistant when moving to a raspberry 4. Restored a complete backup from a session that used external data disk before.Don't now if that's from interest.

However, I removed all partitions from the ssd.

agners, 2022-04-16:
> I had the data partition laying on that SSD before (via datactl move command). Had to reinstall Home assistant when moving to a raspberry 4. Restored a complete backup from a session that used [...]

We use the file system label of that partition to mount the data partition. If you attach it to the system with that name, then the system might choose the external data disk over the internal data partition. However, when properly removed it shouldn't be a problem.

Are there any hints in `ha su logs`?

dan1el-k, 2022-04-16:
Experienced the same issue with closing of the popup when trying via UI.

When trying via the cli running```ha os datadisk move /dev/sda```, no matter if from within the container or from the homeassistant os host I got this error
Post "http://supervisor/os/datadisk/move": context deadline exceeded (Client.Timeout exceeded while awaiting headers).

Nothing in the supervisor logs.

MartyTremblay, 2022-04-25:
I'm having the same issue but on an ODROID-XU4 . However, using the CLI was successful.

capstan1, 2022-04-27:
CLI for me was succesful, too. Hence I couldn't try look into @agners suggestion.
However, I did the same steps over and the same issue occurs.
`ha su logs` does show nothing interesting other than detecting the usb drive when I plugged it in. 

Anything more to debug?

agners, 2022-05-06:
> However, I did the same steps over and the same issue occurs.

You mean you tried using `ha os datadisk move /dev/sda` again after it has already been moved?

Be aware: Once your disk has been used as a external data disk, it might get detected also by a new installation. To avoid any interference, you should format the data disk before trying to reproduce the issue.

capstan1, 2022-05-06:
> You mean you tried using `ha os datadisk move /dev/sda` again after it has already been moved?
>
> Be aware: Once your disk has been used as a external data disk, it might get detected also by a new [...]

No, I tried using the UI to move the data partition to a freshly formatted drive. Didn't try cli again, as it worked the first time.

To clarify: I shutdown my working instance, etched HA to another SD card, restored a backup which had a moved data partition, attached an empty USB drive (without partitions) to it and tried to move it via UI. Again, the drive is shown but when you try to select it in the drop-down the popup closes.

agners, 2022-05-23:
@capstan1 ok got it now :)

I was able to reproduce the issue here. This seems a frontend issue, so I reported it in the frontend repository at https://github.com/home-assistant/frontend/issues/12729.

closed 2022-05-23 by agners

Gby56, 2022-05-26:
had the exact same issue today, I'm trying to have an external hard drive with the SD install, and the lsblk was showing the disk fine. Through the cli it didn't seem to work because `ha os info` didn't show the new /dev/sda...
I managed to get the UI to cooperate by using TAB and just typing in the text field `/dev/sda` and it let me trigger the move, will see if that works...