home-assistant/operating-system #1852: version 8.0.rc2

bschatzow, 2022-04-15:
### Describe the issue you are experiencing

Installed RC.2 and system does not come back on line.
2-04-15 18:48:01 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/config:

Can't restore from CLI.
Error: 'OSManager.update' blocked from execution, system is not running - CoreState.STARTUP

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.0 RC2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Installed the update
2. System never came back on line
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-04-15 18:48:01 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/config:

Error: 'OSManager.update' blocked from execution, system is not running - CoreState.STARTUP
```

### Anything in the Host logs that might be useful for us?

```txt
# ha host logs
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8 (builder@72088471553b) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -gc62532023) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Fri Apr 15 20:11:05 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 256 pages in unavailable ranges
[    0.000000] percpu: Embedded 28 pages/cpu s77272 r8192 d29224 u114688
[    0.000000] pcpu-alloc: s77272 r8192 d29224 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=511fe888ff364f9e8755d41305ab32fe cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1360x768M@60 smsc95xx.macaddr=DC:A6:32:75:DC:C6 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootfstype=squashfs ro rauc.slot=B
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[... 616 lines cut ...]
[   70.745388] audit: type=1325 audit(1650062705.385:198): table=nat family=2 entries=0 op=xt_register pid=8366 subj==unconfined comm="iptables"
[   70.745422] audit: type=1300 audit(1650062705.385:198): arch=c00000b7 syscall=209 success=yes exit=0 a0=4 a1=0 a2=40 a3=7fddb87720 items=0 ppid=8342 pid=8366 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[   70.745440] audit: type=1327 audit(1650062705.385:198): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4C002D6E
[   70.767158] audit: type=1325 audit(1650062705.401:199): table=filter family=2 entries=0 op=xt_register pid=8370 subj==unconfined comm="iptables"
[   70.767207] audit: type=1300 audit(1650062705.401:199): arch=c00000b7 syscall=209 success=yes exit=0 a0=4 a1=0 a2=40 a3=7feabda580 items=0 ppid=8342 pid=8370 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[   70.767225] audit: type=1327 audit(1650062705.401:199): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D4C002D6E
[   70.789660] audit: type=1325 audit(1650062705.429:200): table=nat family=2 entries=5 op=xt_replace pid=8378 subj==unconfined comm="iptables"
[   70.789710] audit: type=1300 audit(1650062705.429:200): arch=c00000b7 syscall=208 success=yes exit=0 a0=4 a1=0 a2=40 a3=557856da60 items=0 ppid=8342 pid=8378 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[   70.789727] audit: type=1327 audit(1650062705.429:200): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4E00444F434B45525F4F5554505554
[   70.797255] audit: type=1325 audit(1650062705.437:201): table=nat family=2 entries=7 op=xt_replace pid=8386 subj==unconfined comm="iptables"
[   70.939133] eth0: renamed from veth75d4c81
[   70.979860] IPv6: ADDRCONF(NETDEV_CHANGE): veth59e0ef0: link becomes ready
[   70.980048] hassio: port 11(veth59e0ef0) entered blocking state
[   70.980066] hassio: port 11(veth59e0ef0) entered forwarding state
[   79.443159] kauditd_printk_skb: 20 callbacks suppressed
[   79.443182] audit: type=1334 audit(1650062714.080:208): prog-id=0 op=UNLOAD
[   79.443215] audit: type=1334 audit(1650062714.080:209): prog-id=0 op=UNLOAD
[   79.517605] audit: type=1334 audit(1650062714.156:210): prog-id=0 op=UNLOAD
[   79.517645] audit: type=1334 audit(1650062714.156:211): prog-id=0 op=UNLOAD
#
```

### System Health information

Can't get there

agners, 2022-04-16:
My `rpi4-64` test instance came up fine, I wonder what is exactly blocking on your end. Can you get the Supervisor logs (`ha su logs`)?

bschatzow, 2022-04-16:
The only error is posted above. 22-04-15 18:48:01 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/config:

I have restored so I can't get more information. If you want me to try again I can.

bschatzow, 2022-04-16:
I just tried again and it work via CLI.  Not sure why it failed before, but it did.

closed 2022-04-16 by bschatzow