home-assistant/operating-system #1853: Intel AX210NGW Wifi/Bluetooth adapter not working

marciogranzotto, 2022-04-16:
### Describe the issue you are experiencing

I've added an Intel AX210 NGW Wifi + BLE adapter to my Ryzen Mini-PC running Home Assistant OS but it fails to initialize it.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install an Intel AX210NGW PCI adapter
2. Turn on the system
3. Check the logs
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-04-16 14:02:31 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-3 / failes to initialize by udev
22-04-16 14:02:31 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-3 / failes to initialize by udev
22-04-16 14:02:32 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/072 - None
22-04-16 14:02:32 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/072 - None
22-04-16 14:02:32 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-3 / failes to initialize by udev
22-04-16 14:02:32 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-3 / failes to initialize by udev
22-04-16 14:02:32 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/073 - None
22-04-16 14:02:32 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/073 - None
22-04-16 14:02:33 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/074 - None
22-04-16 14:02:33 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/074 - None
22-04-16 14:02:33 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-3 / failes to initialize by udev
22-04-16 14:02:33 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/075 - None
22-04-16 14:02:33 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/075 - None
22-04-16 14:02:34 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-3 / failes to initialize by udev
22-04-16 14:02:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/076 - None
22-04-16 14:02:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/076 - None
```

### Anything in the Host logs that might be useful for us?

```txt
[ 1115.302782] usb 3-3: USB disconnect, device number 32
[ 1115.695657] usb 3-3: new full-speed USB device number 33 using xhci_hcd
[ 1115.837467] usb 3-3: New USB device found, idVendor=8087, idProduct=0032, bcdDevice= 0.00
[ 1115.837471] usb 3-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1115.845479] Bluetooth: hci0: Reading Intel version information failed (-22)
[ 1115.845489] Bluetooth: hci0: Intel Read version failed (-22)
[ 1115.845527] Bluetooth: hci0: Intel reset sent to retry FW download
[ 1115.846781] usb 3-3: USB disconnect, device number 33
[ 1116.240598] usb 3-3: new full-speed USB device number 34 using xhci_hcd
[ 1116.382473] usb 3-3: New USB device found, idVendor=8087, idProduct=0032, bcdDevice= 0.00
[ 1116.382477] usb 3-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1116.390533] Bluetooth: hci0: Reading Intel version information failed (-22)
[ 1116.390549] Bluetooth: hci0: Intel Read version failed (-22)
[ 1116.390701] Bluetooth: hci0: Intel reset sent to retry FW download
```

### System Health information

## System Health

version | core-2022.4.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | America/Sao_Paulo

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4060
Installed Version | 1.24.5
Stage | running
Available Repositories | 1068
Downloaded Repositories | 36

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | April 16, 2022, 9:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 232.2 GB
disk_used | 121.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.10.0), Glances (0.15.0), Samba NAS (9.5.1-nas4), Valetudo Mapper (1.10.1), Home Assistant Google Drive Backup (0.106.2), Dasshio (0.3.8), PS4WakerBridge (0.0.2), Mosquitto broker (6.0.1), SSH & Web Terminal (10.1.1), AdGuard Home (4.5.0), Tautulli (2.3.0), MariaDB (2.4.0), Node-RED (11.1.1), Studio Code Server (4.2.0), WireGuard (0.6.0), Assistant Relay (0.7.4), TasmoBackup (1.04.26), File editor (5.3.3), motionEye (0.18.0), Custom deps deployment (1.2.0), JupyterLab (0.7.0), Duck DNS (1.14.0), Nightscout (dev), Nginx Proxy Manager (0.11.0), Lovelace Kindle Screensaver (1.0.0), jackett (0.32.0), radarr (0.10.0), sonarr (0.8.0), Transmission NAS (1.1.0), Plex Media Server with NAS (dev), Filebrowser (2.21.1-2), room-assistant (2.20.0), Telegram Bot Amazon (dev), SQLite Web (3.3.0), HassOS SSH port 22222 Configurator (0.9), Zigbee2mqtt (1.25.0-1), ESPHome (2022.3.2), Tailscale (0.6.1), PrusaSlicer (dev), Portainer (2.11.1-3), Grocy (0.17.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 23
views | 34
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

Here's a picture of the module:
[image]

marciogranzotto, 2022-04-16:
I've updated to HA OS 8.0 RC2 and it's now working!
Seems that this driver is only available on the newer kernel.

agners, 2022-04-19:
Hm, that this model works is a bit surprising since I actually forgot to [backport a patch](https://lore.kernel.org/all/20220312205055.GO283544@scaer/T/) I sent to the mailing list to add firmware for this models :sweat_smile: 

Do you have another operating system loaded on that machine and did a reboot? It could be that the firmware already has been loaded. Did you test WiFi?

In any case, I will add more firmwares for Intel WiFi 22000 series models, just to be sure they'll work in the future.

marciogranzotto, 2022-04-22:
> Do you have another operating system loaded on that machine and did a reboot? It could be that the firmware already has been loaded. Did you test WiFi?

It's just bare metal HA OS.
I only tested Bluetooth though

agners, 2022-04-22:
WiFi firmware is now part of the OS 8.x as well, just checked on my Intel NUC Gen11 which comes with a Wi-Fi 6 AX201 adapter. Closing this then.

closed 2022-04-22 by agners

snow-mix, 2022-07-19:
Good afternoon. Faced with a similar error. After connecting the m2 module wi-fi 6 ax210, WiFi is not detected in the system. Attached the dmesg log.
[dmesg_wifi_ax210_error.log](https://github.com/home-assistant/operating-system/files/9144373/dmesg_wifi_ax210_error.log)

<html>
<body>

Version | core-2022.7.5
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.10.5
Operating System Family | Linux
Operating System Version | 5.15.45
CPU Architecture | x86_64
Configuration Directory | /config

</body>
</html>

<details><summary>hassio</summary>

<html>
<body>

Host Operating System | Home Assistant OS 8.2
-- | --
Update Channel | stable
Supervisor Version | supervisor-2022.07.0
Agent Version | 1.2.1
Docker Version | 20.10.14
Disk Total | 234.0 GB
Disk Used | 3.0 GB
Healthy | true
Supported | true
Board | generic-x86-64
Supervisor API | ok
Version API | ok
Installed Add-ons | File editor (5.3.3), Terminal & SSH (9.6.0)

</body>
</html>

</details>

agners, 2022-07-19:
Please create a new issue as the OS version seems to have the correct firmwares, but there seems to be a kernel issue.