home-assistant/operating-system #1859: HAOS fails to start when LAN is not connected to the internet

ashald, 2022-04-21:
### Describe the issue you are experiencing

I had this issue with HAOS 7.1 and haven't tried to reproduce with recent releases, though I haven't seen any relevant changes in release notes since then.

I had an issue where my ISP connection dropped and while my LAN was working fine, it was disconnected from the internet.
I observed 2 major issues:
1. Local DNS resolution did not work properly
2. HAOS couldn't start on reboot

The sequence of events went as following.
1. I noticed missing sensor data from my ESPHome nodes (I use local DNS names to connect to them in HA, and my LAN DNS server advertised via DHCP was operating fine)
2. I couldn't figure out what's causing it and decided to reboot HA VM
3. HA did not come back online, so I connected to VM serial console and saw that it's stuck waiting for connection to NTP server. It never went past that as is.
4. As a workaround, I disconnected the VM from any network whatsoever, and upon reboot it did start eventually
5. I re-connected VM to the network and I observed that it consistently cannot resolve any local DNS names defined in my LAN DNS server
6. I remember checking DNS logs and I don't recall what exactly they were saying, but it was obvious that it's not attempting to use the LAN DNS server advertised via DHCP

I understand that my description of the issue lacks lots of important detail, but unfortunately I lost logs I collected. Also, I believe it suggests that there is rather a fundamental issue that should be easy to reproduce in an isolated manner (and is tricky to do for me as my family doesn't like when HA is down).

At the very least I'm looking for 2 things:
1. Make sure HAOS can boot when LAN is not connected to the internet (do I need to run local NTP and configure it somehow?)
2. Force local DNS server to be used for all DNS resolution

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Make sure that HA is connected to LAN that is not connected to the internet
2. Reboot HA
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I lost logs. :(
```

### Anything in the Host logs that might be useful for us?

```txt
I lost logs. :(
```

agners, 2022-04-21:
> 1. Local DNS resolution did not work properly

With "Local DNS resolution did not work properly" you mean that you promote a local DNS server via DHCP and use a custom top-level domain for resolution, is that correct?

There was a problem that sometimes the dns-plugin would fallback erroneously. However, this should be fixed since  https://github.com/home-assistant/plugin-dns/pull/82. This is fixed in plugin-dns version `2022.04.0` currently available on the beta channel.

> 2. HAOS couldn't start on reboot
> [...]
> 3. HA did not come back online, so I connected to VM serial console and saw that it's stuck waiting for connection to NTP server. It never went past that as is.

That should not be the case. Yes, we do wait until time is synchronized, but the wait time is limited to 90s. (FWIW, there are related discussions in this thread https://github.com/home-assistant/operating-system/issues/1686).

What happened in your case after the 90s passed?

ashald, 2022-04-24:
> With "Local DNS resolution did not work properly" you mean that you promote a local DNS server via DHCP and use a custom top-level domain for resolution, is that correct?

Yes, all network clients get my local DNS server via DHCP.
I have a domain like `mydomain.com`, and I use `lan.mydomain.com` for my LAN devices. Normally, it works without a hitch. But when my LAN was disconnected from the internet, none of the names were resolvable for HomeAssistant - and logs suggested that it was not talking to my local DNS server at all.

Thanks, I'll follow the thread for dns-plugin.
Does it mean that the issue was not in HAOS but in HA on its own, and the fix will be part of HA release?

> What happened in your case after the 90s passed?
It indeed was showing something like "waiting for time synchronization 50s / 1m 30s."
I don't recall what exactly happened, but  HA was effectively down. I think it went ahead, but eventually HA did not start - I got dropped into a shell (no the HA cli as it happens normally, but bash or whatever was that), and none of HA services seemed to run (e.g., web UI did not ever load etc).

agners, 2022-04-25:
> Does it mean that the issue was not in HAOS but in HA on its own, and the fix will be part of HA release?

It's likely related to the dns-plugin. Supervisor (and its plugins) are released separately. 

> I don't recall what exactly happened, but HA was effectively down. I think it went ahead, but eventually HA did not start - I got dropped into a shell (no the HA cli as it happens normally, but bash or whatever was that), and none of HA services seemed to run (e.g., web UI did not ever load etc).

Hm, the HA cli should come up, so it seems something went wrong there. If you experience this again, you can check the state of the system using the observers web interface (`http://<ha-host>:4357`, `4357` representing `help` in telephone keypad). 

On the shell, outputs of the following commands are helpful:
```
docker ps
journalctl -u hassos-supervisor
```

buczoprazy, 2022-05-05:
coz this is ducking shite! that crap do not boot when offline!!! Which idiot post on first page privacy first?! peace shit where no internet on boot cli not start and jump in to emergency console!

closed 2022-08-11 by github-actions[bot]

nepozs, 2022-08-17:
It isn't ressolved at all, HA lost ability to run offline.
@agners Please just point me where current issue should be created, as HAOS based installations were able to run offline many versions ago (last test when HAOS+HA have started properly offline on NUC platform was december 2020).

agners, 2022-08-18:
@nepozs create a issue in this repository: https://github.com/home-assistant/operating-system