home-assistant/operating-system #1861: Failed to start Journal and Network name resolution after few hours

malosaa, 2022-04-24:
### Describe the issue you are experiencing

Well  2 days ago i had a power outage, so i put home assistant back up, and now every couple of hours i get this error in the screen.
Aswell sometimes that there is not enough memory.. 
The strage thing is Home Assistant runs in vm with 3gb reserved and it consumes max 18% ram of my host.
[image]

I never had troubles at all.. 

What could be the problem ? 

regards 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

it just happens 

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No logs as homeassistant fully block, and when opening logs nothing strange in it.
```

### System Health information

## System Health

version | core-2022.4.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4834
Installed Version | 1.24.5
Stage | running
Available Repositories | 1036
Downloaded Repositories | 25

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | beta
supervisor_version | supervisor-2022.04.2
docker_version | 20.10.9
disk_total | 31.3 GB
disk_used | 15.2 GB
healthy | true
supported
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Tor (3.0.5), AirCast (3.3.0), ESPHome (2022.3.1), Grafana (7.5.2), InfluxDB (4.4.1), MariaDB (2.4.0), Mosquitto broker (6.0.1), Portainer (2.0.0), SSH & Web Terminal (10.1.1), Samba Backup (5.0.0), Samba share (9.5.1), TasmoAdmin (0.17.0), Terminal & SSH (9.3.0), Zigbee2mqtt (1.25.0-1), Portainer (2.9.2), Frigate NVR Proxy (1.3), SQLite Web (3.3.0), AdGuard Home (4.5.1), NGINX Home Assistant SSL proxy (3.1.1), Nginx Proxy Manager (0.11.0), Duck DNS (1.14.0), Node-RED (11.1.1), Assistant Relay (0.7.4), Google Assistant Webserver (0.0.4), WireGuard (0.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 16
views | 26
mode | storage

</details>

### Additional information

[image]

malosaa, 2022-04-25:
Well this morning i checked again and did see i got that low memory error again.. 

Is there a way to check what is causing this ? 
When i look at usage of home assistant and ram its both on 1.8% the other one is 3.5%

agners, 2022-04-25:
> Aswell sometimes that there is not enough memory..
... 
> Well this morning i checked again and did see i got that low memory error again..

Where do you see the low memory error/what is it saying exactly?

> Is there a way to check what is causing this ?

You can observe memory usage of individual addons via UI or using the shell and check with Linux commands such as top. But both cases require you to actively check the values.

malosaa, 2022-04-25:
> You can observe memory usage of individual addons via UI or using the shell and check with Linux commands such as top. But both cases require you to actively check the values.

it says that memory issue something with pythin killed process etc. i accedently forget to make a screenshot.
I see this problem on many systems arround forums.. 

I only 1 time had that issue, so i upped memory to 3gb. but this makes no sense, as my cup and ram in the gui are very very low...

agners, 2022-04-25:
> it says that memory issue something with python killed process etc. i accedently forget to make a screenshot.
I see this problem on many systems arround forums..

When a process uses more and more memory, Linux at one point invokes the OOM killer (out-of-memory killer). The OOM killer tries to identify a "bad actor", which usually means the process which uses all the memory. Typically, the OOM killer kills "the right" process. In other words: It seems that in your system a Python process was using all the memory.

In Home Assistant there are multiple Python processes. Most notable Home Assistant Core. To prove that it was Home Assistant Core, you would need to check the logs on the host system specifically. It is not unheard of that Core has memory issues, there are [open and closed issue reports](https://github.com/home-assistant/core/issues?q=is%3Aissue+memory+leak+is%3Aopen). 

It could be that a nightly job needs some more memory.. If it was a one time thing, and it got resolved with adding more memory, I wouldn't worry about.

agners, 2022-04-25:
Assuming the memory thing was a one time thing, is this issue report then about:

> and now every couple of hours i get a error in the screen.

specifically

> Failed to start Network Name Resolution

Correct?

agners, 2022-04-25:
If so, can you type the following in the console:

```
ha > login
# systemctl status systemd-resolved.service
```

malosaa, 2022-04-25:
> Assuming the memory thing was a one time thing, is this issue report then about:
> 
> > and now every couple of hours i get a error in the screen.
> 
> specifically
> 
> > Failed to start Network Name Resolution
> 
> Correct?

Actually that Failed to start Network Name Resolution is gone, don't know  how.

And i remember now that at 00:00 there will be a backup created, but the files are only 300mb, so thats very low.
But i will need to keep an eye tonight and see if the backup is causing problems.

But nothing in the logs, even also not in the supervisor log.

[image]

malosaa, 2022-04-26:
I don't get it at all.

Yesterday i did uupdate to last homeassistant version, and now not even 1 error... 
Its weird as when i had the outage it wass running core-2022.4.6 and i never had issue but then i got every night that memory issue.

now with core-2022.4.7, i just checked my pc and no error at all..
So i will wait 1 more day, and than i probably need to close this issue... as i think its resolved by HASS Core version.

malosaa, 2022-04-26:
@agners 

Right now after i wrote that message, i can't connect to home-assistant.
Getting time-out all the time.

I don't know why, even on the host itself it's unreachable.

[image]

malosaa, 2022-04-26:
I think i found the problem its the Watchmen add-on from hacs.
I did see a video from a homeassistant youtuber, installing this add-on. 
So i wanted to try it out.

So i went to that device and try to add it again and it fully blocks homeassistant and make it unreachable..

So i think this add-on was causing all these problems.

i opened a issue on theyr github aswell. 

[image]

malosaa, 2022-04-26:
Well right now i'm having more and more issue's as i can't use home assistant anymore.
When it boots up i can access, but after some time it gets inaccessable... i deleted that add-on before..

Any idea's how i can check whats going on ?
I have access to the drive by samba plugin.

also the logs showing nothing.

regards

agners, 2022-04-26:
You can check the observer at `http://<home-assistant-instance>:4357/`.

Otherwise getting SSH access to the underlying OS is helpful, see [SSH access to the host](https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host) how to setup SSH access. In the SSH shell you can check the logs of Supervisor using

```
journalctl -u hassos-supervisor.service
```

malosaa, 2022-04-26:
Thanks

> You can check the observer at `http://<home-assistant-instance>:4357/`.
>
> Otherwise getting SSH access to the underlying OS is helpful, see [SSH access to the [...]

Thanks, 
i did the observer url and it said system unhealthy
When i go to the gui inside home assistant it says system healthy True.

I will need to let it run without a restart for a day, as these problems started after my power outage, and before that outage i installed that add-on that i just updated and caused home assistant to not respond on https/http but its still running.

After 20minutes i could access trough https but nothing in the logs.. really weird.

malosaa, 2022-04-27:
Right now i logged in again, and nothingstrange happened but i got 2 new errors,

```
Logger: homeassistant.components.hassio.handler
Source: components/hassio/handler.py:254
Integration: Home Assistant Supervisor ([documentation](https://www.home-assistant.io/integrations/hassio), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+hassio%22))
First occurred: April 26, 2022, 19:55:53 (1 occurrences)
Last logged: April 26, 2022, 19:55:53

/store return code 500
```

```
Logger: homeassistant.components.hassio
Source: components/hassio/__init__.py:633
Integration: Home Assistant Supervisor ([documentation](https://www.home-assistant.io/integrations/hassio), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+hassio%22))
First occurred: April 26, 2022, 19:55:53 (1 occurrences)
Last logged: April 26, 2022, 19:55:53

Can't read Supervisor data:
```

if this doesnt have something to do with the OS version i will close this issue, as the 2 errors from before are fixed

agners, 2022-04-27:
Yeah those seem more Supervisor related. Thanks.

malosaa, 2022-04-27:
> Yeah those seem more Supervisor related. Thanks.

also thanks for assisting

closed 2022-04-27 by malosaa