home-assistant/operating-system #1862: Disk full on boot, new install or not

tompointexe, 2022-04-24:
### Describe the issue you are experiencing

I just restarted my homeassistant install tonight, having the 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Download latest OVA
2. import in VM
3. Start, will start normaly, supervisor will not
4.  type the df command
5. disk is full

### Anything in the Supervisor logs that might be useful for us?

```txt
[WARN] Home Assistant CLI is not starting ! Jump into emergency console
```

### Anything in the Host logs that might be useful for us?

```txt
[WARN] Home Assistant CLI is not starting ! Jump into emergency console
```

agners, 2022-04-25:
> checked multiple time, i'm unable to do anything

How did you check?

The root file system is on a read-only partition, which is always considered "full". But that is regular behavior.

However, the data disk (mounted at `/mnt/data`) shown on the UI, should not be full. Our default VM images have a 32GB disk size by default.

tompointexe, 2022-04-25:
I've downloaded a new .qcow2 file and placed it in the root, so maybe having / full is not the issue, maybe my network config or something else is setting off HASS CLI, and i searched in the wrong direction
any help would be welcomed then :+1:

agners, 2022-04-25:
What kind of network are you using? The recommended setup is to use bridge mode. The system will need internet connectivity on first boot to download Home Assistant Core.

What virtualization environment are you using?

tompointexe, 2022-04-25:
I'm using a bridge in a specific vlan on KVM, host os is Debian / OpenMediaVault

agners, 2022-05-03:
This sounds a lot like a Internet connectivity issue. Home Assistant OS downloads from the GitHub container registry.

Make sure your network provides IP addresses via DHCP, a valid Gateway which routes to the internet, does not block DNS,  http or https traffic.

closed 2022-05-03 by agners