home-assistant/operating-system #1869: Intel Wireless-AC 9462 not detected

piechade, 2022-04-26:
### Describe the issue you are experiencing

Based on the documentation of Intel the Intel Wireless-AC 9462 should be supported. 
It is using the iwlwifi-9000-pu-b0-jf-b0-34.ucode Firmware.
But the OS is not detecting the WIFI Adapter on the new NUC11ATKPE.
The BR2_PACKAGE_LINUX_FIRMWARE_IWLWIFI_9XXX should handle the adpater.

https://ark.intel.com/content/www/us/en/ark/products/217668/intel-nuc-11-essential-kit-nuc11atkpe.html
https://www.intel.de/content/www/us/en/products/sku/125192/intel-wirelessac-9462/specifications.html
https://www.intel.de/content/www/us/en/support/articles/000005511/wireless.html

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 7.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Buy a NUC11ATKPE NUC
2. Install the SSH & Web Terminal Addon from the Community Repo
3. RUN nmcli device wifi

### Anything in the Supervisor logs that might be useful for us?

```txt
22-04-26 17:11:00 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2022.02.0 - Host
22-04-26 17:11:00 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-04-26 17:11:00 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
22-04-26 17:11:00 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/generic-x86-64-homeassistant with version 2022.4.7
22-04-26 17:11:01 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 7.6 / BootSlot A
22-04-26 17:11:01 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
22-04-26 17:11:01 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
22-04-26 17:11:01 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/f1c878cb repository
22-04-26 17:11:01 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
22-04-26 17:11:01 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/abc67f79 repository
22-04-26 17:11:01 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/774437fd repository
22-04-26 17:11:01 INFO (MainThread) [supervisor.store] Loading add-ons from store: 82 all - 82 new - 0 remove
22-04-26 17:11:01 INFO (MainThread) [supervisor.addons] Found 12 installed add-ons
22-04-26 17:11:01 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/amd64 with version 11.1.1
22-04-26 17:11:01 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to f1c878cb/amd64-addon-adsb-multi-portal-feeder with version 1.0.3
22-04-26 17:11:01 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/sqlite-web/amd64 with version 3.3.0
22-04-26 17:11:01 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/home-panel/amd64 with version 2.4.0
22-04-26 17:11:01 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-nginx_proxy with version 3.1.1
22-04-26 17:11:01 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.0.0
22-04-26 17:11:01 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/fredrikbaberg/addon-3dprinter-amd64 with version 2021.11.7
22-04-26 17:11:01 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 10.1.1
22-04-26 17:11:01 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio-amd64 with version 2022.4.0
22-04-26 17:11:01 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.3.3
22-04-26 17:11:01 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-check_config with version 3.10.0
22-04-26 17:11:01 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/spotify/amd64 with version 0.11.0
22-04-26 17:11:01 INFO (MainThread) [supervisor.backups.manager] Found 2 backup files
22-04-26 17:11:01 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
22-04-26 17:11:01 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
22-04-26 17:11:01 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
22-04-26 17:11:01 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-04-26 17:11:01 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-04-26 17:11:01 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-04-26 17:11:01 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
22-04-26 17:11:01 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-04-26 17:11:01 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-04-26 17:11:01 INFO (MainThread) [__main__] Running Supervisor
22-04-26 17:11:01 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
22-04-26 17:11:01 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
[... 154 lines cut ...]
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2022.4.0), Spotify Connect (0.11.0), Node-RED (11.1.1), File editor (5.3.3), Home Panel (2.4.0), NGINX Home Assistant SSL proxy (3.1.1), 3DPrinter (2021.11.7), SSH & Web Terminal (10.1.1), SQLite Web (3.3.0), Studio Code Server (5.0.0), Check Home Assistant configuration (3.10.0), ADS-B Multi-Portal Feeder (1.0.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 14
views | 6
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2022-04-26:
The firmware file in question seems to be present on OS 7.6:

```
# find /usr/lib -name iwlwifi-9000-pu-b0-jf-b0-34.ucode
/usr/lib/firmware/iwlwifi-9000-pu-b0-jf-b0-34.ucode
```

However, what your system is looking for is a different firmware file: `iwlwifi-QuZ-a0-jf-b0-59.ucode`. This is present in OS 8.x. Can you try the latest 8.0.rc3 release? You can upgrade when joining the Beta channel (in Supervisor).

piechade, 2022-04-26:
When it's planet to release 8.0?

agners, 2022-04-26:
Is your main install on an internal disk? Then its _not_ safe to run a test installation on the USB drive: HAOS uses a file system label to mount the data partition. It might use the data partition of the internal disk... It will be safe if you plug-out the internal disk during your test.

agners, 2022-04-26:
> When it's planned to release 8.0?

Probably in the next 2 weeks or so, depends on the progress of #1830.

piechade, 2022-04-26:
> Is your main install on an internal disk? Then its _not_ safe to run a test installation on the USB drive: HAOS uses a file system label to mount the data partition. It might use the data partition of the internal disk... It will be safe if you plug-out the internal disk during your test.

I have the os running on the internal drive. Why it's not save to flash 8.0 on a USB driver and run it? It should only touch the USB drive? Or is there something that I'm not aware of?

agners, 2022-04-26:
> It should only touch the USB drive? Or is there something that I'm not aware of?

Well it doesn't. Linux udev detects all drives automatically, and _may_ pick the data partition of the internal disk... I know its not ideal, but its how it works today :(

piechade, 2022-04-27:
https://linux-hardware.org/?probe=60b915c436
Of topic question, can I get also the rest working?

JasperLake [UHD Graphics]
Jasper Lake LPSS: UART Controller #0
Management Engine Interface

agners, 2022-04-27:
> JasperLake [UHD Graphics]

The drivers should already been enabled, can you test with 8.0.rc3 which comes with the new Linux 5.15 kernel?

> Jasper Lake LPSS: UART Controller #0
> Management Engine Interface

Drivers are not enabled. Those devices don't seem very useful for HAOS. Do you need them? What is the use case?

piechade, 2022-04-27:
Management Engine Interface makes no sense.
But is the UART Controller not the thing on the top of the nuc. You can use it control relay, or I'm wrong.

piechade, 2022-04-27:
> UART stands for Universal asynchronous receiver-transmitter, which is essentially a serial port interface. Not sure what thing you mean ontop of the NUC, but its unlikely to be controlled via UART.

Forgot about it, I never checked the manual. That's a normal front panel header with HDD led, power etc. I have no idea why this outside the case. If Wi-Fi and graphics work, I'm happy.

closed 2022-04-27 by agners