home-assistant/operating-system #1877: VDI image is taking up more space on host than it's actually using.

vandad51, 2022-04-30:
### Describe the issue you are experiencing

Using up-to-date VirtualBox on a VDI HAOS image, it seems as though HA has silently been chewing up my free space on my Windows 10 host.

When I open up glances to see how much the OS thinks it's using, this is what it shows.
> FILE SYS	       Used	             Total
/backup (/dev/sda8)	10.5G	30.8G
/data (/dev/sda8)	10.5G	30.8G
/ssl (/dev/sda8)	10.5G	30.8G
/share (/dev/sda8)	10.5G	30.8G
/media (/dev/sda8)	10.5G	30.8G
/config (/dev/sda8)	10.5G	30.8G
/addons (/dev/sda8)	10.5G	30.8G
_olv.conf (/dev/sda8)	10.5G	30.8G
_hostname (/dev/sda8)	10.5G	30.8G
_tc/hosts (/dev/sda8)	10.5G	30.8G
FOLDERS		Size
/config		1006M
/share		0
/backup		204M
/media		450K

But yet Vbox says it's using up 30.26 of 30GB. And where my VDI file is on my computer, it is also physically 30.26 GB

[image]

I've scoured the internet to find ways to shrink the image, but not sure how to do inside HA. I tried creating a test HA image/instance but every time I go to create it, the haos_ova-7.6.vdi says it's already locked at 32GB and I can't shrink it using the Virtual Media Manager. When I went ahead and created it anyways, the virutal size was 32GB and the actual size was 816MB.

In the host file PasteBin, the host even says theres only 34.1% space used.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

I'm consistently trying to clear space up in HA, but have ran out of solutions.

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
HA Host Logs

https://paste.in/6s1Hui
```

### System Health information

## System Health

version | core-2022.4.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | America/Winnipeg

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.24.0
Stage | running
Available Repositories | 1094
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 30.8 GB
disk_used | 10.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (11.1.1), Samba share (9.6.0), Studio Code Server (5.0.1), Mosquitto broker (6.0.2), Terminal & SSH (9.3.0), Duck DNS (1.14.0), File editor (5.3.3), Log Viewer (0.13.0), Zigbee2mqtt (1.25.0-1), MariaDB (2.4.0), Nginx Proxy Manager (0.11.0), NGINX Home Assistant SSL proxy (3.1.1), Alsa & PulseAudio Fix (3.4), VLC (0.1.3), AdGuard Home (4.5.0), Home Assistant Google Drive Backup (0.106.2), Zigbee2mqtt Edge (edge), Glances (0.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 4
views | 4
mode | storage

</details>

agners, 2022-05-02:
HAOS makes use of fstrim/discard mechanism. It seems that VirtualBox does not support the fstrim by default. First, you need to use a SATA emulated disk controller instead of IDE. Then, you need to enable the `discard` option manually using the following command:

```
VBoxManage storageattach <VM-NAME> --storagectl "SATA" --port 0 --device 0 --nonrotational on --discard on
```

agners, 2022-05-03:
I've documented this in the installation instructions, I consider this solved https://github.com/home-assistant/home-assistant.io/pull/22592.

closed 2022-05-03 by agners

vandad51, 2022-05-05:
@agners After following through with those instructions, I was not able to use "SATA" as the controller, rather "AHCI". After inputting the command you now have in the instructions, my setup is now stuck with this at start up.

[image]

Need some direction here. I've tried reloading with the controller IDE, but it still boots into that screen

salzrat, 2022-10-18:
Just went through similar hoops, spent two days debugging it. Probably the boot order got changed by adding the new controller. To fix this, type "exit" in the EFI console, then you will land in the vbox EFI bios. In the Boot Manager you can select the vdi image and press enter, then HA should boot. If this works, you should change the boot order so you don't have to do this every time. You can do that in the Boot Manager Preferences (or it is called something similar). Just make sure you move the HDD to the top spot.