home-assistant/operating-system #1879: Z-Wave JS add-on automatically reinstalls itself breaking Z-Wave service provided by Z-Wave JS to MQTT

deftdawg, 2022-05-03:
### Describe the issue you are experiencing

I have `Z-Wave JS to MQTT` installed on my Home Assistant instance, I was under the impression that it supersedes 
the `Z-Wave JS`, so I uninstalled the `Z-Wave JS`...  Everything worked fine.

Unfortunately, something (guessing applying an HA core update) is causing `Z-Wave JS` to be reinstalled automatically.  Each time it is reinstalled, it starts on boot and breaks Z-Wave connectivity in Home Assistant.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 7.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Shutdown and uninstall ZWave JS
2.  Install and start ZWave-to-MQTT
3. Update Home Assistant

### Anything in the Supervisor logs that might be useful for us?

```txt
Including the Host log GitHub errors on issue create: `There was an error creating your Issue: body is too long (maximum is 65536 characters).`. Doesn't look like anything interesting in the log anyway though.
```

### Anything in the Host logs that might be useful for us?

```txt
Including the Host log GitHub errors on issue create: `There was an error creating your Issue: body is too long (maximum is 65536 characters).`. Doesn't look like anything interesting in the log anyway though.
```

### System Health information

## System Health

version | core-2022.4.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | America/Toronto

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.21.0
Stage | running
Available Repositories | 1031
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 29.0 GB
disk_used | 21.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.14.0), Check Home Assistant configuration (3.10.0), Studio Code Server (5.0.2), Terminal & SSH (9.4.0), Mosquitto broker (6.0.2), Snapdrop (0.1-alpha), Z-Wave JS to MQTT (0.39.0), ESPHome (2022.4.0), WireGuard (0.6.0), AdGuard Home (4.5.1), Samba Backup (5.0.0), File editor (5.3.3), Z-Wave JS (0.1.57)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 11
mode | storage

</details>

deftdawg, 2022-05-03:
Upon further investigation I can confirm it is `core` restarts that trigger this behaviour...

Each time core is restarted, if `zwave_core_js` does not exist, it is created with `start on boot`...  since its slow to create `Z-wave to MQTT` succeeds in starting up the first time, but subsequent restarts `zwave_core_js` is started ahead of `Z-wave to MQTT` -> this breaks all zwave in HA.

Is this a bug in HA core or HA OS?  I'm not sure.

KevinCathcart, 2022-05-04:
This would be a bug in Supervisor, not OS, assuming if it were not a core issue. However, this would in fact be a core issue or at a minimum of core-relevant support request.

I think you may have a weird configuration. Like you either have two config entries for the ZWaveJS configuration, one configured for zwave2mqtt, and the other configured to use the addon, or you somehow have a single configuration that somehow both has the MQTT websocket address, while also having the `use_addon` flag set to true.

If it is the former, then in your integrations list instead of seeing a message like "# devices and # entities" under the ZWave JS for the zwave js listing, it would list two or more lines with arrows. If this is happening, you simply need to figure out which one is the one that is trying to use the Zwave JS addon, instead of ZWaveJS2MQTT, and delete it. 

Otherwise, assuming you did not hand-edit anything in the `.storage` folder, you came across a core bug, and should file an issue over in core specifying the zwave_js integration, so they can fix whatever allowed the invalid config entry. (although in this case you can probably could just fix things yourself by clicking the configure button on the integration card, making sure the use official addon box is not checked, and making sure the zwavejs2mqtt address is specified for the websocket. I would expect doing this would result in a fixed config entry.

closed 2022-05-05 by github-actions[bot]