home-assistant/operating-system #1880: The VMware virtual machine cann't run the system

lff110, 2022-05-04:
### Describe the issue you are experiencing

My computer is a win10 system. There is always an error prompt shown in the figure below when using VMware virtual machine, but using other virtual machines in win10 system or win11 using VMware virtual machine can connect to the router's network normally without this error. What's the matter with it?

[image]
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

6.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

The prompt when starting the system is shown in the figure below.
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
22-05-04 18:43:39 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
22-05-04 18:43:39 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
22-05-04 18:43:39 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-05-04 18:43:39 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-05-04 18:43:39 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-04 18:43:39 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
22-05-04 18:43:39 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-05-04 18:43:39 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-04 18:43:39 INFO (MainThread) [__main__] Running Supervisor
22-05-04 18:43:39 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
22-05-04 18:43:39 INFO (MainThread) [supervisor.supervisor] Fetching AppArmor profile https://version.home-assistant.io/apparmor.txt
22-05-04 18:43:44 ERROR (MainThread) [supervisor.supervisor] Can't fetch AppArmor profile https://version.home-assistant.io/apparmor.txt: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
22-05-04 18:43:44 CRITICAL (MainThread) [supervisor.supervisor] Abort update because of an issue with AppArmor: Can't fetch AppArmor profile https://version.home-assistant.io/apparmor.txt: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
22-05-04 18:43:44 CRITICAL (MainThread) [supervisor.core] Can't update Supervisor! This will break some Add-ons or affect future versions of Home Assistant!
22-05-04 18:43:44 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
22-05-04 18:43:44 INFO (MainThread) [supervisor.core] Supervisor reboot detected
22-05-04 18:43:44 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
22-05-04 18:43:44 INFO (MainThread) [supervisor.core] Supervisor is up and running
22-05-04 18:43:44 INFO (MainThread) [supervisor.host.info] Updating local host information
22-05-04 18:43:44 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.module] Create new suggestion SuggestionType.CREATE_FULL_BACKUP - ContextType.SYSTEM / None
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-05-04 18:43:44 INFO (MainThread) [supervisor.jobs] 'CheckAddonPwned.run_check' blocked from execution, no supervisor internet connection
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.PLUGIN
22-05-04 18:43:44 INFO (MainThread) [supervisor.host.services] Updating service information
22-05-04 18:43:44 INFO (MainThread) [supervisor.host.network] Updating local network information
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.CORE
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-05-04 18:43:44 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-04 18:43:44 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not healthy
22-05-04 18:43:44 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-05-04 18:43:44 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-05-04 18:44:44 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Multicast plugin!
22-05-04 18:44:44 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
22-05-04 18:44:44 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_multicast application
22-05-04 18:44:44 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2021.04.0 - Host
22-05-04 18:45:44 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Multicast plugin!
22-05-04 18:45:44 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
[... 417 lines cut ...]
[   94.516969] hassio: port 2(veth02be998) entered blocking state
[   94.516970] hassio: port 2(veth02be998) entered disabled state
[   94.517166] device veth02be998 entered promiscuous mode
[   94.517333] hassio: port 2(veth02be998) entered blocking state
[   94.517335] hassio: port 2(veth02be998) entered forwarding state
[   94.797865] eth0: renamed from veth542b29f
[   94.801609] hassio: port 2(veth02be998) entered disabled state
[   94.801663] IPv6: ADDRCONF(NETDEV_CHANGE): vethe71ae6e: link becomes ready
[   94.801705] docker0: port 1(vethe71ae6e) entered blocking state
[   94.801706] docker0: port 1(vethe71ae6e) entered forwarding state
[   94.801785] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
[   94.868627] eth1: renamed from vethbdc6c2a
[   94.877395] IPv6: ADDRCONF(NETDEV_CHANGE): veth02be998: link becomes ready
[   94.877416] hassio: port 2(veth02be998) entered blocking state
[   94.877417] hassio: port 2(veth02be998) entered forwarding state
[  101.100750] kauditd_printk_skb: 69 callbacks suppressed
[  101.100751] audit: type=1334 audit(1651663319.547:87): prog-id=11 op=LOAD
[  101.100753] audit: type=1334 audit(1651663319.547:88): prog-id=12 op=LOAD
[  101.197337] audit: type=1334 audit(1651663319.644:89): prog-id=13 op=LOAD
[  101.197413] audit: type=1334 audit(1651663319.644:90): prog-id=14 op=LOAD
```

### Additional information

I tried version 6.6 and 6.4, but they can't be used

lff110, 2022-05-04:
Other host logs 1
[    0.000000] Linux version 5.10.98 (builder@73cd380cf5ea) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g8bbb32c1) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Tue Feb 8 16:57:14 UTC 2022
[    0.000000] Command line: apparmor=1 security=apparmor console=ttyS0 console=tty1 bootchooser.active=A  root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait systemd.machine_id=6bb6d7b5beeb4a80af95afff38f13c48 zram.enabled=1 zram.num_devices=3
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000efaafff] usable
[    0.000000] BIOS-e820: [mem 0x000000000efab000-0x000000000efaefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000efaf000-0x000000000efbcfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000efbd000-0x000000000efc1fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000efc2000-0x000000000efc6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000efc7000-0x000000000fee5fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fee6000-0x000000000ff55fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ff56000-0x000000000ff71fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ff72000-0x000000000ff75fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000ff76000-0x000000007fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x0e5f7018-0x0e5ff057] usable ==> usable
[    0.000000] e820: update [mem 0x0e5f7018-0x0e5ff057] usable ==> usable
[    0.000000] e820: update [mem 0x0e5f4018-0x0e5f6057] usable ==> usable
[    0.000000] e820: update [mem 0x0e5f4018-0x0e5f6057] usable ==> usable
[    0.000000] e820: update [mem 0x0e5f2018-0x0e5f3857] usable ==> usable
[    0.000000] e820: update [mem 0x0e5f2018-0x0e5f3857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000000e5f2017] usable
[    0.000000] reserve setup_data: [mem 0x000000000e5f2018-0x000000000e5f3857] usable
[    0.000000] reserve setup_data: [mem 0x000000000e5f3858-0x000000000e5f4017] usable
[    0.000000] reserve setup_data: [mem 0x000000000e5f4018-0x000000000e5f6057] usable
[    0.000000] reserve setup_data: [mem 0x000000000e5f6058-0x000000000e5f7017] usable
[    0.000000] reserve setup_data: [mem 0x000000000e5f7018-0x000000000e5ff057] usable
[    0.000000] reserve setup_data: [mem 0x000000000e5ff058-0x000000000efaafff] usable
[    0.000000] reserve setup_data: [mem 0x000000000efab000-0x000000000efaefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000efaf000-0x000000000efbcfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000efbd000-0x000000000efc1fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000efc2000-0x000000000efc6fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000efc7000-0x000000000fee5fff] usable
[    0.000000] reserve setup_data: [mem 0x000000000fee6000-0x000000000ff55fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000ff56000-0x000000000ff71fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000000ff72000-0x000000000ff75fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000ff76000-0x000000007fffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] efi: EFI v2.40 by VMware, Inc.
[    0.000000] efi: SMBIOS=0xefc2000 ACPI 2.0=0xff63000 MEMATTR=0xf47ba98 
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.18452719.B64.2108091906 08/09/2021
[    0.000000] vmware: hypercall mode: 0x02
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 2903.998 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using clock offset of 4222163520 ns
[    0.000015] tsc: Detected 2903.998 MHz processor
[    0.025288] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.025292] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.025305] last_pfn = 0x80000 max_arch_pfn = 0x400000000
[    0.025318] MTRR default type: uncachable
[    0.025319] MTRR fixed ranges enabled:
[    0.025320]   00000-9FFFF write-back
[    0.025320]   A0000-FFFFF uncachable
[    0.025321] MTRR variable ranges enabled:
[... 1128 lines cut ...]
[... cut ...]
[    0.523880] pci_bus 0000:20: resource 2 [mem 0xf8700000-0xf87fffff 64bit pref]
[    0.523949] pci_bus 0000:21: resource 0 [io  0x4000-0x4fff]
[    0.524017] pci_bus 0000:21: resource 1 [mem 0xfcc00000-0xfccfffff]
[    0.524077] pci_bus 0000:21: resource 2 [mem 0xf8600000-0xf86fffff 64bit pref]
[    0.524147] pci_bus 0000:22: resource 0 [io  0x3000-0x3fff]
[    0.524200] pci_bus 0000:22: resource 1 [mem 0xfcb00000-0xfcbfffff]
[    0.524260] pci_bus 0000:22: resource 2 [mem 0xf8500000-0xf85fffff 64bit pref]

agners, 2022-05-04:
> My computer is a win10 system. There is always an error prompt shown in the figure below when using VMware virtual machine, but using other virtual machines in win10 system or win11 using VMware [...]

So what you are saying is that Windows 10 + VMware causes issues? But other combinations work?

Can you try a more recent version of Home Assistant OS (e.g. 7.6 or even better 8.0.rc3)?

lff110, 2022-05-05:
I've tried the version 7.6 you said, and it can't work normally. I haven't tried another version, and I don't know where to download that version. I saw on our domestic website that it is more convenient for others to use homeassistant system, so I don't know whether the other system you mentioned can support Xiaomi series plug-ins.

buczoprazy, 2022-05-05:
coz this is ducking shite! that crap do not boot when offline!!! Which idiot post on first page privacy first?! peace shit!

lff110, 2022-05-06:
I just used this version of the system you said, but it still has the same problem

lff110, 2022-05-06:
> 您可以从此处的 GitHub 发布页面下载 RC 版本：https://github.com/home-assistant/operating-system/releases

I just used this version of the system you said, but it still has the same problem.

agners, 2022-05-06:
Hm, in this case it seems that the system cannot access the NTP Server. We are using 1.1.1.1 as NTP server, can your network reach that IP as NTP server?

Alternatively, 8.0.rc3 should pick up a DHCP provided NTP server.

lff110, 2022-05-07:
I just used the ping command to Ping the address you mentioned. I found that packet loss is serious. Sending 122 pieces of data can normally receive 97 pieces, losing 20% of the data, and the delay is relatively large, almost 300ms. Since I am in China, our domestic data access to foreign countries may be relatively strict, but I also see that our anchors are also using this service, and they have no problem using it, so I don't know what the problem is.

agners, 2022-05-08:
What kind of network emulation are you using? Maybe it is worth testing with various virtual network card types.

lff110, 2022-05-09:
I don't know what you mean. Can you take a screenshot to explain it?

lff110, 2022-05-29:
Now that the latest version of the HA system is used, the previous problem is gone, but there is a new problem, that is, the problem of pictures.I have decompressed it several times, but this error message appears every time. How can we solve it
[image]
?

closed 2022-09-03 by github-actions[bot]