home-assistant/operating-system #1884: Raspberry PI DHCP issue when HassOS is booting before the router

AdriaanBoshoff, 2022-05-05:
### Describe the issue you are experiencing

Environment:

- Raspberry Pi 4 B 4GB.
- Wired Network.
- 64GB USB 3.1
- No SD card
- Static IP on the router DHCP (not on the pi)

I experienced a power outage recently and thought my SD card got corrupted. Afterwords I set the Pi to boot from USB and bought myself a USB 3.1 stick and loaded HassOS on there.

Fast forward a day later I get home to discover I had another power outage. I also noticed I couldn't boot HassOS after the power came back on. I thought it was strange that this could happen a second time but this time with a USB.

I plugged in a monitor and discovered that it boots normally into the HassOS CLI despite some errors related to the network. I then proceeded to check if the pi lists under my router and it doesn't despite the Ethernet port data light flickering. I also tried pinging the pi as I set a static IP on my router for it without any luck.

Alright. So I did some googling and found several posts mentioning people not being able to boot after a power outage.

In my case it was clear:

1. The network port "eth0" gets detected.
2. No gateway detected.
3. No IPv4 assigned.
4. No DNS assigned.

The 4 points above made me realize this isn't a issue where the OS gets corrupted but rather a network issue.

I start searching for more answers on the web on how to set a static IP on HassOS CLI and eventually got the holy grail which is: [Using nmcli to set a static IPv4 address](https://github.com/home-assistant/operating-system/blob/7e702a2ab2ee7e76ed7f3a1be6f98a3c832b557e/Documentation/network.md#using-nmcli-to-set-a-static-ipv4-address)

I set my static IP, gateway and DNS then reboot the PI via the CLI.

SUCCESS!

While writing this I had another power outage. Nice right. HassOS was able pick up the network as soon as my router became available again without any issues.

I have attempted to reproduce this with success. See below.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.0.rc3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Set a static IP in you router for the Pi.
2. Unplug both pi and router to simulate a power outage.
3. plug both in at the same time.

You should see the HassOS boots to the CLI but has no network configuration. Rebooting the PI after the router is online doesn't make a difference.

### Anything in the Supervisor logs that might be useful for us?

```txt
No recent logs
```

### Anything in the Host logs that might be useful for us?

```txt
No recent logs
```

### System Health information

## System Health

version | core-2022.5.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | Africa/Johannesburg

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.0.rc3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.0
docker_version | 20.10.14
disk_total | 56.6 GB
disk_used | 3.5 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.1.1), File editor (5.3.3), Duck DNS (1.14.0), TasmoAdmin (0.17.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

buczoprazy, 2022-05-05:
coz this is ducking shite! that crap do not boot when offline!!! Which idiot post on first page privacy first?! peace shit

AdriaanBoshoff, 2022-05-06:
> 

You are one of the people that make developers not want to continue with their work. If you think it's so easy to develop something like this then do it yourself. You're getting this great platform for FREE! and all you're doing is making clutter in this bug report instead of contributing. I suggest you look for attention somewhere else because no one here cares about your opinion.

The developers of HassOS did a great job at making it as user friendly and expandible as possible. I commend them for this and because of you I will also donate to them.

agners, 2022-05-06:
> You should see the HassOS boots to the CLI but has no network configuration. 

Static IP via DHCP still needs the router to be up and running to get an IP. Static IP via DHCP just means that the DHCP server makes a static mapping of a MAC address to an IP, but it's the DHCP server which does that assignment. So the banner not showing an IP address is expected if the router is not online.

[image]

However, the HAOS (or rather the NetworkManager running on HAOS) will retry to contact the DHCP server to get an IP assignment. In my tests I can see that this does happen (using `login`/`ip addr show dev eth0` does show an IP address gets assigned). On the console you can use `ha network info` in the Home Assistant CLI to check the IP address again, or  `ha network reload` to update the network info (this only refreshes the networking info of Supervisor).

If you want a truly static IP, you need to configure static IP on Home Assistant side.

Also note: When you have a fresh installation, you need Internet access to download Home Assistant Core at first. The Home Assistant image only contains a first version of the Supervisor, the Core will be installed on first boot basically. This is so users can start with the latest Core version, even when OS is some months old.

agners, 2022-05-06:
> Thanks for your amazing work. Is there a way I can donate to you?

I do get paid by Nabu Casa, so if you have a Nabu Casa subscription, you will support me already :smile: . I do have GitHub sponsor active, if you want to support me directly.

Thanks for the update! Please feel free to reopen in case it turns out that the HAOS DHCP client misbehaves or similar.

closed 2022-05-06 by agners