home-assistant/operating-system #1885: Possible issue with DNS loop when using VLAN

alekslyse, 2022-05-06:
### Describe the issue you are experiencing

Network setup:

Unifi network with multiple VLAN. The port HA is plugged to has all tags passed through as it should so it can use resources on all VLANS. The issue then is that resolver in OS goes crazy and updates the DNS non stop. Example of journal. Something is happening since on my NUC the idle HA is using 0-0.1% CPU, but CORE/OS is idling on 25-35%CPU (no cameras, frigate etc)

`Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Hostname conflict, changing published hostname from 'homeassistant1724737' to 'homeassistant1724739'.
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Detected conflict on homeassistant1724739.local IN AAAA fe80::5e3a:9053:e5f:1bb0
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Hostname conflict, changing published hostname from 'homeassistant1724739' to 'homeassistant1724745'.
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Detected conflict on homeassistant1724745.local IN AAAA fe80::5e3a:9053:e5f:1bb0
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Hostname conflict, changing published hostname from 'homeassistant1724745' to 'homeassistant1724752'.
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Detected conflict on homeassistant1724752.local IN AAAA fe80::5e3a:9053:e5f:1bb0
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Hostname conflict, changing published hostname from 'homeassistant1724752' to 'homeassistant1724762'.
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Detected conflict on homeassistant1724762.local IN AAAA fe80::5e3a:9053:e5f:1bb0
Apr 30 03:47:24 homeassistant.local systemd-resolved[350]: Hostname conflict, changing published hostname from 'homeassistant1724762' to 'homeassistant1724764'.`

This is just a small part of a mile long log. If I force LAN tag to the port it does not do this.

Home Assistant itself finds the second VLAN its using: See photo.
[image]

The system info is added below:

## System Health

version | core-2022.5.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/Oslo

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4610
Installed Version | 1.24.5
Stage | running
Available Repositories | 1104
Downloaded Repositories | 61

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 23 May 2022 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.0
docker_version | 20.10.9
disk_total | 234.0 GB
disk_used | 42.3 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.6.1), Mosquitto broker (6.1.1), AdGuard Home (4.5.1), ESPHome (2022.3.1), File editor (5.3.3), Home Assistant Google Drive Backup (0.107.1), SSH & Web Terminal (10.1.3), Z-Wave JS (0.1.58), Z-Wave JS to MQTT (0.39.0), Frigate NVR Proxy (1.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 34
views | 45
mode | storage

</details>

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Setup multiple VLAN
2. Let HA access multiple VLAN tags ("any tag")
3. HA will spam the DNS with new listings
...

### Anything in the Supervisor logs that might be useful for us?

```txt
See in original
```

### Anything in the Host logs that might be useful for us?

```txt
See in original
```

### System Health information

## System Health

version | core-2022.5.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/Oslo

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4610
Installed Version | 1.24.5
Stage | running
Available Repositories | 1104
Downloaded Repositories | 61

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 23 May 2022 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.0
docker_version | 20.10.9
disk_total | 234.0 GB
disk_used | 42.3 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.6.1), Mosquitto broker (6.1.1), AdGuard Home (4.5.1), ESPHome (2022.3.1), File editor (5.3.3), Home Assistant Google Drive Backup (0.107.1), SSH & Web Terminal (10.1.3), Z-Wave JS (0.1.58), Z-Wave JS to MQTT (0.39.0), Frigate NVR Proxy (1.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 34
views | 45
mode | storage

</details>

agners, 2022-05-06:
Are you using a mDNS reflector or similar? Then it might be related to #1397.

closed 2022-08-11 by github-actions[bot]

titiviking, 2024-01-24:
This issue still exists using HA 2024.1.5 on HAOS.

finchlt, 2024-02-16:
I'm having this issue too

titiviking, 2024-02-16:
Side question :
Do you see both interfaces (ip's) in the Unifi interface?
I only see the main interface, but the vlan is absent in the unifi clients... Very annoying as it's unable to provide statistics on the traffic for the different vlan interfaces...

finchlt, 2024-02-16:
I see the interfaces for both VLANs in the UniFi interface. M rational for dual home in two VLANs was to make service discovery easier on some annoying devices on my IOT network. Though this log issue is making me rethink. Filling up my storage and already made it hard to debug something else I was working on.

titiviking, 2024-02-16:
I killed the loop by adjusting the listed networks for mdns... May not be adequate for your use case from what I understand...
Or you disable it and occasionally enable when needed to discover devices?

pcmike, 2024-05-02:
I'm still having this issue on latest HAOS VM (on proxmox) on a unifi network too. The three bridges are each tagged with a specific vlan and the issue persists..

sle118, 2024-11-19:
Chiming in as well. It was probably happening for a while without me noticing, but this has caused me some troubles recently. Disabling mDNS reflector doesn't seem like a viable option as connectivity to all esphome devices (amongst other things) will require a lot of manual settings.

Hoping some day this will get the attention it deserves.

Archomeda, 2025-01-28:
I recently noticed this log spam as well. Running under Proxmox with 3 network interfaces, each with their own VLAN.

usvi, 2025-02-22:
I do not use home assistant but I have a mDNS storm happening. I have a small embedded box having one physical interface I aliase with macvlan to 2 separate interfaces, 'wan' and 'lmp'. wan is the regular interface providing actual connectivity and lmp is for local management with link-local addresses. I have configured stuff so that mDNS is disabled on wan and enabled on lmp.

Now what seems to be happening that after 16 to 30 days the system goes automatically haywire and starts a mDNS storm. To my shocking surprise the mDNS flag on wan interface has been automatically flipped ON. I can emulate the storm by commanding resolvectl to enable mDNS on wan interface in addition to the lmp interface and then fiddling the LAN cable.

I'm trying currently to debug this, but it is a pain because it needs to brew almost a month to occur. I was able to circumvent the issue by exclusively allowing mDNS 5353 only on one interface (lmp).