home-assistant/operating-system #1887: HAOS fails to start with NVMe connected

zdenekchobot, 2022-05-06:
### Describe the issue you are experiencing

HAOS stops booting when NVMe SSD is attached. 
Without an NVMe SSD, HA works perfectly
My configuration:
Rassperry CM4 4GB RAM / 8 GB eMMC on Waveshare IO base https://www.waveshare.com/CM4-IO-BASE-B.htm
HAOS 7.4 64 bit on eMMC
NVMEe Hynix M.2 NVMe 512GB with latest firmware
CM4 has latest bootloader

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install HAOS 7.4 on eMMC (without a NVMe SSD attached)
2. Start HA, upgrade all Core, Supervisor etc.
3. Attach NVMe SSD
4. HAOS stops starting ...
...

### Anything in the Supervisor logs that might be useful for us?

```txt
HAOS Debug message on Serial with insterted SSD (HDMI display does not show anything):

-Boot 2021.10 (Mar 28 2022 - 11:43:41 +0000)

DRAM: 3.9 GiB
RPI Compute Module 4 (0xc03140)
MMC: mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from nowhere… OK
In: serial
Out: serial
Err: serial
Net: eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
PCI: Failed autoconfig bar 10
PCI: Failed autoconfig bar 14
PCI: Failed autoconfig bar 18
PCI: Failed autoconfig bar 1c
PCI: Failed autoconfig bar 20
PCI: Failed autoconfig bar 24
starting USB…
No working controllers found
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1…
Found U-Boot script /boot.scr
2411 bytes read in 13 ms (180.7 KiB/s)

Executing script at 02400000
switch to partitions #0, OK
mmc0(part 0) is current device
loading env…

MMC read: dev # 0, block # 1214464, count 32 … 32 blocks read: OK
Trying to boot slot A, 2 attempts remaining. Loading kernel …
23079424 bytes read in 1328 ms (16.6 MiB/s)
storing env…

MMC write: dev # 0, block # 1214464, count 32 … 32 blocks written: OK
Starting kernel
Moving Image from 0x80000 to 0x200000, end=1950000

Flattened Device Tree blob at 2eff2c00
Booting using the fdt blob at 0x2eff2c00
Using Device Tree in place at 000000002eff2c00, end 000000002f002f38

Starting kernel …
*****************************************************************************
HAOS Debug message on Serial without SSD (HAOS starts fine):
U-Boot 2021.10 (Mar 28 2022 - 11:43:41 +0000)

DRAM: 3.9 GiB
RPI Compute Module 4 (0xc03140)
MMC: mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from nowhere… OK
In: serial
Out: serial
Err: serial
Net: eth0: ethernet@7d580000
PCIe BRCM: link down
starting USB…
No working controllers found
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1…
Found U-Boot script /boot.scr
2411 bytes read in 13 ms (180.7 KiB/s)

Executing script at 02400000
switch to partitions #0, OK
mmc0(part 0) is current device
loading env…

MMC read: dev # 0, block # 1214464, count 32 … 32 blocks read: OK
Trying to boot slot A, 1 attempts remaining. Loading kernel …
23079424 bytes read in 1327 ms (16.6 MiB/s)
storing env…

MMC write: dev # 0, block # 1214464, count 32 … 32 blocks written: OK
Starting kernel
Moving Image from 0x80000 to 0x200000, end=1950000

Flattened Device Tree blob at 2eff2c00
Booting using the fdt blob at 0x2eff2c00
Using Device Tree in place at 000000002eff2c00, end 000000002f002f38

Starting kernel .
```

### Anything in the Host logs that might be useful for us?

```txt
No, the system does not start and does not log
```

### Additional information

Raspberry Pi OS 64-bit RPI starts and works with the NVMe SSD properly

agners, 2022-05-06:
Try using [8.0.rc4](https://github.com/home-assistant/operating-system/releases/tag/8.0.rc4) from the GitHub release page. It includes changes to U-Boot which should make HAOS boot with NVMe.

zdenekchobot, 2022-05-06:
thank you for the quick suggestion.
With 8.0.rc4, the debug log is changed. But HAOS still doesn't start and display stays blank.

U-Boot 2022.01 (May 06 2022 - 13:09:13 +0000)

DRAM:  3.9 GiB
RPI Compute Module 4 (0xc03140)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
No working controllers found
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2411 bytes read in 12 ms (195.3 KiB/s)
## Executing script at 02400000
switch to partitions #0, OK
mmc0(part 0) is current device
loading env...

MMC read: dev # 0, block # 1214464, count 32 ... 32 blocks read: OK
Trying to boot slot A, 2 attempts remaining. Loading kernel ...
23618048 bytes read in 1366 ms (16.5 MiB/s)
storing env...

MMC write: dev # 0, block # 1214464, count 32 ... 32 blocks written: OK
Starting kernel
Moving Image from 0x80000 to 0x200000, end=1980000
## Flattened Device Tree blob at 2eff2c00
   Booting using the fdt blob at 0x2eff2c00
   Using Device Tree in place at 000000002eff2c00, end 000000002f002f62

Starting kernel

agners, 2022-05-07:
Hm, it seems the kernel crashes/isn't able to boot. I wonder if that board needs a specific kernel version or something? Can you boot Raspberry Pi OS on it without any changes to config.txt/or special kernel?

Can you try setting the device tree in config.txt (on the first FAT partition) to CM4 specifically:

```
device_tree=bcm2711-rpi-cm4.dtb
```

zdenekchobot, 2022-05-07:
No change with device_tree=bcm2711-rpi-cm4.dtb in config.txt
Yes, Raspberry Pi OS Lite 64-bit boots and uses NVMe SSD without any modification. Just "flash and run".

agners, 2022-05-08:
> Yes, Raspberry Pi OS Lite 64-bit boots and uses NVMe SSD without any modification. Just "flash and run".

That is somewhat surprising, as we use the very same kernel and even almost the same configuration. I why RPi OS works and HAOS not.

Can you edit `cmdline.txt` and add `earlycon` to the list?

Is there anything on the NVMe or is it empty?

Do you maybe have a different M.2 NVMe model to try if its realted to the particular NVMe model?

agners, 2022-05-09:
Uh, the kernel indeed crashes very early. I haven't seen that in my testing. I wonder what is different in your hardware which makes that appear.

Do you happen to have a CM4 IO Board along with [PCI Express x1 zu M.2 M-Key](https://www.delock.de/produkt/64105/merkmale.html?g=1140) adapter?

I'll do some more poking to see if I can reproduce it here.

zdenekchobot, 2022-05-10:
Unfortunately I only have the board [https://www.waveshare.com/CM4-IO-BASE-B.htm](url) with integrated M.2 KEY.

bschatzow, 2022-06-09:
@zdenekchobot I opened a similar issue with my RPI and SSD attached. My system also does not restart. If I unplug the SSD, restart the Pi, turn off the pi, plug back in the SSD everything works.

zdenekchobot, 2022-06-09:
Thank you @bschatzow  I let HA fully initialize without SSD. After connecting the SSD and restarting, the kernel still crashes

agners, 2022-06-10:
> @zdenekchobot I opened a similar issue with my RPI and SSD attached. My system also does not restart. If I unplug the SSD, restart the Pi, turn off the pi, plug back in the SSD everything works.

This is unrelated to your problem @bschatzow: This is a different environment (CM4 + Waveshare IO base), and the SSD is connected completely different (USB 3 vs. PCIe directly).

JumpMaster, 2022-06-11:
Confirming this has fixed the issue for me.  CM4 + Waveshare IO base + NVMe SSD.  Failed with 8.1 but works flawlessly with 8.2.

zdenekchobot, 2022-06-12:
@agners I understand. Any u-boot/kernel update plan/roadmap? In the meantime, thank you for your support.

zdenekchobot, 2022-06-12:
@JumpMaster it sounds great! What kind of Waveshare IO base and NVMe SSD do you use exactly? Thanks for sharing your success.

JumpMaster, 2022-06-12:
@zdenekchobot 
I bought the cheapest Waveshare IO Base.  The cm4‑io‑base‑b.  I did buy a decent NVME drive though even though there's no need for a lot of space.  I plan to buy Home Assistant yellow so will be moving the NVME drive over.

There is an issue with the waveshare IO base in that the NVME drive doesn't fit as it's too long.  But it will do until HA yellow is released.

Samsung 980 250 GB PCIe 3.0 (up to 2,900 MB/s) NVMe M.2 Internal Solid State Drive (SSD) (MZ-V8V250BW)

zdenekchobot, 2022-06-12:
Same board.
No success with the Samsung EVO 870 512GB or the NVMEe Hynix M.2 NVMe 512GB:-(

zdenekchobot, 2022-07-11:
@p6sfrx725z28knfy @agners 
No change with HAOS 8.3. I solved the issue by installing Home Assistant Supervised on Raspberry Pi OS Lite 64-bit. It works fine. I manually moved the datadisk and the Docker files to NVMe (the 'ha os datadisk move' command does not work in the supervised installation).
By the way: I tested the Bullseye from https://raspi.debian.net/tested-images/ (which is recommended for Supervised), but the one with NVMEe crashed completely like HAOS.

p6sfrx725z28knfy, 2022-07-12:
> @p6sfrx725z28knfy @agners No change with HAOS 8.3. I solved the issue by installing Home Assistant Supervised on Raspberry Pi OS Lite 64-bit. It works fine. I manually moved the datadisk and the Docker files to NVMe (the 'ha os datadisk move' command does not work in the supervised installation). BTW: I tested the Bullseye from https://raspi.debian.net/tested-images/ (which is recommended for Supervised), but the one with NVMEe crashed completely like HAOS.

Ah interesting, hopefully the issues with HAOS can be resolved as the supervised installation mode may not be for everyone..

Do I understand from the fact this bug has _not_ been resolved in HAOS, and if so should this issue be closed? :)

zdenekchobot, 2022-07-12:
> Odesláno: úterý 12. července 2022 13:10
> Komu: home-assistant/operating-system ***@***.***>

Yes, you're right, it's a solution only for me (and experienced users :-)). I closed the issue because I have a solution and @agners did not plan to actively solve this problem (see previous communication)

agners, 2022-07-13:
> Although I am curious why the changes @agners made in 8.2 fixed the issue observed by @JumpMaster, while still occurs on both my CM4-IO-BASE-A and CM4-IO-CASE-B with either 128gb or 256gb NVME :(

My change added NVMe support to U-Boot for the Raspberry Pi 4 images as well. It seems that the with those drivers in place, U-Boot works properly for @JumpMaster. Not sure, maybe @JumpMaster can comment on what exact version of base board etc. he uses exactly.

Is the error for all of you the same as the OP (`PCI: Failed autoconfig bar` in serial console)? Also on OS 8.2/8.3?

p6sfrx725z28knfy, 2022-07-13:
> Is the error for all of you the same as the OP (`PCI: Failed autoconfig bar` in serial console)? Also on OS 8.2/8.3?

I am using 8.2 and with the NVME attached the console output stops at "Starting Kernel...", I will try adding earlycon shortly..

```
U-Boot 2022.01 (Jun 09 2022 - 01:50:26 +0000)

DRAM:  3.9 GiB
RPI Compute Module 4 (0xc03140)
MMC:   mmc@7e340000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
No working controllers found
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2411 bytes read in 13 ms (180.7 KiB/s)
 Executing script at 02400000
switch to partitions #0, OK
mmc0(part 0) is current device
loading env...

MMC read: dev # 0, block # 1214464, count 32 ... 32 blocks read: OK
Trying to boot slot A, 2 attempts remaining. Loading kernel ...
23618048 bytes read in 1367 ms (16.5 MiB/s)
storing env...

MMC write: dev # 0, block # 1214464, count 32 ... 32 blocks written: OK
Starting kernel
Moving Image from 0x80000 to 0x200000, end=1980000
  Flattened Device Tree blob at 2eff2c00
   Booting using the fdt blob at 0x2eff2c00
   Using Device Tree in place at 000000002eff2c00, end 000000002f002fd2

Starting kernel ...
```

p6sfrx725z28knfy, 2022-07-13:
Still nothing obvious, to me at least...

```

Read start4.elf bytes  2241824 hnd 0x1ec
Read fixup4.dat bytes     5352 hnd 0x55
Firmware: e5a963efa66a1974127860b42e913d2374139ff5 Mar 24 2022 13:19:26
0x00c03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+

MESS:00:00:03.806307:0: arasan: arasan_emmc_open
MESS:00:00:03.997524:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.001016:0: brfs: File read: 2274 bytes
MESS:00:00:04.121455:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.130966:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.140480:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.149987:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.159501:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.169010:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.178524:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.188031:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.197545:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.207054:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.211559:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.217667:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.222923:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:04.229139:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:04.257223:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:04.261860:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:04.269602:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:04.277051:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:04.772932:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:04.778579:0: *** Restart logging
MESS:00:00:04.781222:0: brfs: File read: 2274 bytes
MESS:00:00:04.849920:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.859954:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.869990:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.880020:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.895433:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.905499:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.915569:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.925630:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.935699:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.945764:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.950820:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.961464:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.971528:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.981598:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.991658:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.001727:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.011791:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.021860:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.031920:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.041990:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.052054:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.057111:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.062711:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.071471:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:05.076944:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:05.083396:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:05.094496:0: dtb_file 'bcm2711-rpi-cm4.dtb'
MESS:00:00:05.104746:0: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
MESS:00:00:05.108012:0: Loading 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xcc30
MESS:00:00:05.139243:0: brfs: File read: 52272 bytes
MESS:00:00:05.144225:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:05.282138:0: brfs: File read: 1783 bytes
MESS:00:00:05.284992:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.290817:0: brfs: File read: 2274 bytes
[... 38 lines cut ...]
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2411 bytes read in 13 ms (180.7 KiB/s)
## Executing script at 02400000
switch to partitions #0, OK
mmc0(part 0) is current device
loading env...

MMC read: dev # 0, block # 1214464, count 32 ... 32 blocks read: OK
Trying to boot slot A, 1 attempts remaining. Loading kernel ...
23618048 bytes read in 1367 ms (16.5 MiB/s)
storing env...

MMC write: dev # 0, block # 1214464, count 32 ... 32 blocks written: OK
Starting kernel
Moving Image from 0x80000 to 0x200000, end=1980000
## Flattened Device Tree blob at 2eff2c00
   Booting using the fdt blob at 0x2eff2c00
   Using Device Tree in place at 000000002eff2c00, end 000000002f002ff6

Starting kernel ...

```

zdenekchobot, 2022-07-14:
> Still nothing obvious, to me at least...
>
> ```
>
> Read start4.elf bytes  2241824 hnd 0x1ec
> Read fixup4.dat bytes     5352 hnd 0x55
> Firmware: e5a963efa66a1974127860b42e913d2374139ff5 Mar 24 2022 [...]

with the same setting earlycon and console=tty1 in cmdline.txt (and enable_uart=1 in config.txt of course) I also received messages about the start (and crash) of the kernel on the serial port. See attachments in my previous posts.

zdenekchobot, 2022-07-14:
> > Yes, you're right, it's a solution only for me (and experienced users :-)). I closed the issue because I have a solution and @agners did not plan to actively solve this problem (see previous communication)
> 
> Fair enough, although perhaps as you have experienced similar issues with RasPi Bullseye then it could be an upstream issue rather than with HAOS itself..
> 
> Although I am curious why the changes @agners made in 8.2 fixed the issue observed by @JumpMaster, while still occurs on both my CM4-IO-BASE-A and CM4-IO-CASE-B with either 128gb or 256gb NVME :(

I think Bullseye from https://raspi.debian.net/tested-images/ contains an older kernel version than Raspberry Pi OS Lite 64-bit and HAOS. That's probably why it doesn't work with NVMe.
Raspberry Pi OS Lite 64-bit works with CM4-IO-BASE-B and NVMEe Hynix M.2 NVMe 512GB on "first power on". Just flash and run:-)

agners, 2022-07-14:
@p6sfrx725z28knfy hm, it seems earlycon did not really work in your case. Can you add just `earlycon` (without `=uart8250...`)? I think that should be enough for Arm platforms.

zdenekchobot, 2022-07-14:
> @p6sfrx725z28knfy hm, it seems earlycon did not really work in your case. Can you add just `earlycon` (without `=uart8250...`)? And/or it seems you are using device tree overlays (like `disable-bt`), can you try without?

Attachments that worked for me
[config.txt](https://github.com/home-assistant/operating-system/files/9109677/config.txt)
[cmdline.txt](https://github.com/home-assistant/operating-system/files/9109678/cmdline.txt)

p6sfrx725z28knfy, 2022-07-14:
FYI I tried RaspiOS Lite which booted and detected the NVME drive no problems, I will try HAOS again later with the suggested earlycon syntax :)

```

$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

$ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

$ lsblk 
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0  14.6G  0 disk 
├─mmcblk0p1  179:1    0   256M  0 part /boot
└─mmcblk0p2  179:2    0  14.3G  0 part /
mmcblk0boot0 179:32   0     4M  1 disk 
mmcblk0boot1 179:64   0     4M  1 disk 
nvme0n1      259:0    0 238.5G  0 disk 
└─nvme0n1p1  259:1    0 238.5G  0 part 
```

p6sfrx725z28knfy, 2022-07-14:
> Attachments that worked for me [config.txt](https://github.com/home-assistant/operating-system/files/9109677/config.txt) [...]

Thanks @zdenekchobot, I used those config and cmdline files with HAOS 8.2 with the following output:

```

U-Boot 2022.01 (Jun 09 2022 - 01:50:26 +0000)

DRAM:  3.9 GiB
RPI Compute Module 4 (0xc03140)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
No working controllers found
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2411 bytes read in 13 ms (180.7 KiB/s)
## Executing script at 02400000
switch to partitions #0, OK
mmc0(part 0) is current device
loading env...

MMC read: dev # 0, block # 1214464, count 32 ... 32 blocks read: OK
## Error: bad CRC, import failed
Trying to boot slot A, 2 attempts remaining. Loading kernel ...
23618048 bytes read in 1362 ms (16.5 MiB/s)
storing env...

MMC write: dev # 0, block # 1214464, count 32 ... 32 blocks written: OK
Starting kernel
Moving Image from 0x80000 to 0x200000, end=1980000
## Flattened Device Tree blob at 2eff2c00
   Booting using the fdt blob at 0x2eff2c00
   Using Device Tree in place at 000000002eff2c00, end 000000002f002f86

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8 (builder@5b9b6c5b4698) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g799554151) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Thu Jun 9 01:50:59 2
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe215040 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] SError Interrupt on CPU0, code 0xbf000002 -- SError
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 5.15.32-v8 #1
[    0.000000] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[    0.000000] pstate: 600000c5 (nZCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    0.000000] pc : setup_arch+0x164/0x5a0
[    0.000000] lr : setup_arch+0x15c/0x5a0
[    0.000000] sp : ffffffc009493d90
[    0.000000] x29: ffffffc009493d90 x28: 00000000012b0018 x27: 000000003b364f68
[    0.000000] x26: 0000000000000000 x25: 000000003b3d5ab0 x24: 000000003aff1d98
[    0.000000] x23: ffffffc00949a180 x22: ffffffc0094a2bc0 x21: ffffffc00968c000
[    0.000000] x20: fffffffdfddf2cc8 x19: ffffffc008010000 x18: ffffffffffffffff
[    0.000000] x17: 0000000000001800 x16: 0000000000001400 x15: ffffffc0894938a7
[    0.000000] x14: 0000000000000000 x13: 64656c62616e6520 x12: 0000000000000000
[    0.000000] x11: ffffffc008cda9d0 x10: 0101010101010101 x9 : 0000000000000041
[    0.000000] x8 : 0101010101010101 x7 : 0000000001e34ef5 x6 : fefefefeff3fff73
[    0.000000] x5 : 0000000000000000 x4 : 0000000000000072 x3 : 0000000000000063
[    0.000000] x2 : 0000000000000072 x1 : 0000000000000000 x0 : 00000000000000c0
[    0.000000] Kernel panic - not syncing: Asynchronous SError Interrupt
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 5.15.32-v8 #1
[    0.000000] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[    0.000000] Call trace:
[    0.000000]  dump_backtrace+0x0/0x1b0
[    0.000000]  show_stack+0x20/0x30
[    0.000000]  dump_stack_lvl+0x8c/0xb8
[    0.000000]  dump_stack+0x18/0x34
[    0.000000]  panic+0x178/0x370
[    0.000000]  nmi_panic+0x94/0xa0
[    0.000000]  arm64_serror_panic+0x6c/0x80
[    0.000000]  do_serror+0x3c/0x90
[    0.000000]  el1h_64_error_handler+0x38/0x50
[    0.000000]  el1h_64_error+0x78/0x7c
[    0.000000]  setup_arch+0x164/0x5a0
[    0.000000]  start_kernel+0x9c/0x6e8
[    0.000000]  __primary_switched+0xa0/0xa8

```

maxromanovsky, 2022-08-21:
Hi!
I am experiencing the same issue exactly with [CM4-IO-BASE-B](https://www.waveshare.com/wiki/CM4-IO-BASE-B).
HAOS v8.5 boots w/o NVMe attached, however I don't see any way to modify the `config.txt` and other params:
- USB ports on the board are disabled by default, so I can neither enable SSH on port `22222`, nor connect a physical keyboard
- I have macOS, and it doesn't see `boot` partition after flashing HAOS

Any ideas on how to connect to the board?
- Will UART via GPIO help me here?
- Will replacing macOS with Linux machine help me if CM4 is connected to it it USB-C connection mode (i.e. for flashing eMMC)?

TimWeyand, 2022-10-17:
Hi,
has anyone found a solution, to run HAOS with an NVMe on a waveshare CM4-IO-BASE-A or CM4-IO-BASE-B?

HAOS (9.2) runs on eMMC without NVMe. As soon as i install the NVMe, it’s not booting anymore. It does not matter which Bootpriority and if installed on eMMC or NVMe.

(raspbian works just fine, both scenarios)

maxromanovsky, 2022-10-18:
@TimWeyand here's what helped me: https://github.com/home-assistant/operating-system/issues/2075#issuecomment-1245763776

TimWeyand, 2022-10-18:
@maxromanovsky Thanks, which haos version are you running? 

I tried it with the latest (yesterday), but there seems to be a new nvme problem

maxromanovsky, 2022-10-18:
@TimWeyand 
```
Home Assistant 2022.10.4
Supervisor 2022.10.0
Operating System 9.2
```

I've installed v9.0, and then upgraded it as usual via HA app

JumpMaster, 2022-10-18:
Hi all,

I have a CM4-IO-BASE-B with a CM4 (4GB RAM, 16GB MMC) set to boot from NVME first.  The NVME is a Samsung 980 250GB (MZ-VBV250).  It's not in use at the moment so I can install any OS and try any upgrade method.

Let me know if anything needs testing!

zdenekchobot, 2022-10-18:
Great offer, thank you @JumpMaster. I would be very interested in the test result (clean installation HAOS on NVMe). Today I have to use a supervised HA installation instead of HAOS.

JumpMaster, 2022-10-18:
So I've approached this as if it was a new pi/nvme/carrier board.

I used Jeff Geerlings guide on updating the cm4 bootloader and changing the boot order.
https://www.jeffgeerling.com/blog/2022/how-update-raspberry-pi-compute-module-4-bootloader-eeprom

in boot.conf I set the boot order to:
BOOT_ORDER=0xf25416

I replaced pieeprom.original.bin in the recovery folder with the latest pieeprom.bin
https://github.com/raspberrypi/rpi-eeprom/releases/download/v2022.04.26-138a1/rpi-boot-eeprom-recovery-2022-04-26-vl805-000138a1.zip

Then ran ./update-pieeprom.sh to create the new pieeprom.bin

To flash the new eeprom I used "rpiboot -d recovery" with the carrier board connected via USB C and the switch set to 1.

To install the OS I put the nvme drive into a USB carrier and used RPI Imager to install the latest 9.2 x64 OS
https://github.com/home-assistant/operating-system/releases/download/9.2/haos_rpi4-64-9.2.img.xz

Set the carrier board switch back to 0.  Install the NVME drive and it booted hasos first time.

zdenekchobot, 2022-10-21:
> So I've approached this as if it was a new pi/nvme/carrier board.
>
> I used Jeff Geerlings guide on updating the cm4 bootloader and changing the boot order. [...]

> So I've approached this as if it was a new pi/nvme/carrier board.
>
> I used Jeff Geerlings guide on updating the cm4 bootloader and changing the boot order. [...]

Thanks for sharing. My CM4 and Waveshare IO-BASE won't boot from my NVMe SK Hynix :-( No HDMI & UART output when preferring NVMe as boot device. Bootloader is latest version, NVMe has correct content and other boot methods work fine. I think my NVMe SK Hynix is not 100% compatible.
I'm going back to the supervised installation. With HAOS, I might still have to deal with other problems with the peripherals on the IO-BASE (especially the fan control) and the Waveshare UPS HAT.
Thanks again everyone for your help.

agners, 2022-10-24:
> My CM4 and Waveshare IO-BASE won't boot from my NVMe SK Hynix :-( 

@zdenekchobot was that the same NVMe you've posted logs from at https://github.com/home-assistant/operating-system/issues/1887#issuecomment-1121404519?

I've received reports of problems with SK Hynix NVMe's on Yellow as well (see #2178).

p6sfrx725z28knfy, 2022-10-31:
FYI NVME in the Waveshare board seem seems to be working under HAOS 9.3:

> nvme0n1
/dev/disk/by-id/nvme-BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
Subsystem:
block
Device path:
/dev/nvme0n1
ID:
/dev/disk/by-id/nvme-BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
Attributes:
DEVLINKS: >-
  /dev/disk/by-id/nvme-BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
  /dev/disk/by-id/nvme-nvme.1c5c-4e4a38424e303735333132323042414346-4243353031204e564d6520534b2068796e6978203132384742-00000001
  /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
DEVNAME: /dev/nvme0n1
DEVPATH: >-
  /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/nvme/nvme0/nvme0n1
DEVTYPE: disk
DISKSEQ: '28'
ID_MODEL: BC501 NVMe SK hynix 128GB
ID_PART_TABLE_TYPE: gpt
ID_PART_TABLE_UUID: 757f2a9d-af6c-4131-a86b-725bec926732
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-nvme-1
ID_REVISION: 80001C00
ID_SERIAL: BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
ID_SERIAL_SHORT: xxxxxxxxxxx
ID_WWN: >-
  nvme.1c5c-4e4a38424e303735333132323042414346-4243353031204e564d6520534b2068796e6978203132384742-00000001
MAJOR: '259'
MINOR: '0'
SUBSYSTEM: block
TAGS: ':systemd:'
USEC_INITIALIZED: '3142039'

And I was able to use the UI to move the data disk from the eMMC:

> $ cat /proc/version 
Linux version 5.15.61-v8 (builder@102ccd9fd38e) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g48c0c35f09) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Fri Oct 21 00:45:05 UTC 2022

> $ lsblk /dev/nvme0n1
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
nvme0n1     259:0    0 119.2G  0 disk 
└─nvme0n1p1 259:1    0 119.2G  0 part /var/log/journal
                                      /etc/pulse/client.conf
                                      /etc/asound.conf
                                      /etc/hostname
                                      /etc/hosts
                                      /run/audio
                                      /etc/resolv.conf
                                      /addons
                                      /ssl
                                      /config
                                      /share
                                      /data
                                      /backup
                                      /media

> $ df -h
Filesystem                Size      Used Available Use% Mounted on
overlay                 117.3G      7.0G    105.5G   6% /
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /media
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /backup
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /data
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /share
devtmpfs                  1.7G         0      1.7G   0% /dev
tmpfs                     1.9G         0      1.9G   0% /dev/shm
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /config
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /ssl
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /addons
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/resolv.conf
tmpfs                   758.6M      1.4M    757.2M   0% /run/dbus
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /run/audio
tmpfs                   758.6M      1.4M    757.2M   0% /run/docker.sock
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/hosts
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/hostname
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/asound.conf
tmpfs                     1.9G         0      1.9G   0% /dev/shm
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/pulse/client.conf
tmpfs                   758.6M      1.4M    757.2M   0% /run/log/journal
/dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /var/log/journal
tmpfs                     1.9G         0      1.9G   0% /proc/asound
devtmpfs                  1.7G         0      1.7G   0% /proc/keys
devtmpfs                  1.7G         0      1.7G   0% /proc/latency_stats
devtmpfs                  1.7G         0      1.7G   0% /proc/timer_list
tmpfs                     1.9G         0      1.9G   0% /sys/firmware

zdenekchobot, 2022-10-31:
> > My CM4 and Waveshare IO-BASE won't boot from my NVMe SK Hynix :-(
> 
> @zdenekchobot was that the same NVMe you've posted logs from at [#1887 (comment)](https://github.com/home-assistant/operating-system/issues/1887#issuecomment-1121404519)?
> 
> I've received reports of problems with SK Hynix NVMe's on Yellow as well (see #2178).

Yes, it was the same one. SK_Hynix BC711 512GB

zdenekchobot, 2022-10-31:
> FYI NVME in the Waveshare board seem seems to at least be detected under HAOS 9.3:
> 
> > nvme0n1
> > /dev/disk/by-id/nvme-BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
> > Subsystem:
> > block
> > Device path:
> > /dev/nvme0n1
> > ID:
> > /dev/disk/by-id/nvme-BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
> > Attributes:
> > DEVLINKS: >-
> > /dev/disk/by-id/nvme-BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
> > /dev/disk/by-id/nvme-nvme.1c5c-4e4a38424e303735333132323042414346-4243353031204e564d6520534b2068796e6978203132384742-00000001
> > /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
> > DEVNAME: /dev/nvme0n1
> > DEVPATH: >-
> > /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/nvme/nvme0/nvme0n1
> > DEVTYPE: disk
> > DISKSEQ: '28'
> > ID_MODEL: BC501 NVMe SK hynix 128GB
> > ID_PART_TABLE_TYPE: gpt
> > ID_PART_TABLE_UUID: 757f2a9d-af6c-4131-a86b-725bec926732
> > ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
> > ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-nvme-1
> > ID_REVISION: 80001C00
> > ID_SERIAL: BC501_NVMe_SK_hynix_128GB_NJ8BN07531220BACF
> > ID_SERIAL_SHORT: xxxxxxxxxxx
> > ID_WWN: >-
> > nvme.1c5c-4e4a38424e303735333132323042414346-4243353031204e564d6520534b2068796e6978203132384742-00000001
> > MAJOR: '259'
> > MINOR: '0'
> > SUBSYSTEM: block
> > TAGS: ':systemd:'
> > USEC_INITIALIZED: '3142039'
> 
> And I was able to use the UI to move the data disk from the eMMC:
> 
> > $ cat /proc/version
> > Linux version 5.15.61-v8 (builder@102ccd9fd38e) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g48c0c35f09) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Fri Oct 21 00:45:05 UTC 2022
> 
> > $ lsblk /dev/nvme0n1
> > NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
> > nvme0n1     259:0    0 119.2G  0 disk
> > └─nvme0n1p1 259:1    0 119.2G  0 part /var/log/journal
> > /etc/pulse/client.conf
> > /etc/asound.conf
> > /etc/hostname
> > /etc/hosts
> > /run/audio
> > /etc/resolv.conf
> > /addons
> > /ssl
> > /config
> > /share
> > /data
> > /backup
> > /media
> 
> > $ df -h
> > Filesystem                Size      Used Available Use% Mounted on
> > overlay                 117.3G      7.0G    105.5G   6% /
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /media
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /backup
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /data
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /share
> > devtmpfs                  1.7G         0      1.7G   0% /dev
> > tmpfs                     1.9G         0      1.9G   0% /dev/shm
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /config
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /ssl
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /addons
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/resolv.conf
> > tmpfs                   758.6M      1.4M    757.2M   0% /run/dbus
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /run/audio
> > tmpfs                   758.6M      1.4M    757.2M   0% /run/docker.sock
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/hosts
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/hostname
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/asound.conf
> > tmpfs                     1.9G         0      1.9G   0% /dev/shm
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /etc/pulse/client.conf
> > tmpfs                   758.6M      1.4M    757.2M   0% /run/log/journal
> > /dev/nvme0n1p1          117.3G      7.0G    105.5G   6% /var/log/journal
> > tmpfs                     1.9G         0      1.9G   0% /proc/asound
> > devtmpfs                  1.7G         0      1.7G   0% /proc/keys
> > devtmpfs                  1.7G         0      1.7G   0% /proc/latency_stats
> > devtmpfs                  1.7G         0      1.7G   0% /proc/timer_list
> > tmpfs                     1.9G         0      1.9G   0% /sys/firmware

Thanks for sharing the experience and hope for the future :-) I will occasionally buy another NVMe and test everything again. I currently have a different model than yours (BC711) - maybe the reason for my trouble?
p.s. Supervised HA installation works great. 

P: /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/nvme/nvme0/nvme0n1
N: nvme0n1
L: 0
S: disk/by-id/nvme-SKHynix_HFM512GD3HX015N_FNAAN43291020A84G
S: disk/by-id/nvme-eui.ace42e00164fb04a2ee4ac0000000001
S: disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
E: DEVPATH=/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/nvme/nvme0/nvme0n1
E: DEVNAME=/dev/nvme0n1
E: DEVTYPE=disk
E: DISKSEQ=25
E: MAJOR=259
E: MINOR=0
E: SUBSYSTEM=block
E: USEC_INITIALIZED=4657817
E: ID_SERIAL_SHORT=FNAAN43291020A84G
E: ID_WWN=eui.ace42e00164fb04a2ee4ac0000000001
E: ID_MODEL=SKHynix_HFM512GD3HX015N
E: ID_REVISION=41040C20
E: ID_SERIAL=SKHynix_HFM512GD3HX015N_FNAAN43291020A84G
E: ID_PATH=platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
E: ID_PATH_TAG=platform-fd500000_pcie-pci-0000_01_00_0-nvme-1
E: ID_PART_TABLE_UUID=21c21642
E: ID_PART_TABLE_TYPE=dos
E: DEVLINKS=/dev/disk/by-id/nvme-SKHynix_HFM512GD3HX015N_FNAAN43291020A84G /dev/disk/by-id/nvme-eui.ace42e00164fb04a2ee4ac0000000001 /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-nvme-1
E: TAGS=:systemd:
E: CURRENT_TAGS=:systemd:

maxromanovsky, 2023-01-30:
The issue still persists

rluengen, 2023-02-01:
Confirmed with Home Assistant OS 9.5 that this issue still exists.

floion, 2023-03-31:
For anybody still interested in this issue, I have created this issue in upstream rpi firmware repo: https://github.com/raspberrypi/firmware/issues/1800

maxromanovsky, 2023-03-31:
@floion just for my understanding. How is that related to RPI firmware? Vanilla Raspberry PI OS x64 in my case booted up correctly from eMMC and could use NVMe drive on Waveshare board.
All this time I thought it's an issue with how HAOS was built and configured.

maxromanovsky, 2023-04-14:
@floion In this case, how does home assistant yellow boot with both eMMC and NVMe?

agners, 2023-04-14:
> @floion In this case, how does home assistant yellow boot with both eMMC and NVMe?

Afaik, this issue was always a combination of U-Boot + Waveshare board: On Yellow, things seem to work :man_shrugging: 

> For anybody still interested in this issue, I have created this issue in upstream rpi firmware repo: [raspberrypi/firmware#1800](https://github.com/raspberrypi/firmware/issues/1800)

@floion hm, from this issue it seems that boot order configured is very much relevant. We actually don't adjust boot order on Yellow, so we often boot from eMMC and let the Linux kernel initialize PCIe/NVMe. I guess that works on Waveshare as well?

maxromanovsky, 2023-04-18:
@agners 

Is it just me, or does anyone else have issues after upgrade to HAOS 10.0?

I'm using RPI CM4 with eMMC on Waveshare board with NVMe.
eMMC is unused due to known issues with NVMe, however after upgrade OS doesn't boot at all.
If connected, screen is blank, and as I understand it all ends on U-Boot.
Am I getting it right?

1. PI EEPROM is configured to boot from NVMe, and successfully does that by passing control to U-Boot (which was updated as a part of HAOS - #2234)
2. U-Boot fails to detect the NVMe for some reason (`Device 0: unknown device`?) and tries to boot via network (PXE)
3. HAOS is not even booted

```
...................................................................

U-Boot 2023.01 (Apr 17 2023 - 11:52:08 +0000)

DRAM:  512 MiB (effective 3.4 GiB)
RPI Compute Module 4 (0xc03141)
Core:  208 devices, 16 uclasses, devicetree: board
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
No working controllers found
switch to partitions #0, OK
mmc0(part 0) is current device
** No partition table - mmc 0 **
Couldn't find partition mmc 0:1
Card did not respond to voltage select! : -110
MMC Device 2 not found
no mmc device at slot 2

Device 0: unknown device
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
starting USB...
No working controllers found
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: pxeuuid
Retrieving file: pxelinux.cfg/01-e4-5f-01-b8-d2-51
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/00000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/0000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/00000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/0000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/00
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/0
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/default-arm-bcm283x-rpi
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/default-arm-bcm283x
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/default-arm
ethernet@7d580000 Waiting for PHY auto negotiation to complete....
```

floion, 2023-04-19:
> Afaik, this issue was always a combination of U-Boot + Waveshare board: On Yellow, things seem to work man_shrugging
>
>
> @floion hm, from this issue it seems that boot order configured is very much [...]

@agners in our case, the linux kernel would crash when initializing the PCIe when using the boot order with eMMC set before NVMe. From this report I see logs similar for you too.

I am not familiar with what the Yellow you are referring to is.

maxromanovsky, 2023-04-19:
@floion [Yellow](https://www.home-assistant.io/yellow/) is a custom CM4 board.

floion, 2023-04-19:
@maxromanovsky I see.
Well it should suffer from the same issue if it uses u-boot. Maybe the CM4s Yellow uses have a different boot order set?

maxromanovsky, 2023-04-22:
I was able to boot from internal eMMC and use M.2 eMMC (not NVMe) as a data disk - see magic instructions in this comment: https://github.com/home-assistant/operating-system/issues/2479#issuecomment-1518544887

@floion this is question to @agners. I don't have Yellow.
> Well it should suffer from the same issue if it uses u-boot. Maybe the CM4s Yellow uses have a different boot order set?

closed 2023-07-28 by github-actions[bot]

xyklex, 2023-08-03:
I'm facing the same issue but in my case, I'm trying to boot the OS directly from the NVMe disk, similar to [this](https://github.com/home-assistant/operating-system/issues/2178)

I have tried it on two boards:
- Waveshare IO base B
- Raspberry Pi CM4 Official IO Board

I'm using u-boot version 2023.07.

I get the next logs

<details><summary>rpi boot logs</summary>

```
  Device: mmc@7e340000
  Manufacturer ID: 15
  OEM: 0
  Name: BJTD4R
  Bus Speed: 52000000
  Mode: MMC High Speed (52MHz)
  Rd Block Len: 512
  MMC version 5.1
  High Capacity: Yes
  Capacity: 29.1 GiB
  Bus Width: 4-bit
  Erase Group Size: 512 KiB
  HC WP Group Size: 8 MiB
  User Capacity: 29.1 GiB WRREL
  Boot Capacity: 4 MiB ENH
  RPMB Capacity: 4 MiB ENH
  Boot area 0 is not write protected
  Boot area 1 is not write protected
  U-Boot> Read start4.elf bytes  2250656 hnd 0xae17
  Read fixup4.dat bytes     5400 hnd 0x11059
  0x00c03141 0x00000000 0x00001fff
  MEM GPU: 76 ARM: 948 TOTAL: 1024
  Starting start4.elf @ 0xfec00200 partition 0
  NVME off
  +

  MESS:00:00:03.216911:0: VID 0x1c5c MN SKHynix_HFM512GD3HX015N
  MESS:00:00:03.221055:0: NVME on
  MESS:00:00:03.496984:0: brfs: File read: /mfs/sd/config.txt
  MESS:00:00:03.500184:0: brfs: File read: 1261 bytes
  MESS:00:00:03.527745:0: HDMI0:EDID error reading EDID block 0 attempt 0
  MESS:00:00:03.532248:0: HDMI0:EDID giving up on reading EDID block 0
  MESS:00:00:03.549158:0: HDMI1:EDID error reading EDID block 0 attempt 0
  MESS:00:00:03.553665:0: HDMI1:EDID giving up on reading EDID block 0
  MESS:00:00:03.565309:0: brfs: File read: /mfs/sd/config.txt
  MESS:00:00:03.568472:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
  MESS:00:00:03.574975:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
  MESS:00:00:03.746340:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
  MESS:00:00:03.750980:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
  MESS:00:00:03.758722:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
  MESS:00:00:03.766171:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
  MESS:00:00:04.071913:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
  MESS:00:00:04.077561:0: *** Restart logging
  MESS:00:00:04.080202:0: brfs: File read: 1261 bytes
  MESS:00:00:04.098203:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
  MESS:00:00:04.103229:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
  MESS:00:00:04.113852:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
  MESS:00:00:04.118881:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
  MESS:00:00:04.124480:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
  MESS:00:00:04.138269:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
  MESS:00:00:04.143299:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
  MESS:00:00:04.153921:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
  MESS:00:00:04.158945:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
  MESS:00:00:04.164543:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
  MESS:00:00:04.173307:0: HDMI0: hdmi_pixel_encoding: 300000000
  MESS:00:00:04.178780:0: HDMI1: hdmi_pixel_encoding: 300000000
  MESS:00:00:04.184577:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
[... 54 lines cut ...]
  ** No partition table - mmc 0 **
  Couldn't find partition mmc 0:1
  Card did not respond to voltage select! : -110
  MMC Device 2 not found
  no mmc device at slot 2
  starting USB...
  Bus usb@7e980000: dwc2_usb usb@7e980000: Can't get reset: -2
  dwc2_usb usb@7e980000: Core Release: 2.80a
  USB DWC2
  scanning bus usb@7e980000 for devices... 2 USB Device(s) found
        scanning usb for storage devices... 0 Storage Device(s) found

  Device 0: unknown device
  PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
  U-Boot> pci
  BusDevFun  VendorId   DeviceId   Device Class       Sub-Class
  _____________________________________________________________
  00.00.00   0x14e4     0x2711     Bridge device           0x04
  01.00.00   0x1c5c     0x174a     Mass storage controller 0x08
  U-Boot> nvme scan
  U-Boot> nvme info
```
</details>

As you can see from the u-boot `pci` output, the device is recognized but when trying `nvme scan` it doesn't output any error but no devices were found. The NVMe has two partitions (1st FAT, 2nd Ext4) I can boot directly to it using `kernel=vmlinuz` and it will boot ubuntu 22.04 normally, so something is wrong with u-boot, but don't know what exactly (nvme driver, or what else?). I've also tried changing the `BOOT_ORDER` as reported [in this](https://github.com/raspberrypi/firmware/issues/1800) issue, but no changes.

agners, 2023-08-03:
@xyklex hm seems like the U-Boot NVMe implementation isn't able to detect that particular NVMe, for some reason. Can you open a new issue along with information what NVMe you are using?

xyklex, 2023-08-03:
@agners thanks for you answer, Should I create the issue here or on u-boot?

Rogash1, 2025-01-19:
Has this issue ever been solved?
I just reinstalled HASSOS on my CM4 8GB on a Wavshare CM4-IO-BASE-A since it was a really old setup from mid-2023 and the upgrade to the newest version failed. The old setup was successful running from an Transcend TS1TMTE400S NVME back then. The Hassos version 10.0 I think. Whenever I now plug the SSD into the Wavshare IO-Base I get the same "Kernel panic - not syncing: Asynchronous SError Interrupt" error. The bootloader of my CM4 is on the latest version as well.