home-assistant/operating-system #1889: Can't update from 6.1 to 7.6 - Can't find ext4 filesystem

doron1, 2022-05-06:
### Describe the issue you are experiencing

Trying to update from HA OS 6.1 to 7.6. Fails repeatedly (from GUI and from CLI).
On console:
`EXT4-fs (sda2): VFS: Can't find ext4 filesystem`

In supervisor log:
`Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: failed to mount slot: failed to run mount: Child process exited with code 32`

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

6.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.os update
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: failed to mount slot: failed to run mount: Child process exited with code 32
```

### Anything in the Host logs that might be useful for us?

```txt
EXT4-fs (sda2): VFS: Can't find ext4 filesystem
```

agners, 2022-05-07:
Can you try updating to the latest version 6.6 before upgrading? You can do that from the Home Assistant terminal using:

```
ha os update --version=6.6
```

doron1, 2022-05-07:
Yes, it did the trick - thanks!
Still worth addressing IMO, as most users will try the update from the GUI and keep failing.

agners, 2022-05-10:
So far upgrade across major release releases did mostly work, in fact I think that is the first restriction I am aware of.

The cause are two changes in the update system:

- In 6.2 HAOS changed the kernel partition from being marked as `ext4` to `raw`
- In 7.6 HAOS added a rauc `post-install` hook

It seems that the post install hook causes rauc from trying to mount the partition before calling the hook. Even though we don't need the hook for the kernel partition, it causes the error when upgrading from 6.1 and older to 7.6 and newer.

Probably the best way is to enforce major OS upgrades only. It was in the back of my mind a while, but this makes it more important. This needs to be implemented in Supervisor though. I created a RFC here: https://github.com/home-assistant/supervisor/issues/3629.

closed 2022-05-10 by agners