home-assistant/operating-system #1900: Bluetooth not working on virtualbox vm

bobvmierlo, 2022-05-12:
### Describe the issue you are experiencing

Since installing OS8.0 the internal bluetooth chip of my Intel NUC is not working anymore in Home Assistant.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Home Assistant OS can't seem to use the internal Bluetooth chip of my Intel NUC, since installing 8.0.
Tried disattaching the "USB" device from the Virtualbox VM and reattaching. This allways solved issues with the internal BT chip.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing related to BT issues
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.259527] Bluetooth: Core ver 2.22
[    0.259534] NET: Registered PF_BLUETOOTH protocol family
[    0.259537] Bluetooth: HCI device and connection manager initialized
[    0.259539] Bluetooth: HCI socket layer initialized
[    0.259542] Bluetooth: L2CAP socket layer initialized
[    0.259545] Bluetooth: SCO socket layer initialized
[    4.002661] Bluetooth: hci0: Found device firmware: intel/ibt-17-16-1.sfi
[    4.002754] Bluetooth: hci0: Boot Address: 0x40800
[    4.002757] Bluetooth: hci0: Firmware Version: 60-44.21
[    6.035532] Bluetooth: hci0: command 0xfc01 tx timeout
[   14.355434] Bluetooth: hci0: FW download error recovery failed (-110)
[  306.696246] Bluetooth: hci0: Found device firmware: intel/ibt-17-16-1.sfi
[  306.696284] Bluetooth: hci0: Boot Address: 0x40800
[  306.696285] Bluetooth: hci0: Firmware Version: 60-44.21
[  308.754237] Bluetooth: hci0: command 0xfc01 tx timeout
[  316.946372] Bluetooth: hci0: FW download error recovery failed (-110)
```

### System Health information

## System Health

version | core-2022.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.15.38
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4247
Installed Version | 1.24.5
Stage | running
Available Repositories | 1103
Downloaded Repositories | 58

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 27

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 29 May 2022, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.0
-- | --
update_channel | beta
supervisor_version | supervisor-2022.05.2
docker_version | 20.10.14
disk_total | 73.1 GB
disk_used | 46.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.6.1), Mosquitto broker (6.1.2), TasmoAdmin (0.17.0), AirCast (3.3.0), File editor (5.3.3), deCONZ (6.13.0), SSH & Web Terminal (10.1.3), Vaultwarden (Bitwarden) (0.16.0), phpMyAdmin (0.7.1), WireGuard (0.6.0), Glances (0.15.0), Node-RED (11.1.2), Home Assistant Google Drive Backup (0.107.2), AdGuard Home (4.5.1), Check Home Assistant configuration (3.10.0), Let's Encrypt (4.12.2), Nginx Proxy Manager (0.11.0), chrony (2.3.0), ESPHome (2022.3.1), TimescaleDB (2.0.1), pgAdmin4 (2.0.0), MariaDB (2.4.0), Studio Code Server (5.0.2), Zigbee2mqtt (1.25.1-1), Frigate NVR (3.1), UniFi Network Application (2.2.0), DSMR Reader (1.2.0), Genie (3.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 38
views | 33
mode | storage

</details>

agners, 2022-05-12:
It seems that OS 7.6 provided the same firmware (`ibt-17-16-1.sfi`). So the only thing which changes is the upstream kernel (from LTS 5.10 to 5.15). It is a bit surprising to me that this would break Bluetooth, especially on Intel hardware, their drivers are usually very well tested.

FWIW, -110 means timeout... So probably some communication issue or similar.

Can you downgrade again to 7.6 to double check that things indeed work with older OS versions?

```
ha os update --version 7.6
```

bobvmierlo, 2022-05-14:
It seems the internal BT chip of my Intel NUC has died on the same day I updated my Home Assistant OS VM. So not an OS issue.

closed 2022-05-14 by bobvmierlo