home-assistant/operating-system #1905: Since 2022.5, static IP address does not hold internet connectivity

ZainAjam, 2022-05-15:
### Describe the issue you are experiencing

I have been running on static IP address for months (if not years) now, and in the last week or two, I noticed my app could not access my home assistant when I am outside of the house. After trying different settings, I eventually realized that when I switched the ETH0 network to DCHP and reserved the address in my router, it solved the problem. For some reason, setting static IP address causes temporary internet access for the first few minutes after setting it, and then it loses the internet connection. 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

The static IP address details I use is 192.168.86.2/24 as the IP, 192.168.86.1 as the DHCP server (my router) and 8.8.8.8 as my DNS server, and tried 192.168.86.1 as DNS server too, with no difference. I'm saddened that I cannot use my own static IP settings now and have to use DHCP with reserved IP in my router.

1. Set static IP details as specified in the description of the issue.
2. Port forward 443 on router
3. Access home assistant remotely. It may work in the first few minutes. Access home assistant remotely 5 minutes later. It should not be possible

### Anything in the Supervisor logs that might be useful for us?

```txt
22-05-15 15:40:13 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-05-15 15:40:13 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-05-15 15:40:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-05-15 15:40:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-05-15 15:40:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-05-15 15:40:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-05-15 15:40:15 INFO (MainThread) [supervisor.host.services] Updating service information
22-05-15 15:40:15 INFO (MainThread) [supervisor.host.network] Updating local network information
22-05-15 15:40:15 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-05-15 15:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-15 15:40:15 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-05-15 15:40:15 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-05-15 15:40:15 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-05-15 15:40:15 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-05-15 15:40:16 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-05-15 15:40:17 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-15 15:40:17 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-05-15 15:40:17 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-05-15 15:44:54 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-15 15:49:57 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-15 15:53:30 INFO (SyncWorker_1) [supervisor.docker.interface] Stopping addon_a0d7b954_adguard application
22-05-15 15:53:35 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_a0d7b954_adguard application
22-05-15 15:53:52 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/adguard/aarch64 with version 4.5.0
22-05-15 15:53:53 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:62518 ssl:default [Connect call failed ('172.30.32.1', 62518)]
22-05-15 15:53:56 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_adguard
22-05-15 15:53:56 INFO (MainThread) [supervisor.api.middleware.security] /network/interface/eth0/info access from a0d7b954_adguard
22-05-15 15:53:57 INFO (MainThread) [supervisor.api.middleware.security] /network/interface/wlan0/info access from a0d7b954_adguard
22-05-15 15:54:36 INFO (SyncWorker_3) [supervisor.docker.interface] Stopping addon_a0d7b954_adguard application
22-05-15 15:54:40 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_a0d7b954_adguard application
22-05-15 15:54:59 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-15 15:56:30 INFO (MainThread) [supervisor.host.network] Updating local network information
22-05-15 15:57:30 ERROR (MainThread) [supervisor.host.network] Can't update config on eth0: ipv4.dns: this property is not allowed for 'method=disabled'
22-05-15 15:57:44 ERROR (MainThread) [supervisor.host.network] Can't update config on eth0: ipv4.dns: this property is not allowed for 'method=disabled'
22-05-15 15:59:35 INFO (MainThread) [supervisor.host.network] Updating local network information
22-05-15 15:59:45 INFO (MainThread) [supervisor.host.network] Updating local network information
22-05-15 16:00:01 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-15 16:00:53 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/adguard/aarch64 with version 4.5.0
22-05-15 16:00:57 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_adguard
22-05-15 16:00:57 INFO (MainThread) [supervisor.api.middleware.security] /network/interface/eth0/info access from a0d7b954_adguard
22-05-15 16:00:58 INFO (MainThread) [supervisor.api.middleware.security] /network/interface/wlan0/info access from a0d7b954_adguard
22-05-15 16:05:04 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-15 16:09:44 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-05-15 16:10:07 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
[... 49 lines cut ...]
[ 1827.518140] bcmgenet fd580000.ethernet eth0: Link is Down
[ 1973.952048] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

### System Health information

Home Assistant Core 2022.5.4
Home Assistant Supervisor 2022.05.1
Home Assistant OS 7.6
Kernel version 5.10.103-v8
Agent version 1.2.1
Path to configuration.yaml: /config

[Developed by a bunch of awesome people.](https://www.home-assistant.io/developers/credits/)

Published under the Apache 2.0 license
Source: [server](https://github.com/home-assistant/core) — [frontend-ui](https://github.com/home-assistant/frontend)

Built using [Python 3](https://www.python.org/), [Lit](https://lit.dev/), Icons by [Google](https://fonts.google.com/icons?selected=Material+Icons) and [Material Design Icons](https://materialdesignicons.com/).

Frontend version: 20220504.1 - latest

### Additional information

None

agners, 2022-05-16:
That is really weird, I am sure a lot of people us static IP, I in fact use static IP since years as well, without any issue accessing it from the outside.

Can it be that when using static IP on device side, that the router assigns the IP via DHCP to another device, hence creating an IP address conflict? Can you check the DHCP IP range?

agners, 2022-05-16:
Hm, maybe this is related to #1902. Can you check `nmcli` profiles as explained in https://github.com/home-assistant/operating-system/issues/1902#issuecomment-1127405708.

ZainAjam, 2022-05-16:
> Let me try. Looks like using DHCP IP reservation on router (instead of
> static IP) does not solve the internet connectivity problem to home
> assistant device. I'm struggling to get add-on updates run and I'm even
> struggling to get the ssh & web terminal add-on installed because internet
> conneciton comes and goes. I've also just run a supervisor update, so maybe
> that will help too. Will run the OS update soon too
>
> *Zain Ajam*
> 0845819728

I've updated the supervisor and OS just now. I'll set it as static IP and
try again.

Before I do this, this is what I currently see in SSH & Web Terminal. What
does this mean, if anything?

[image: image.png]

*Zain Ajam*
***@***.***
0845819728

agners, 2022-05-16:
It means you have two network configurations: One which has been shipped by older OS version, and one generated by Supervisor. You can safely delete the now unused default configuration (`nmcli connection delete "HassOS default"`). I don't think this is the problem though, but worth trying.

ZainAjam, 2022-05-16:
It seems to be working fine now again.

I'm unsure what fixed it, but I did the following things since it stopped
working:

1. Set the network as DHCP on eth0 and reserved the IP address on my
router. Left this overnight and for the better part of the morning the next
day, before switching back to static on eth0. But also did this switch
back, after having done point 2 and point 3 below

2. Deleted the old Hass os default network connection using nmcli
connection delete "hassos default"

3. Upgraded to the latest supervisor and os version that became available
today

I'm not sure which of these 3 steps fixed it, but I am glad it is fixed

>
> Message ID: ***@***.***
> .com>
>

ZainAjam, 2022-05-17:
After a reboot, the problem has returned.

I cannot access the home assistant instance remotely but I can access it on
192.168.86.2:8123

When I switch to DHCP on eth0, it works. When I switch to static IP on
eth0, it stops working after some time or after a reboot

agners, 2022-05-18:
Can you try `nmcli connection delete "HassOS default"` and reboot, and see if that helps.

closed 2022-09-13 by github-actions[bot]

ZainAjam, 2022-10-11:
I've gone through all my devices in my router list and no other is assigned
the 192.168.86.2 address.

Attached is my DHCP pool range on the Google WiFi router.

ZainAjam, 2022-10-11:
Let me try. Looks like using DHCP IP reservation on router (instead of
static IP) does not solve the internet connectivity problem to home
assistant device. I'm struggling to get add-on updates run and I'm even
struggling to get the ssh & web terminal add-on installed because internet
conneciton comes and goes. I've also just run a supervisor update, so maybe
that will help too. Will run the OS update soon too

*Zain Ajam*
***@***.***
0845819728

ZainAjam, 2022-10-11:
To make sure I don't have a conflict with that ip address, I've unplugged
the home assistant raspberry Pi from the power and run an IP address
scanner and 192.168.86.2 is shown as not used.

agners, 2022-10-11:
How do you access your instance outside of your home? Are you using Home Assistant Cloud?