home-assistant/operating-system #1908: Upgrade to 8.0 stuck

sjmotew, 2022-05-16:
### Describe the issue you are experiencing

Trying to upgrade to HA OS 8.0, running on VirtualBox, keeps getting stuck on "A start job is running for Docker Application Container Engine", does not progress during this for one of the components, runs over 15 min....

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Using HA UI, selected update HA OS to 8.0.  Timed out.
2. Looked at VM terminal, shows stuck at docker application container engine, several of these did complete 
3. Attempted reboot, stuck at same place again
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't access
```

### Anything in the Host logs that might be useful for us?

```txt
Can't access
```

julianjd-uk, 2022-05-16:
My Pi 4 is stuck booting.
No error messages, but the last console message reads
[  218.304584] random: crng init done

Nothing since

agners, 2022-05-16:
> Trying to upgrade to HA OS 8.0, running on VirtualBox

Which VirtualBox version are you running? There are some reports that 6.1.32 causes issue (but used to be the case with 7.x already, see #1737 and #1782).

> generic-x86-64 (Generic UEFI capable x86-64 systems)

If you are using VirtualBox, that should be `ova` then I assume?

agners, 2022-05-16:
> My Pi 4 is stuck booting.

@julianjd-uk this is about VirtualBox image being stuck at a later stage. Yours looks like a different issue. Please create a new issue.

sjmotew, 2022-05-16:
Running 6.1.26, yes sorry ova.

agners, 2022-05-16:
So it is really stuck, even after 90s? If you power it off 3 times in that state, the bootloader should switch back to the previous installed OS.

sjmotew, 2022-05-16:
Reset on VM 3 times, each iteration got further on startup.  Successfully launched now with HA OS 8.0,  Learning here.....be patient, I suspect there were some VM-related Docker reconfigures that just needed time to clean up.