home-assistant/operating-system #1909: Raspberry Pi 4 not booting after upgrade to 8.0

julianjd-uk, 2022-05-16:
### Describe the issue you are experiencing

Raspberry Pi 4 8gb not booting after upgrade from HA OS 7.6 to HA OS 8.0
Unable to access any logs as no CLI or web interface
[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.reboot raspberry pi
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Not available
```

### Anything in the Host logs that might be useful for us?

```txt
Not available
```

### System Health information

Not available

### Additional information

Same image as above. Nothing more I can add, sorry.

[image]

agners, 2022-05-16:
From the logs it seems you are using a USB SSD. Do you use it as a boot device or do you boot from the SD card?

From the logs it seems you are using a device which needs to be run with USB quirks enabled (`174c:55aa`). Can you add the following line to `cmdline.txt` on the first partition of your boot device?

`usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:55aa:u`

This is added in a new installation by default, however we do not add it on upgrade. It is kinda interesting that this used to work before.

bschatzow, 2022-05-16:
@julianjd-uk I opened a similar issue during testing of the RCs. Only one other person has mentioned this issue.  Still working with @agners to isolate the issue.  What works for me is unplugging the power for 10 seconds. It has always booted correctly once power is restored. Yours goes farther than mine as my display stays blank on the restart.

jherby2k, 2022-05-16:
I have the "same" adapter (narrator: he probably doesn't. These ASMedia adapters are a mess) and it is working well on 8.0 with the above quirk. i actually found this issue prior to upgrading so didn't try without the quirks. I was not running with quirks under the previous OS.

julianjd-uk, 2022-05-16:
I'm running with an SSD and no SD card.
The USB adaptor looks like a generic Chinese one from StarTech.com and has USB3S2SAT3CB on it.
The SSD is a Kingston 240GB, sorry can't easily read the numbers on it.

So I restored my last backup, from 02:30 this morning and got everything working fine.
Updated cmdline.txt as above and rebooted with success.
Reinstalled HA OS 8.0 and now I'm stuck with the Raspberry Pi bootloader still on screen

[image]
.

julianjd-uk, 2022-05-16:
Well I was very naughty and pulled the SSD, waited and reinserted it.
It booted up all okay, then I did a controlled reboot and everything is running again.
The only update I have that isn't installed is Ring-MQTT with Video Streaming V 5.0.1, the V 5.x.x is a breaking change and breaks my system.

Please close if you think this is all now clear.
Thanks

agners, 2022-05-16:
> Well I was very naughty and pulled the SSD, waited and reinserted it.

It seems like the Raspberry Pi Firmware wasn't able to enumerate properly. That is before anything HAOS related is loaded. If you want that addressed, you'd need to update the firmware (e.g. using a [Raspberry Pi OS rpi-update](https://www.raspberrypi.com/documentation/computers/os.html#rpi-update)). If it still persists, you'd [need to report to the Raspberry Pi firmware developers](https://github.com/raspberrypi/firmware).

> It booted up all okay, then I did a controlled reboot and everything is running again.

Perfect, that means HAOS is working fine in your case: The UMS quirks got applied (since you reinstalled from scratch) and that made it boot successfully.

closed 2022-05-16 by agners