home-assistant/operating-system #1910: HA OS 8.0 fails to boot

bruvv, 2022-05-16:
### Describe the issue you are experiencing

When installing 8.0 from the GUI it fails to boot. There is no IP so I cannot access it remotely nor can I attach a monitor to see why it fails to boot. When I remotely switch the power off and on it boots again after about 90s.

Ps the question:  _What operating system image do you use?_ is not working anymore there is no `board value`

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install 8.0
2. Wait 1 hour
3. Nothing comes online
4. Power off and power back on
5. Works and HA OS is back to 7.6
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing usefull
```

### Anything in the Host logs that might be useful for us?

```txt
https://pastebin.com/TrLb5agg
```

### System Health information

Home Assistant Core 2022.5.4
Home Assistant Supervisor 2022.05.2
Home Assistant OS 7.6
Kernel version 5.10.108
Agent version 1.2.1

Path to configuration.yaml: /config

Developed by a bunch of awesome people.

Published under the Apache 2.0 license
Source: [server](https://github.com/home-assistant/core) — [frontend-ui](https://github.com/home-assistant/frontend)

Built using [Python 3](https://www.python.org/), [Lit](https://lit.dev/), Icons by [Google](https://fonts.google.com/icons?selected=Material+Icons) and [Material Design Icons](https://materialdesignicons.com/).

Frontend version: 20220504.1 - latest

bschatzow, 2022-05-16:
I removed power on mine for 19 seconds and then it booted correctly.

bruvv, 2022-05-16:
And are you updated to 8.0 of 7.6 again?

agners, 2022-05-16:
Anything special on the display when the system is hanging?

bruvv, 2022-05-16:
@agners as mentioned in the first sentence: _There is no IP so I cannot access it remotely nor can I attach a monitor to see why it fails to boot_ so I cannot see why it is hanging.

vinnyspb, 2022-05-16:
I believe I experienced the same problem after upgrading to 8.0. My installation is in Virtualbox (host macOS).
I attached to serial console and the screen was full of messages:
```
Bus error (core dumped)
```
After several minutes of these messages it fall back to the emergency console where practically any command was not possible to execute (same error message).
I rebooted and manually chose "Slot B" in bootloader which allowed me to restore previous OS version.

agners, 2022-05-16:
@vinnyspb this seems a different problem as it is using VirtualBox. Can you open a new issue?

bruvv, 2022-05-16:
@agners I know. But as it is remote I cannot physically access it to attach a monitor to it as I'm on vacation

agners, 2022-05-16:
Ah got it now. Somehow I interpreted "remotely" meaning just via network/from another room.

Do you have the SSH access on host OS (port 22222) enabled? Then we might lift some logs from the system. Otherwise it's probably easier to debug it once you have local display access.

bruvv, 2022-05-16:
No worries I've machine access via 22222 so what can I do to debug this:)?

realthk, 2022-05-16:
> I am running it on a HP T620

Ah, no wonder then why the same thing has happened with my HP T620...

Started the upgrade to HassOS 8, forgot about it a bit, but after about half an hour the system was still not online (not just Home Assistant, but the box itself was not on the network), there is no display connected to it, so I had to restart with the power button.
After a few minutes, normal startup, but still with Hassos 7.6.

Unfortunately, I cannot attach a display to it in the next few days, so have to wait if someone has a solution - HP T620 is a popular platform to run HassOS

ahjcarroll, 2022-05-16:
Had the same problem today, tried to upgrade my system to OS ver. 8 and it didn't load up after a reboot

The Laptop (Dell E5440) I am running Hassos on, seemed to hang.  I assumed it was the SSD, so removed it and tested it, SSD passed all tests. So downloaded the latest image of Hassos but once again it won't load, just displayed about 15 lines on a black background.  Re-imaged the SSD twice, just to be sure the problem wasn't between the keyboard and the chair!  Then checked the network cable in another laptop, and it worked fine. Connected back into the Hassos Laptop running version 8 and noticed that there were no lights on the network card.

Tried the previous version, 7.6 and it loaded correctly and am now restoring from a backup.

So it would appear, in my case at least, that version 8 is missing the Ethernet drivers for my Dell Latitude E5440.

jeggermont, 2022-05-17:
> @bruvv from what I can tell the HP T620 has a graphics card. What I meant is connecting a display to that system and see if that shows something.

I have tried installing 8.0rc1, 8.0rc4 and 8.0,  (using an lubuntu livecd and dd) as well as upgrading from 7.6 to 8.0 on an HP T620. In all cases I see something like SLOT A or SLOT B and then the screen goes black.

phillprice, 2022-05-17:
I have the same system and problem, I will grab a monitor and see if I can see any output other than the previously known problem on the community forum: https://community-assets.home-assistant.io/original/4X/8/f/5/8f5b37d90eba61c1b2ce4dfcf32d02d8f8e9ecf1.jpeg

pickonedev, 2022-05-17:
If I try update to 8.0, the computer will restart, then it will try boot on Slot B, which fails... If I restart and boot from slot A, I am back to 7.6 ...

Big Fail, any solution?

agners, 2022-05-18:
> If I try update to 8.0, the computer will restart, then it will try boot on Slot B, which fails...

Any specific error message showing in this case?

> Big Fail, any solution?

IMHO, big win for A/B fallback option. At least a fallback works :thinking:

bruvv, 2022-05-20:
Indeed 8.1 fixed the error and it's now booting.

closed 2022-05-20 by bruvv

BCsabaEngine, 2022-05-20:
Solves the problem. Thx!!!