home-assistant/operating-system #1913: Can't boot into HAOS after upgrade

LeoSko, 2022-05-16:
### Describe the issue you are experiencing

I was trying to upgrade HAOS to 8.0 but after a while noticed it didn't actually boot up properly and not it is stuck at several variants of BIOS loading without reaching past that.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Easier said than done since everything already seem to be gone. :(
Here is what I did:

1. I believe I had HAOS 7.6
2. Started upgrading to 8.0 together with some another Home-Assistant like icon to upgrade + Mosquitto broker, all at the same time
3. After 10-15 minutes noticed a bunch of error messages in VM log that it can't read from device
4. Restarted VM and now I can't get past "loading Boot0002..." screen, see screenshots attached

### Anything in the Supervisor logs that might be useful for us?

```txt
I can't reach it now.
```

### Anything in the Host logs that might be useful for us?

```txt
I can't reach it now.
```

### System Health information

I can't reach it now.

### Additional information

My configuration and current screens I get:
[image]
[image]
[image]
[image]

VM disk is back from 7.4 hence the name is still 7.4.

I tried **cloning** the VM and I get different result, but nothing seems useful in the menus following [discussion about UEFI Interactive Shell](https://community.home-assistant.io/t/cannot-boot-into-ha-anymore-stuck-in-uefi-interactive-shell/296605/6)
[image]

agners, 2022-05-16:
Hm, just tried to reproduce using VirtualBox 6.1.34 and IDE disk controller. It does seem to boot fine here.

That said, SATA should be the better supported controller at this point, can you remove the IDE controller and add a SATA controller and attach the disk file to that instead?

[image]

DJAMIRSAM, 2022-05-16:
I have the same issue

LeoSko, 2022-05-17:
@agners I tried but it didn't anything. First, I don't get "SATA" label when I add "AHCI (SATA)" - it says AHCI instead unlike on your screenshot.
[image]
Second, after booting it was also first stuck at UEFI Interactive Shell and after changing boot order to boot from HDD, same result as before:
[image]

LeoSko, 2022-05-17:
Also tried downgrading VirtualBox to 6.1.30 (from latest 6.1.32), had to change USB controller to 1.1 instead of 3.0 but it didn't help anyway with same result.

agners, 2022-05-17:
@LeoSko hm, I guess the errors in the VM log really lead to corruption of the boot partition then.

If you have a recent backup, then downloading HAOS 8.0 vdi image and restoring from that backup is probably the quickest fix.

If you don't have a backup, it should be possible to restore the relevant OS parititons without loosing data. Just to be safe, its probably worth making a copy of your disk drive before.

1. Download [Ubuntu Desktop](https://ubuntu.com/download/desktop) which provides a Live system
2. Attach it as a CD-ROM drive

[image]

3. Start the system and choose "Try or Install Ubuntu"
4. In the graphical Install screen, choose "Try Ubuntu"
5. Once the system is up and running, open the browser and download [haos_ova-8.0.vdi.zip](https://github.com/home-assistant/operating-system/releases/download/8.0/haos_ova-8.0.vdi.zip)
6. Press Windows -> Type "Terminal" and Enter to get to the Terminal
7. Use `cd Downloads` to get to the download folder.
8. Type `unzip haos_ova-8.0.vdi.zip` to unpack the disk image
9. Type `sudo apt install qemu-utils` to install utilities
10. Type `sudo modprobe nbd` to load the a kernel driver necessary to mount the disk image
11. Type `sudo qemu-nbd -c /dev/nbd0 haos_ova-8.0.vdi` to mount the disk image
12. Type `sudo dd if=/dev/nbd0 of=/dev/sda bs=512 count=1427456` this copies all system partitions from the disk image to your drive.
13. Type `reboot`
14. Press `Enter` to confirm that you remove the Ubuntu image (its should remove automatically, otherwise remove it manually from the virtual CD-ROM drive.

agners, 2022-05-17:
> I have the same issue

@DJAMIRSAM Did you had issues during installation as well?

The above steps will restore the system partitions, so it might help you too.

AbeltjeNL, 2022-05-17:
I'm also running OVA in VB and didn't encounter any issues. By reading this issue I first changed (my old VB config) from IDE to AHCI (SATA), then changed from Linux other 64x to Linux Linux 2.6 / 3.x / 4.x (64-bit), I then had to reconfigure the boot order (interactive shell) wait 5 seconds, type exit and configure the boot order, booted my system, performed the update to 8.0 and everything works like a charm.

jbtls, 2022-05-17:
Hi, im trying to install it from scratch witch HA8.0 and VB 6.1.34 and its also stuck.

LeoSko, 2022-05-17:
@agners, thanks for helpful steps for case with no backup - they were clear and helped.
One note that would like to leave for others who can face same issue - I used to run HAOS with 1 CPU 2 Gb RAM. However, for running Ubuntu you actually need 2 CPU 4 Gb RAM, so change it before launching into Ubuntu image. I'll probably keep 2-4 config for HAOS as well now, as this might potentially cause those issues I guess.

closed 2022-05-17 by LeoSko

alexbelgium, 2022-06-22:
Hi, I installed a clean image of the vdi on the website which ld to the same error