home-assistant/operating-system #1914: Board 'intel-nuc' not found, using default 'ova'

yang3535, 2022-05-17:
### Describe the issue you are experiencing

I have Home Assistant running on an Intel NUC DC53427HYE

I see "Board 'intel-nuc' not found, using default 'ova'" in my logs since updating the Home Assistant OS to the latest today.  It seems to run fine, but inserts the log entry.  

If it is working like it should and the log entry is normal, please unclutter the log.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: pyhaversion
Source: /usr/local/lib/python3.9/site-packages/pyhaversion/supervisor.py:72
First occurred: 10:38:55 AM (1 occurrences)
Last logged: 10:38:55 AM

Board 'intel-nuc' not found, using default 'ova'
```

### Anything in the Host logs that might be useful for us?

```txt
see above
```

### System Health information

Home Assistant Core 2022.5.4
Home Assistant Supervisor 2022.05.2
Home Assistant OS 8.0
Kernel version 5.15.38
Agent version 1.2.1
Path to configuration.yaml: /config

[Developed by a bunch of awesome people.](https://www.home-assistant.io/developers/credits/)

Published under the Apache 2.0 license
Source: [server](https://github.com/home-assistant/core) — [frontend-ui](https://github.com/home-assistant/frontend)

Built using [Python 3](https://www.python.org/), [Lit](https://lit.dev/), Icons by [Google](https://fonts.google.com/icons?selected=Material+Icons) and [Material Design Icons](https://materialdesignicons.com/).

Frontend version: 20220504.1 - latest

mbo18, 2022-05-18:
This log seems from the `version` Integration. `Intel_nuc` has been removed, you should now use `generic-x86-64` Or setup this integration from the UI instead of using yaml

agners, 2022-05-18:
Yeah this is related to the Home Assistant Core `version` integration. Did you update the system from a `intel-nuc` image to the new `generic-x86-64` image? It could be that a stale entry of the version integration is causing this. Can you try to remove the `version` integration and add again? If the problem still persists, please reopen it in the Home Assistant Core repository.