home-assistant/operating-system #1916: search domain is unset when IP is configured as static

NODeeJay, 2022-05-18:
### Describe the issue you are experiencing

The issue here is the missing search domain under network settings, because with a static IP the search domain not part of the DHCP offer and it cannot be set through the UI.
It seems that as soon as no search domain is set, resolved is not able to resolve any FQDN that is handled by my DNS server.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Assuming my DNS servers answer in a split DNS config authoritative for `iot.home` and `mydomain.net`. My DHCP server provides `iot.home` as search domain in the DHCP offer.
- set a DHCP IP in the UI under /config/network
- run `resolvectl status`, you will get Scopes, Protocols, Current and available DNS Servers AND the DNS Domain (if the search domain is part of the DHCP offer, which it is in my case)
- a `resolvectl query test.mydomain.net` will work
- a `resolvectl query test.iot.home` will work
- a `resolvectl query www.google.com` will work
---
- now change to static IP in the UI
- run `resolvectl status`, the DNS Domain is missing
- a `resolvectl query test.mydomain.net` will NOT work
- a `resolvectl query test.iot.home` will NOT work
- a `resolvectl query www.google.com` will work

any domain that is answered by the DNS server set only seems to work when also a search domain is set.

### Anything in the Supervisor logs that might be useful for us?

```txt
does not show up in supervisor
```

### Anything in the Host logs that might be useful for us?

```txt
except the change from DHCP to DNS nothing of interest
```

Sunny2102, 2022-05-18:
The best description for a well-known Problem.
HA does not have the skill to solve it.

La mejor descripción de un problema conocido!
HA no tiene la capacidad de solucionarlo.

agners, 2022-05-18:
Currently HA does not support custom DNS suffix.

However, if you configure a static IP (along with the DNS server), a FQDN should still resolve just fine. So this:

```
resolvectl query test.mydomain.net
```

should resolve.

In fact, I just tested in my production instance, static IP:

[image]

In resolvectl status, the DNS Domain is missing (obviously). But that is *NOT* a problem if you use FQDN.

FQDN resolves fine:

```
# resolvectl query living-room-sensor.lan
living-room-sensor.lan: 192.168.80.95          -- link: eth0
```

What does not work is using only a single label DNS name (e.g. `living-room-sensor`). That is when DNS suffixes would come in.

If your issue is about resolving host-names without FQDN, than that is a legitmate feature request.

@Sunny2102 great, sounds like you have it. Patch welcome :stuck_out_tongue:

NODeeJay, 2022-05-18:
@agners in fact, im my case HA supports custom domain suffix and even single-label DNS names with custom domain suffix - if applied by DHCP:
[image]

the same when a static IP is set (to make the difference between DHCP and static clearly visible I omit 192.168.103.31 in the static setting):
[image]
in that case even the single-label DNS name got resolved properly by LLMNR

And regarding using the FQDN to avoid the issue: in my case it is a problem if I use the FQDN with a valid top level domain that resolves in a different IP for request from the internet than requests from the internal network (split DNS) and my DNS server is authoritative for both domains.

Test cases:
- All requests can be answered by the local DNS servers, it's hosting the .local and the toplevel domain
- Query
  - single-label with .local,
  - FQDN with .local,
  - FQDN with top level

1. I set DHCP or static in the UI
2. I sip from my coffee (delay ~5000ms)
3. I flush the cache
4. I verify the UI settings are applied
5. I run the test cases
[image]

This is all pretty odd, is it just me? I will setup a test VM network and try to replicate there, but maybe someone has a similar setup and can confirm or roast my findings please.

agners, 2022-05-18:
> in fact, im my case HA supports custom domain suffix and even single-label DNS names with custom domain suffix - if applied by DHCP:

I meant manually configured DNS suffix. I'd say DNS suffix are always custom (in the sense, they are environment specific), whether they are DHCP provided or manually configured.

In the static configured case, it seems that your DNS is not answering at all. The single label use case uses LLMNR, which is a multicast resolution protocol not relying on your DNS server. In the second case, when the system tries to reach out to your DNS server, the DNS server does not seem to respond.

It seems that you have 2 or 3 IPs configured as DNS servers: `192.168.103.32`, `192.168.103.33` and via DHCP also `192.168.103.31`. Are these all reachable?

Can you try nslookup directly, addressing the individual DNS servers?

```
nslookup  gna.<beep>.local 192.168.103.31
nslookup  gna.<beep>.local 192.168.103.32
nslookup  gna.<beep>.local 192.168.103.33
```

NODeeJay, 2022-05-18:
I have several DNS servers configured in an Active Directory. They all host <beep>.local, <beep>.net and other public and "private" domains like iot.home. They also correctly host the reverse lookup zones for all domains.
3 of them are on-site and are pushed as DHCP option together with the search domain, time server, boot file, next server, offset, VLAN IDs for certain manufacturers and so on.
In my tests I used for static just 2 of them that I can see a clear configuration difference when enabling DHCP or static. To answer your question, with DHCP set, at the red line I switch from DHCP to static in the UI with the usual coffee break:
[image]

Active Directory with misconfigured or unreachable DNS would create far bigger issues than a couple of hostnames from some ESPs not getting resolved.

### DNS server logs:
at 1254 I set the DHCP, at 1504 I changed to static and started with flushdns at 1505 the test for static
```
May 18 12:54:43 HASS systemd-resolved[351]: enp0s3: Bus client set DNS server li                                                                                        st to: 192.168.103.33, 192.168.103.32, 192.168.103.31
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client reset search doma                                                                                        in list.
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client set default route                                                                                         setting: no
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client set MulticastDNS                                                                                         setting: no
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client reset DNS server                                                                                         list.
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client set default route                                                                                         setting: yes
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client set MulticastDNS                                                                                         setting: yes
May 18 15:04:37 HASS systemd-resolved[351]: enp0s3: Bus client set DNS server li                                                                                        st to: 192.168.103.32, 192.168.103.33
May 18 15:05:14 HASS systemd-resolved[351]: Flushed all caches.
```

### systemd-resolved debug, starting with DHCP:
```
# resolvectl flush-caches
# SYSTEMD_LOG_LEVEL=debug resolvectl query gna.< beep >.local
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Resolving gna.< beep >.local (family *, interface *).
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.983 path=n/a interface=n/a member=n/a cookie=1 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.resolve1 path=/org/freedesktop/resolve1 interface=org.freedesktop.resolve1.Manager member=ResolveHostname cookie=2 reply_cookie=0 signature=isit error-name=n/a error-message=n/a
Got message type=method_return sender=:1.4 destination=:1.983 path=n/a interface=n/a member=n/a cookie=160326 reply_cookie=2 signature=a(iiay)st error-name=n/a error-message=n/a
gna.< beep >.local: 192.168.103.61             -- link: enp0s3

-- Information acquired via protocol DNS in 23.7ms.
-- Data is authenticated: no; Data was acquired via local or encrypted transport: no
-- Data from: network
Bus n/a: changing state RUNNING → CLOSED
# SYSTEMD_LOG_LEVEL=debug resolvectl query gna
Bus n/a: changing state UNSET → OPENING
sd-bus: starting bus by connecting to /run/dbus/system_bus_socket...
Bus n/a: changing state OPENING → AUTHENTICATING
Resolving gna (family *, interface *).
Bus n/a: changing state AUTHENTICATING → HELLO
Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=Hello cookie=1 reply_cookie=0 signature=n/a error-name=n/a error-message=n/a
Got message type=method_return sender=org.freedesktop.DBus destination=:1.984 path=n/a interface=n/a member=n/a cookie=1 reply_cookie=1 signature=s error-name=n/a error-message=n/a
Bus n/a: changing state HELLO → RUNNING
Sent message type=method_call sender=n/a destination=org.freedesktop.resolve1 path=/org/freedesktop/resolve1 interface=org.freedesktop.resolve1.Manager member=ResolveHostname cookie=2 reply_cookie=0 signature=isit error-name=n/a error-message=n/a
Got message type=method_return sender=:1.4 destination=:1.984 path=n/a interface=n/a member=n/a cookie=160346 reply_cookie=2 signature=a(iiay)st error-name=n/a error-message=n/a
gna: 192.168.103.61                            -- link: enp0s3
     (gna.< beep >.local)

-- Information acquired via protocol DNS in 15.1ms.
-- Data is authenticated: no; Data was acquired via local or encrypted transport: no
-- Data from: cache
Bus n/a: changing state RUNNING → CLOSED
```

[... cut ...]
Bus n/a: changing state RUNNING → CLOSED
```
which also did not arrive at my DNS but resolved, which is most probably caused by the fallback in resolved - I will block connections on the firewall with log to see if it arrives there...
By accident I tried an nslookup to a local FQDN and it resolved. I tried above test cases and they all resolved, internal as external ones. I counterverfied with resolvectl and again domains hosted by the local DNS did not resolve.

Is there a difference between nslookup and resolvectl?

NODeeJay, 2022-08-06:
After blocking aside of DNS also DoH and DoT I can now confirm systemd-resolved uses it's own fallback, as we cannot set the internal search domain in the UI when static IP is configured.

From what I understand I can use the file **_my-network_** in the **_network_** folder on a usb stick. While the config works well for SSH keys and time servers, it somehow does not take over my manual network config - at least it is not shown in the UI.
```
[ipv4]
method=manual
address=192.168.1.111/24
dns=192.168.103.33;192.168.103.32;
dns-search=<beeb>.local

[ipv6]
method=disabled
```

Do I also need to add
```
[connection]
id=Home Assistant OS default
uuid=f62bf7c2-e565-49ff-bbfc-a4cf791e6add
type=802-3-ethernet
llmnr=2
mdns=2
```
to make it work?

Merci vielmal für Dini Zit und Hilf.

closed 2022-11-11 by github-actions[bot]