home-assistant/operating-system #1917: HAOS 8.0 fails to boot after update on  Raspberry Pi 4 Model B 2GB ARM with a WesternDigital Elements SE SSD 480GB

raforo, 2022-05-18:
### Describe the issue you are experiencing

After upgrading to version 8.0, the installation fails, the reboot repeats in a loop.

I turned off the power and on subsequent reboot attempts, it doesn't get past the initial Raspberry boot screen, displaying this screen.

I've backed up the SSD and reinstalled version 7.6 and the latest data backup, but I've lost Portainer CE and Postgres installed in Docker

[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update HAOS 8 over a  Raspberry Pi 4 Model B 2GB ARM with a WesternDigital Elements SE SSD 480GB 
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
nil
```

### Anything in the Host logs that might be useful for us?

```txt
nil
```

agners, 2022-05-18:
So the system is stuck at this screen? It seems that it can load the Raspberry Pi firmware (`start4.efl`), but then the firmware gets stuck somehow. We use the latest firmware from the Raspberry Pi team (https://github.com/raspberrypi/firmware/tree/1.20220331).

Does the latest Raspberry Pi OS work on the same system?

raforo, 2022-05-18:
With version 7.6 of HAOS reinstalled, I have installed version 8.0 and this time it has completed the installation correctly, but I'm still without Portainer CE and I've lost the Postgres data installed in Dockers
Could it be that the Docker not supported by the Supervisor caused the problem?
Will I be able to find the Postgres data somewhere in the HA backup or SSD backup I've made? They contain very important data for me.

agners, 2022-05-18:
> but I'm still without Portainer CE and I've lost the Postgres data installed in Dockers
> Could it be that the Docker not supported by the Supervisor caused the problem?

I don't think that Supervisor forcefully deletes containers/data. However, we do *not* backup other containers, so your data won't be in HA backup.

You'll likely be able to find it when accessing the underlying OS using SSH (port 22222) and run docker commands directly. However, Portainer and custom container is unsupported, I can't help you.

closed 2022-05-18 by agners