home-assistant/operating-system #1919: Interface enx0 does not exist

mxkissnr, 2022-05-18:
### Describe the issue you are experiencing

Can't change IP Settings it shows me this error Message: Interface enx0 does not exist. The AdGuard home plugin is also no longer working, my DNS requests are being rejected. Other plugins like Samba share do not work either.
Hardware: Dell R520
OS: XCP-NG 8.2.1
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Server restart did not help

### Anything in the Supervisor logs that might be useful for us?

```txt
22-05-18 09:02:41 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-05-18 09:02:42 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-18 09:02:42 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-05-18 09:02:42 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-05-18 09:04:31 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:09:35 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:14:38 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:15:33 INFO (MainThread) [supervisor.host.info] Updating local host information
22-05-18 09:15:33 INFO (MainThread) [supervisor.host.services] Updating service information
22-05-18 09:15:33 INFO (MainThread) [supervisor.host.network] Updating local network information
22-05-18 09:15:34 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-05-18 09:15:34 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-05-18 09:19:41 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:24:45 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:29:48 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:32:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-05-18 09:34:52 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:39:56 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:44:59 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:50:03 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 09:55:06 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:00:10 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:02:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-05-18 10:02:16 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from f0e7e328_hassio_google_drive_backup
22-05-18 10:02:16 INFO (MainThread) [supervisor.api.middleware.security] /backups access from f0e7e328_hassio_google_drive_backup
22-05-18 10:02:42 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-05-18 10:02:42 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-05-18 10:02:42 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-05-18 10:02:42 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-05-18 10:02:42 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-05-18 10:02:42 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-05-18 10:03:02 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-05-18 10:03:02 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-05-18 10:03:02 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-05-18 10:03:02 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-18 10:03:02 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-05-18 10:03:02 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-05-18 10:05:13 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:10:16 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:15:20 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:20:23 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:25:26 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:30:29 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-18 10:32:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
[... 141 lines cut ...]
[    0.084488] PM: hibernation: Registered nosave memory: [mem 0xf0000000-0xfbffffff]
[    0.084489] PM: hibernation: Registered nosave memory: [mem 0xfc000000-0xfcffffff]
[    0.084491] PM: hibernation: Registered nosave memory: [mem 0xfd000000-0xffffffff]
[    0.084493] [mem 0xf0000000-0xfbffffff] available for PCI devices
[    0.084495] Booting paravirtualized kernel on Xen HVM
[    0.084500] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1910969940391419 ns
[    0.090817] setup_percpu: NR_CPUS:64 nr_cpumask_bits:64 nr_cpu_ids:64 nr_node_ids:1
[    0.096572] percpu: Embedded 55 pages/cpu s184600 r8192 d32488 u262144
[    0.096593] pcpu-alloc: s184600 r8192 d32488 u262144 alloc=1*2097152
[    0.096596] pcpu-alloc: [0] 00 01 02 03 04 05 06 07 [0] 08 09 10 11 12 13 14 15 
[    0.096606] pcpu-alloc: [0] 16 17 18 19 20 21 22 23 [0] 24 25 26 27 28 29 30 31 
[    0.096616] pcpu-alloc: [0] 32 33 34 35 36 37 38 39 [0] 40 41 42 43 44 45 46 47 
[    0.096625] pcpu-alloc: [0] 48 49 50 51 52 53 54 55 [0] 56 57 58 59 60 61 62 63 
[    0.096676] xen: PV spinlocks enabled
[    0.096680] PV qspinlock hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.096695] Built 1 zonelists, mobility grouping on.  Total pages: 2060958
[    0.096697] Policy zone: Normal
[    0.096701] Kernel command line: BOOT_IMAGE=(hd0,gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=6c807c0ccef74ef9bc6d3f9e4489eaed cgroup_enable=memory fsck.repair=yes console=ttyS0 console=tty1 rauc.slot=B
[    0.096943] Unknown kernel command line parameters "BOOT_IMAGE=(hd0,gpt4)/bzImage cgroup_enable=memory", will be passed to user space.
[    0.096946] printk: log_buf_len individual max cpu contribution: 4096 bytes
[    0.096947] printk: log_buf_len total cpu_extra contributions: 258048 bytes
[    0.096949] printk: log_buf_len min size: 262144 bytes
[    0.097636] printk: log_buf_len: 524288 bytes
[    0.097638] printk: early log buf free: 252856(96%)
```

mxkissnr, 2022-05-18:
Full Host Log:
[    0.000000] Linux version 5.15.38 (builder@59ab1699e95d) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -gee5615946) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Wed May 11 09:52:49 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=6c807c0ccef74ef9bc6d3f9e4489eaed cgroup_enable=memory fsck.repair=yes console=ttyS0 console=tty1 rauc.slot=B
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000ee90bfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ee90c000-0x00000000ee915fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000ee916000-0x00000000ee91afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ee91b000-0x00000000ef99afff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ef99b000-0x00000000ef9cafff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000ef9cb000-0x00000000ef9f2fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ef9f3000-0x00000000ef9fafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000ef9fb000-0x00000000ef9fefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ef9ff000-0x00000000effdefff] usable
[    0.000000] BIOS-e820: [mem 0x00000000effdf000-0x00000000efffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000effff000-0x00000000efffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000fc000000-0x00000000fcffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000020fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: ACPI=0xef9fa000 ACPI 2.0=0xef9fa014 SMBIOS=0xef9cc000 MEMATTR=0xee2edb98 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: Xen HVM domU, BIOS 4.13 05/11/2022
[    0.000000] Hypervisor detected: Xen HVM
[    0.000000] Xen version 4.13.
[    0.000000] platform_pci_unplug: Netfront and the Xen platform PCI driver have been compiled for this kernel: unplug emulated NICs.
[    0.000000] platform_pci_unplug: Blkfront and the Xen platform PCI driver have been compiled for this kernel: unplug emulated disks.
[    0.000000] You might have to change the root device
[    0.000000] from /dev/hd[a-d] to /dev/xvd[a-d]
[    0.000000] in your root= kernel command line option
[    0.000010] HVMOP_pagetable_dying not supported
[    0.041016] tsc: Fast TSC calibration using PIT
[    0.041023] tsc: Detected 2399.968 MHz processor
[    0.041025] tsc: Detected 2400.004 MHz TSC
[    0.041043] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.041048] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.041055] last_pfn = 0x20fc00 max_arch_pfn = 0x400000000
[    0.041143] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.041156] last_pfn = 0xf0000 max_arch_pfn = 0x400000000
[    0.052055] Using GB pages for direct mapping
[    0.052295] Secure boot could not be determined
[    0.052299] ACPI: Early table checksum verification disabled
[    0.052305] ACPI: RSDP 0x00000000EF9FA014 000024 (v02 Xen   )
[    0.052311] ACPI: XSDT 0x00000000EF9F90E8 00004C (v01 Xen    HVM      00000000      01000013)
[    0.052318] ACPI: FACP 0x00000000EF9F8000 0000F4 (v04 Xen    HVM      00000000 HVML 00000000)
[    0.052325] ACPI: DSDT 0x00000000EE90C000 0092A3 (v02 Xen    HVM      00000000 INTL 20160527)
[    0.052329] ACPI: FACS 0x00000000EF9FE000 000040
[    0.052333] ACPI: APIC 0x00000000EF9F7000 000260 (v02 Xen    HVM      00000000 HVML 00000000)
[    0.052337] ACPI: HPET 0x00000000EF9F6000 000038 (v01 Xen    HVM      00000000 HVML 00000000)
[    0.052340] ACPI: WAET 0x00000000EF9F5000 000028 (v01 Xen    HVM      00000000 HVML 00000000)
[    0.052344] ACPI: BGRT 0x00000000EF9F4000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.052347] ACPI: Reserving FACP table memory at [mem 0xef9f8000-0xef9f80f3]
[    0.052350] ACPI: Reserving DSDT table memory at [mem 0xee90c000-0xee9152a2]
[    0.052351] ACPI: Reserving FACS table memory at [mem 0xef9fe000-0xef9fe03f]
[    0.052353] ACPI: Reserving APIC table memory at [mem 0xef9f7000-0xef9f725f]
[    0.052354] ACPI: Reserving HPET table memory at [mem 0xef9f6000-0xef9f6037]
[    0.052356] ACPI: Reserving WAET table memory at [mem 0xef9f5000-0xef9f5027]
[    0.052357] ACPI: Reserving BGRT table memory at [mem 0xef9f4000-0xef9f4037]
[    0.059431] No NUMA configuration found
[    0.059434] Faking a node at [mem 0x0000000000000000-0x000000020fbfffff]
[    0.059440] NODE_DATA(0) allocated [mem 0x20fbfb000-0x20fbfffff]
[    0.059485] Zone ranges:
[    0.059486]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.059489]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[... 707 lines cut ...]
[... cut ...]
[15301.363390] audit: type=1334 audit(1652858133.408:251): prog-id=24 op=LOAD
[15301.602266] audit: type=1334 audit(1652858133.647:252): prog-id=25 op=LOAD
[15301.602378] audit: type=1334 audit(1652858133.647:253): prog-id=26 op=LOAD
[15331.677461] audit: type=1334 audit(1652858163.724:254): prog-id=0 op=UNLOAD
[15331.677469] audit: type=1334 audit(1652858163.724:255): prog-id=0 op=UNLOAD
[15331.856500] audit: type=1334 audit(1652858163.903:256): prog-id=0 op=UNLOAD
[15331.856507] audit: type=1334 audit(1652858163.903:257): prog-id=0 op=UNLOAD

agners, 2022-05-18:
This seems a rather special setup. I wonder if the capitalization (`enX0` vs. `enx0`) is the culprit, the log seems to suggest that the interface name is actually `enX0`.

```
[ 5.788852] vif vif-0 enX0: renamed from eth0
```

Can you install the `SSH & Web Terminal` add-on from the community store and run the following commands and post their output?

```
ip addr
ha network info
```

mxkissnr, 2022-05-18:
thank you for the quick reply enclosed the data:
ip addr
[image]

ha network info
[image]

mxkissnr, 2022-05-18:
When I try this I get the error that this interface does not exist

`ha network update enX0 --ipv4-method=static --ipv4-address 10.1.10.95/24 `
`Error: Interface enx0 does not exist`

agners, 2022-05-18:
This seems to be a Supervisor issue. I am transferring this issue.