home-assistant/operating-system #1920: Home Assistant OS 8.0 fails to boot completely on DELL 3090 MFF

agners, 2022-05-18:
> > @pickonedev
> > > Big Fail, any solution?
> > 
> > 
> > It call that big win, the fallback did work and your system is still running!
> > > If I try update to 8.0, the computer will restart, then it will try boot on Slot B, which fails...
> > 
> > 
> > How odes it fail exactly? Any specific error message showing up?
> 
> @agners Well... I just click on install from Home Assistant, then the computer restarts and shows me two different boot options, where the computer it is selecting the Slot B, instead Slot A. After this, the computer freezes with a screen where the text it is cutted. After this, I force restart the computer and HA it is still on v7.6 . I have even updated my bios to the last version available. I am using a DELL 3090 MFF Please, take a look at how the computer is after it is trying to update. [image] [image]

Exactly the same problem here, non-readable message and computer freeze. Using a Lenovo Yoga and HAOS at 7.6 verision. Any solution?

_Originally posted by @Apinyot in https://github.com/home-assistant/operating-system/issues/1912#issuecomment-1129862486_

agners, 2022-05-18:
@Apinyot is the kernel also booting in your case (log messages with the timestamps on the left side as the screenshot of @pickondev is showing?

Apinyot, 2022-05-18:
> 

Yes, exactly the same in the screenshot posted. I need to switch the laptop off (pressing power button for several seconds) to reach the bootloader screen again. If selecting the first option I can boot to the v7.6 HAOS but if I select the second, it freezes there.

agners, 2022-05-18:
@Apinyot what laptop make/model is that exactly?

This is most likely related to #1774, and the associated changed to use `SYSFB_SIMPLEFB` and `DRM_SIMPLEDREM`. Looking at other projects, it seems that there have been other similar breakages:
https://bugs.archlinux.org/task/72645
https://bugzilla.redhat.com/show_bug.cgi?id=2071209

@pickonedev @Apinyot Those seem to be specific to NVIDIA graphics cards. Do you happen to know what type of graphics card your system is using?

pickonedev, 2022-05-18:
@agners No, my configuration is:

Mini PC Dell OptiPlex 3090 MFF
Intel® Core™ i7-10700T @4.50 GHz, Comet Lake
32GB DDR4
512GB SSD NVME
512GB SSD SATA
Wi-Fi
Intel® UHD Graphics 630

pickonedev, 2022-05-18:
> @pickonedev (and also @Apinyot) can you try the following in the boot menu, select `Slot B`, press `e`, move the curser to the end of `linux` line and add `initcall_blacklist=simpledrm_platform_driver_init` (no return or backspace needed, that is done automatically by GRUB2). Then press F10. Does this boot into HAOS 8.0 successfully?
> 
> [image]

Yes, it is working. I'm in 8.0 now. But please tell me, I must do this everytime I restart my computer or it will remain saved? And two more qestions :-D It will be ok if I will update to future releases? If I still have two slots in the boot menu, can I have only one? Sorry, I don't know how it is working, thats why so many questions 🥇

agners, 2022-05-18:
> Yes, it is working. I'm in 8.0 now. But please tell me, I must do this everytime I restart my computer or it will remain saved?

It won't be saved. This was a test to confirm what the exact problem is. I am working on fixing it with the next release. So until then, you need to use Slot A or type the work around.

> If I still have two slots in the boot menu, can I have only one or maybe can I get rid of the old version from slot a?

No, you can't get rid of the old version. That is by design, left there as a fallback. If everything work as expected, it will automatically pick the new slot once you update to the next version (8.1) and you won't need to worry about it.

pickonedev, 2022-05-18:
I understand. Ok, I will wait for 8.1 then :-D Then you for your help.

Oh, wait, and when the 8.1 will be released, we will still have two slots? Used in the same way as now? If one "goes down", it will load the other one? I am trying to figure out how it will be in the future.

Apinyot, 2022-05-18:
> @pickonedev (and also @Apinyot) can you try the following in the boot menu, select `Slot B`, press `e`, move the curser to the end of `linux` line and add [...]

OK, it worked for me also. Considering we cannot make a clean boot (by default slot b will not porperly boot and we have to mannually shitch to slot a) it ios better for us to remain in v8.0 or v7.6?

agners, 2022-05-18:
Thanks for testing!

Both should be save, what you prefer. Probably easier to stay in v7.6, so you don't have to type the work around.

pickonedev, 2022-05-18:
Yes, I think I will wait for 8.1, staying to 7.6 till then. Thanks @agners

agners, 2022-05-19:
> By the way, If I modify something in 8.0 it will be in 7.6 available as well?

Yes, the data partition is shared: Adding/removing/updating Home Assitant Core/Supervisor or Add-ons and any user changes are available in both OS versions. The two OS versions share the same data partition.

agners, 2022-05-20:
This is fixed in Home Assistant OS 8.1 currently on beta channel. For those affected by this issue, can you join the beta channel to update to HAOS 8.1 and confirm that its indeed working for your case? Thanks!

closed 2022-05-20 by agners