home-assistant/operating-system #1921: Strange issue with upgrade to Home Assistant OS 8.0

onlize, 2022-05-18:
### Describe the issue you are experiencing

I see multiple people saying that the boot has an issue after upgrading to OS 8.0, but I think I have an interesting case, so I thought I would share it with you. I have an older fitlet computer that I have used for a few years. It does not have a monitor or a keyboard attached to it.

I always keep my HA up to date, so it has all the necessary updates. Now I tried to upgrade it to OS 8.0. After the upgrade, the computer reboots and never comes back. If I unplug my computer and plug it in again, it comes back with OS v 7.6. 

I wanted to see errors, so I plugged in a monitor. After upgrading to OS 8.0, everything works great, and it comes up with OS v8.0. I can reboot it, and there are no issues.

As soon as I unplug the monitor, it hangs during reboot and comes up only after I unplug and plug power into my computer. And it comes back as OS v 7.6.

I tried it a few times and have the same issues every time.

Thank you.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update to OS 8.0
2. Reboot
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing.
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing.
```

agners, 2022-05-18:
This is probably related to #1920 which also report issues related to graphics. Maybe you system as well has problems, only in your case when no monitor is conected.

What type of system are you using exactly (e.g. graphics card)?

onlize, 2022-05-20:
OS 8.1 fixed the problem.

Thank you.

closed 2022-05-20 by onlize