home-assistant/operating-system #1922: Move data to SSD not working

istornz, 2022-05-18:


agners, 2022-05-19:
In my case I am unable to select the datadisk in the frontend, but that seems independent. I created an issue for this:
https://github.com/home-assistant/frontend/issues/12729

However, when using the console command moving to data disk seems to work on my test: The system reboots, the disk blinks with lots of activity, and I end up with the data being  on the datadisk.

Can you [setup SSH access to the underlying OS](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host) and access the logs of the following services after a failed moving attempt?

```
systemctl status hassos-data.service
```

istornz, 2022-05-20:
Thanks for your answer @agners, I will try again soon 👍

istornz, 2022-05-23:
@agners So I tried but it seems the command was not found...
```
-bash: systemctl: command not found
```

Instead, I tried to watch log inside the "logs" section in the Web UI, for "Supervisor" I see:
```
22-05-23 18:05:03 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/sda1 - /dev/disk/by-id/ata-KINGSTON_SUV400S37120G_50026B776800DF08-part1
22-05-23 18:05:03 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/sda2 - /dev/disk/by-id/ata-KINGSTON_SUV400S37120G_50026B776800DF08-part2
```

Note: The SSD seems to be formatted because it is unreadable when I plug it on my Mac

agners, 2022-05-24:
@istornz are the host log showing anything? What type of system are you using? Is this an upgraded installation or a new installation?

istornz, 2022-05-24:
@agners No there is no errors in the host log...
I'm using a Raspberry Pi 4 Model B Rev 1.4 (with USB boot enabled, maybe I can try to disable it ?)
```
Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1
Kernel version 5.15.32-v8
```
I upgraded recently to 8.1 but it was never working in a fresh install 8.0 (did the install over 7 times 😂)

agners, 2022-05-24:
Some SSD devices are known to cause problems, and we added storage quirks in #1864 to Raspberry Pi's `config.txt`. However, that only gets shipped with a new installation since version 8.0, hence the question. Your installation should have those storage quirks present. Also, since there are no errors in the host log, it seems that the drive is working properly.

At this point I need the logs from the `hassos-data.service` to debug it further.

You can access it by login on the local terminal (screen/keyboard), using `login` then bellow commands. Alternatively, you can enable host SSH access (port 22222) as [described on this page](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host). With that you should be able to execute the commands directly:

```
systemctl status hassos-data.service
journalctl -u hassos-data.service
```

closed 2022-08-29 by github-actions[bot]