home-assistant/operating-system #1925: Home Assistant 8.0 VirtualBox (6.1.30), Fails during boot AND will not shutdown via a acpipowerbutton Command

RJ-Make, 2022-05-18:
### Describe the issue you are experiencing

**_1st Issue:_** During VirtualBox VM (6.1.30) start; Home Assistant will fail during the boot sequence (see screenshot). I have to issue a power off command to make another attempt. 

On the 3rd to 5th attempt it will successfully complete boot. (Seems to get stuck in the same place each time)

**_2nd Issue:_** OS Version 8, inside VirtualBox no longer respects the `acpipowerbutton` command. 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Updated OS from 7.6 (I believe) to 8.0

### Anything in the Supervisor logs that might be useful for us?

```txt
Logs appear to not be written to or are overwritten on a successful boot.
```

### Anything in the Host logs that might be useful for us?

```txt
Logs added below
```

### System Health information

## System Health

version | core-2022.5.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.15.38
arch | x86_64
timezone | America/New_York

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4905
Installed Version | 1.24.5
Stage | running
Available Repositories | 1056
Downloaded Repositories | 14

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | April 20, 2023, 8:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-2.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.2
docker_version | 20.10.14
disk_total | 30.8 GB
disk_used | 18.2 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.6.1), Studio Code Server (5.0.4), InfluxDB (4.4.1), File editor (5.3.3), Nextcloud Backup (0.17.3), SSH & Web Terminal (10.1.3), Samba Backup (5.0.0), Z-Wave JS (0.1.58), Z-Wave JS to MQTT (0.40.0), Network UPS Tools (0.10.0), MariaDB (2.4.0), Grafana (7.5.2), Glances (0.15.0), ESPHome (2022.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 7
views | 28
mode | storage

</details>

### Additional information

[HostLogs.txt](https://github.com/home-assistant/operating-system/files/8718914/HostLogs.txt)

[image]

agners, 2022-05-18:
Duplicate of #1896.

agners, 2022-05-19:
Um, sorry, I guess it was a bit late yesterday :)

Not sure why booting breaks. It seems that the top line of your screenshot (CR2: ...) indicates some crash during startup. Can you share a screenshot of the logs further up?

What if you boot the previous release (by selecting the other boot slot in the GRUB2 boot menu?

RJ-Make, 2022-05-19:
> Um, sorry, I guess it was a bit late yesterday :)
>
> Not sure why booting breaks. It seems that the top line of your screenshot (CR2: ...) indicates some crash during startup. Can you share a [...]

Thank you for taking a look at this. I'm new to all of this so please be patient with me. 

To answer your questions

1) The screen goes by fast, and I'm not sure how to scroll up (or how to see the text that is cut off on the right) once the text is past.

2) Same kind of problem as above, that screen is only present for a bit longer then a blink of my eye, so I'm not sure how to select it.  That said, I didn't have this problem on 7.6, but to be fair, there were also a few addon's/integrations (listed above) that updated also on the same day or day after. 

I'm not sure if this observation is relevant, but with OS 8.0 I notice an extremely long shut down cycle.  Example: from the time I hit "Shutdown Host" in the UI, it now takes over 2 min, 7 seconds to completely close.

If you can give me some direction in how to obtain the information you require, that would be most helpful.  VirtualBox is running in Windows 10.  I'm guessing there are no indication of the problem in the host log file I provided?

agners, 2022-05-19:
Do you get access to a shell at one point? If so, you can display the kernel logs again from `dmesg`. If you use `dmesg | more` you should see the logs page by page (use space to get to the next page).

agners, 2022-05-19:
Can you select the other boot slot in GRUB2 at startup and try to update to the latest development build:

```
ha supervisor options --channel=dev
ha supervisor update
ha os update --version=9.0.dev20220519
```

After that, switch back to the stable channel so you don't get development versions of Supervisor/Core

```
ha supervisor options --channel=stable
ha supervisor update
```

agners, 2022-05-19:
If you interrupt boot early three times, GRUB2 should automatically select the previous version.

RJ-Make, 2022-05-19:
Does it matter how long I let it boot for? As I tried that at the very beginning and it always selects A

RJ-Make, 2022-05-19:
I get this is error `"Error: Error raised from OTA Webserver: 404"` when I use this command `ha os update --version=9.0.dev20220519`

RJ-Make, 2022-05-19:
Seems like it's still failing at the same point with the dev build.

[image]

RJ-Make, 2022-05-20:
I rolled a new 7.6 VM, and recovered my system via HAOS backup and I'm now working reliably again. 

I did save the 8.0 (now 9.0 Dev) VM. 

So what's next?

bruvv, 2022-05-20:
8.1 fixed this probably

RJ-Make, 2022-05-20:
> 8.1 fixed this probably

I would have expected the version 9 of the dev branch to contain those fixes??

agners, 2022-05-23:
Other than the stable kernel update no specific fixes to specifically address this problem went into 8.1. And yes, 9.0 dev builds include all the fixes which go into 8.1.

> On the 3rd to 5th attempt it will successfully complete boot.

Can you get the host logs (`dmesg`) when the boot completes?

RJ-Make, 2022-05-23:
I will attempt the dmesg when I get brave enough to try 8.1+ again.  Probably next week or so.

jeremypsl, 2022-05-26:
Is it possible de disable those news slots selection ?
Sometimes after a reboot of the VM, it load the A slot and give me an old OS...

RJ-Make, 2022-05-26:
Probably best to start a new bug submission, as it doesn't appear to have anything to do with this bug.

RJ-Make, 2022-05-28:
I'm closing this for now as I've moved to VMware and spun up a new 8.1 install. Which doesn't _**seem**_ to have this boot issue.

closed 2022-05-28 by RJ-Make