home-assistant/operating-system #1926: Upgrade to HA OS 8.0 fails

vaibhav2912, 2022-05-19:
### Describe the issue you are experiencing

I saw other bugs but the message I see on screen when reboot gets stuck is different. So creating this issue.

I am trying to update Home Assistant OS to 8.0 using the GUI. I checked my configuration using [Check Home Assistant configuration 3](https://github.com/home-assistant/addons/blob/d133b38bba32467b34d87c7245dcd9cba40bad8a/check_config/README.md) add on as mentioned on [Common Tasks - Operating System](https://www.home-assistant.io/common-tasks/os/).

The update gets stuck and I don’t see lights blinking on the ethernet port so I think it looses internet connection.

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update to HA OS 8.0 from the GUI.
2. Wait for a host reboot. Gets stuck.
3. Unplug host and plug back in.
4. HA boots on version 7.6.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

### System Health information

Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.2
Home Assistant OS 7.6
Kernel version 5.10.108
Agent version 1.2.1
Path to configuration.yaml: /config

[Developed by a bunch of awesome people.](https://www.home-assistant.io/developers/credits/)

Published under the Apache 2.0 license
Source: [server](https://github.com/home-assistant/core) — [frontend-ui](https://github.com/home-assistant/frontend)

Built using [Python 3](https://www.python.org/), [Lit](https://lit.dev/), Icons by [Google](https://fonts.google.com/icons?selected=Material+Icons) and [Material Design Icons](https://materialdesignicons.com/).

Frontend version: 20220504.1 - latest

agners, 2022-05-19:
It does seem similar to #1920, can you try the work around posted in https://github.com/home-assistant/operating-system/issues/1920#issuecomment-1129987911?

agners, 2022-05-20:
This is likely fixed in Home Assistant OS 8.1 currently on beta channel. You should be able to select the other boot slot at startup, and update to HAOS 8.1.

vaibhav2912, 2022-05-20:
> It does seem similar to #1920, can you try the work around posted in [#1920 (comment)](https://github.com/home-assistant/operating-system/issues/1920#issuecomment-1129987911)?

I tried but can't. During the reboot the screen to select different slots is shown for less than a second. Tried violently clicking the down arrow to select the second slot but was defeated.

I will try the HAOS 8.1 once it is released as stable.

brylee123, 2022-05-20:
@agners unfortunately this is the reason why I am here. 8.1 upgrade crashed. I can ping the rpi4 but I cannot connect nor SSH into it. I don't have details yet but I will update as I go along.

agners, 2022-05-23:
@vaibhav2912 

> I will try the HAOS 8.1 once it is released as stable.

HAOS is now on stable channel. Let me know if the update worked for you.

vaibhav2912, 2022-05-23:
@agners I am being cautious and watching the bug reports for HAOS 8.1 upgrade. I see some people who (like me) tried HAOS 8.0 upgrade before, are still having issues after HAOS 8.1 upgrade.
Also issue about Bluetooth on user reported bug on after upgrade to HAOS 8.1.

I only have one instance of HA that I am using. I don't have a test setup. So I am being very cautious about this upgrade.

agners, 2022-05-23:
Keep in mind that at least 30'000 users updated to 8.0 or 8.1 successfully already. Many reports are a very specific setup. Do you happen to have a report which thinks would affect your installation?

> So I am being very cautious about this upgrade.

That make sense, better safe than sorry :) All I am saying that the issue you have been seeing should be addressed with 8.1.

vaibhav2912, 2022-05-23:
> Do you happen to have a report which thinks would affect your installation?

No. Nothing specific. Generic x86-64 installation of HAOS.

> Keep in mind that at least 30'000 users updated to 8.0 or 8.1 successfully already.

OK. That gives me some confidence to go ahead and upgrade.
I will update and report the status.

vaibhav2912, 2022-05-23:
@agners
I updated to HAOS 8.1. Everything went fine. Up and running on HAOS 8.1. Thanks

closed 2022-05-23 by vaibhav2912