home-assistant/operating-system #1934: UI is no longer showing new updates

oneseventhree, 2022-05-21:
### Describe the issue you are experiencing

UI is saying I am running:
Version    core-2022.5.4

But 2022.5.5 is now released - normally it just shows I need to update in UI but it's not showing that anymore.

I've never had this problem before until the .5.x releases. Is there a way to fix it so the update shows up in the UI again? Theres no error logs so I am not sure how to fix this - I can update manually in terminal but that's a bit of a pain every release.

How can I fix this? - Thanks in advance

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Wait for new update

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System Health information

N/A

agners, 2022-05-23:
Sometimes it takes a bit for the update to show up, is this still a problem? Whether updates are available/necessary is determined by the Supervisor. You'll have to open an issue in that repository.

closed 2022-05-23 by github-actions[bot]

oneseventhree, 2022-05-26:
> Sometimes it takes a bit for the update to show up, is this still a problem? Whether updates are available/necessary is determined by the Supervisor. You'll have to open an issue in that repository.

Still a problem - I've logged it here now: https://github.com/home-assistant/supervisor/issues/3657