home-assistant/operating-system #1935: Bluetooth no longer working on VirtualBox after upgrade

cryptton2004, 2022-05-21:
### Describe the issue you are experiencing

While running Homeassistant OS 7.6 on VirtualBox, Bluetooth works (bluetoothctl shows the controller).
After upgrading to 8.1, Bluetooth stops working: "No default controller available"
Downgraded again, works, upgraded: stops working.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install the OVA image on a VirtualBox machine
2. Note: version installed 7.6
3. Add to yaml configuration
```
device_tracker:
  - platform: bluetooth_tracker
```
4. Connect to the VM, hit 'login' and run
`bluetoothctl scan on`
The cli returns the controller and relevant information
5. Upgrade to 8.1 and restart
6. Notice the issue
7. Additionally you can repeat step 4:
`"No default controller available`
Note: none are displayed

### Anything in the Supervisor logs that might be useful for us?

```txt
A warning:
22-05-21 11:59:51 WARNING (MainThread) [supervisor.host.network] Requested to update interface enp0s3 which does not exist or is disabled.
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.194225] acpi PNP0A03:00: fail to add MMCONFIG information, can't access extended PCI configuration space under this bridge.
[    0.194386] PCI host bridge to bus 0000:00
...
[    2.606952] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
..
```

### System Health information

Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.2
Home Assistant OS 8.1
Kernel version 5.15.41
Agent version 1.2.1
Path to configuration.yaml: /config

[Developed by a bunch of awesome people.](https://www.home-assistant.io/developers/credits/)

Published under the Apache 2.0 license
Source: [server](https://github.com/home-assistant/core) — [frontend-ui](https://github.com/home-assistant/frontend)

Built using [Python 3](https://www.python.org/), [Lit](https://lit.dev/), Icons by [Google](https://fonts.google.com/icons?selected=Material+Icons) and [Material Design Icons](https://materialdesignicons.com/).

Frontend version: 20220504.1 - latest
[... 74 lines cut ...]
  File "/usr/local/lib/python3.9/site-packages/bluetooth/bluez.py", line 291, in _gethcisock
    raise BluetoothError ("error accessing bluetooth device")
bluetooth.btcommon.BluetoothError: error accessing bluetooth device
2022-05-21 12:03:04 WARNING (SyncWorker_6) [homeassistant.components.mitemp_bt.sensor] Polling error Could not read data from Mi Temp sensor 58:2D:34:3A:86:79
2022-05-21 12:03:04 WARNING (SyncWorker_4) [homeassistant.components.mitemp_bt.sensor] Polling error Could not read data from Mi Temp sensor 58:2D:34:3A:86:79
2022-05-21 12:03:05 ERROR (SyncWorker_0) [homeassistant.components.bluetooth_tracker.device_tracker] Couldn't discover bluetooth devices: [Errno 19] No such device
2022-05-21 12:03:05 ERROR (MainThread) [homeassistant.components.bluetooth_tracker.device_tracker] Error looking up Bluetooth device
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/bluetooth/bluez.py", line 289, in _gethcisock
    sock = _bt.hci_open_dev (device_id)
_bluetooth.error: no available bluetoot devices
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 167, in perform_bluetooth_update
    friendly_name = await hass.async_add_executor_job(lookup_name, mac)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 127, in lookup_name
    return bluetooth.lookup_name(mac, timeout=5)  # type: ignore[no-any-return]
  File "/usr/local/lib/python3.9/site-packages/bluetooth/bluez.py", line 61, in lookup_name
    sock = _gethcisock ()
  File "/usr/local/lib/python3.9/site-packages/bluetooth/bluez.py", line 291, in _gethcisock
    raise BluetoothError ("error accessing bluetooth device")
bluetooth.btcommon.BluetoothError: error accessing bluetooth device
2022-05-21 12:03:14 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.mitemp_bt_battery is taking over 10 seconds

agners, 2022-05-23:
```
[    2.606952] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
```

Hm, that sounds like there is an issue with forwarding the USB device. It sounds like with Linux 5.15 something broke OS forwarding. I am not aware of any driver changes which could have caused that.

What Host OS do you use? Do you use the VirtualBox Extension Pack?

Can you attach the full host logs (`dmesg`) of 7.6 and 8.1?

ben0551, 2022-05-25:
Confirmed Intel Wireless 9260 Bluetooth does not work on 8.0 and 8.1

Found this? https://bugzilla.kernel.org/show_bug.cgi?id=215167

ben0551, 2022-05-25:
Windows 11, Extension Pack installed.

Puddingpanzer, 2022-05-25:
Same Error for me with 8.0 and 8.1 at Synology Virtual Manager

agners, 2022-05-25:
@ben0551 @Puddingpanzer do your host logs show similar issues to the OT (`usb usb1-port2: Cannot enable. Maybe the USB cable is bad?`)?

cryptton2004, 2022-05-25:
Sorry for the late reply!
Seems that the cable error no longer shows after full shutdown and startup, but there's still something:
`Bluetooth: hci0: command 0xfc01 tx timeout`
Full log from 8.1:
`[    0.000000] Linux version 5.15.41 (builder@58360eda4381) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g2eebdbad5) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Fri May 20 00:38:27 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=bd464feb8d734efbb1ca93700c990a2c cgroup_enable=memory fsck.repair=yes console=ttyS0 console=tty1 rauc.slot=B
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bdd5efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bdd5f000-0x00000000bdd5ffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bdd60000-0x00000000bde90fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bde91000-0x00000000bdf1afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bdf1b000-0x00000000bef9afff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bef9b000-0x00000000befcafff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000befcb000-0x00000000beff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000beff3000-0x00000000beffafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000beffb000-0x00000000beffefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000befff000-0x00000000bf46afff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf46b000-0x00000000bff6afff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000bff6b000-0x00000000bffeefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bffef000-0x00000000bffeffff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: ACPI=0xbeffa000 ACPI 2.0=0xbeffa014 SMBIOS=0xbeffd000 
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 27999001, primary cpu clock
[    0.000001] kvm-clock: using sched offset of 2619379450 cycles
[    0.000005] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000011] tsc: Detected 3491.912 MHz processor
[    0.001874] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001881] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001892] last_pfn = 0xbf46b max_arch_pfn = 0x400000000
[    0.001908] Disabled
[    0.001910] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[    0.001915] CPU MTRRs all blank - virtualized system.
[    0.001920] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.001968] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.003839] Secure boot could not be determined
[    0.003845] ACPI: Early table checksum verification disabled
[    0.003849] ACPI: RSDP 0x00000000BEFFA014 000024 (v02 VBOX  )
[    0.003857] ACPI: XSDT 0x00000000BEFF90E8 000044 (v01 VBOX   VBOXFACP 00000001      01000013)
[    0.003868] ACPI: FACP 0x00000000BEFF7000 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.003880] ACPI: DSDT 0x00000000BEFF4000 002325 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.003889] ACPI: FACS 0x00000000BEFFE000 000040
[    0.003896] ACPI: APIC 0x00000000BEFF8000 00005C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.003906] ACPI: SSDT 0x00000000BEFF3000 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.003915] ACPI: BGRT 0x00000000BDD5F000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.003923] ACPI: Reserving FACP table memory at [mem 0xbeff7000-0xbeff70f3]
[    0.003928] ACPI: Reserving DSDT table memory at [mem 0xbeff4000-0xbeff6324]
[    0.003932] ACPI: Reserving FACS table memory at [mem 0xbeffe000-0xbeffe03f]
[    0.003936] ACPI: Reserving APIC table memory at [mem 0xbeff8000-0xbeff805b]
[    0.003940] ACPI: Reserving SSDT table memory at [mem 0xbeff3000-0xbeff31cb]
[    0.003944] ACPI: Reserving BGRT table memory at [mem 0xbdd5f000-0xbdd5f037]
[    0.004674] No NUMA configuration found
[    0.004677] Faking a node at [mem 0x0000000000000000-0x00000000bf46afff]
[    0.004685] NODE_DATA(0) allocated [mem 0xbf213000-0xbf217fff]
[    0.004730] Zone ranges:
[    0.004732]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004737]   DMA32    [mem 0x0000000001000000-0x00000000bf46afff]
[    0.004741]   Normal   empty
[... 651 lines cut ...]
[  118.433289] audit: type=1327 audit(1653474855.851:167): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4C002D6E
[... cut ...]
[  124.217817] audit: type=1334 audit(1653474861.636:183): prog-id=21 op=LOAD
[  124.217942] audit: type=1334 audit(1653474861.636:184): prog-id=22 op=LOAD
[  154.272547] audit: type=1334 audit(1653474891.690:185): prog-id=0 op=UNLOAD
[  154.272554] audit: type=1334 audit(1653474891.690:186): prog-id=0 op=UNLOAD
[  154.790190] audit: type=1334 audit(1653474892.208:187): prog-id=0 op=UNLOAD
[  154.790196] audit: type=1334 audit(1653474892.208:188): prog-id=0 op=UNLOAD`

Full log from 7.6
`will be here'

ben0551, 2022-05-25:
> @ben0551 @Puddingpanzer do your host logs show similar issues to the OT (`usb usb1-port2: Cannot enable. Maybe the USB cable is bad?`)?

I get 
Bluetooth: hci0: command 0xfc01 tx timeout

cryptton2004, 2022-05-25:
@agners 
Running Virtual Box on W10. No extensions. 
I edited the comment above.

cryptton2004, 2022-06-16:
Hello,

Is there any plan of including the fix?
(https://github.com/home-assistant/operating-system/issues/1935#issuecomment-1136689805)

fdsasdfddsa, 2022-07-11:
Same issue of HAOS version 8.2 (and 8.3, after updated) running VirtualBox (Version 6.1.34 r150636 (Qt5.6.2), with Extension Pack) on Win10 (19043.1766) PC with Intel AX1650i (201NGW). 
It puzzles me for a few days as a newbie of Home Assistant and a person not very familiar with VMs and Linux. I switch to HAOS 7.6 thanks for this post, and bluetooth function works fine on my PC now.

agners, 2022-07-13:
> Is there any plan of including the fix?
> (https://github.com/home-assistant/operating-system/issues/1935#issuecomment-1136689805)

That is not a fix, that is a bug report. There are potential fixes discussed and linked there. Some of them are already part of HAOS 8.1 (e.g. `BTUSB_INTEL_BROKEN_SHUTDOWN_LED` is set for `0x8087`/`0x0a2a`), but they do not seem to fix the problem for you, unfortunately :(

Have you had a chance to try HAOS 8.2?

@fdsasdfddsa @ben0551 can you share the "Host" logs from Settings -> System -> Logs? So we can build a list of devices which seem to affected in our case.

fdsasdfddsa, 2022-07-17:
> Can you share the "Host" logs from Settings -> System -> Logs? So we can build a list of devices which seem to affected in our case.

@agners 
Host logs. For the last lines, I remove and mount the bluetooth back to the VM after its boot process finished.
<details>
<summary>Full logs</summary>

```
[    0.000000] Linux version 5.15.45 (builder@99b125bac596) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g799554151) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Wed Jun 8 23:31:44 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=425f427cf7894ebc9579b4b106380013 cgroup_enable=memory fsck.repair=yes console=ttyS0 console=tty1 rauc.slot=A
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003de38fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003de39000-0x000000003de39fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003de3a000-0x000000003de90fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003de91000-0x000000003df1afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003df1b000-0x000000003ef9afff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ef9b000-0x000000003efcafff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003efcb000-0x000000003eff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003eff3000-0x000000003effafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003effb000-0x000000003effefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003efff000-0x000000003f46afff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f46b000-0x000000003ff6afff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003ff6b000-0x000000003ffeefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ffef000-0x000000003ffeffff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: ACPI=0x3effa000 ACPI 2.0=0x3effa014 SMBIOS=0x3effd000 
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 28595001, primary cpu clock
[    0.000001] kvm-clock: using sched offset of 2469458787 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 2688.000 MHz processor
[    0.000536] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000539] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000545] last_pfn = 0x3f46b max_arch_pfn = 0x400000000
[    0.000555] Disabled
[    0.000556] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[    0.000560] CPU MTRRs all blank - virtualized system.
[    0.000563] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.000628] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.000875] Secure boot could not be determined
[    0.000878] ACPI: Early table checksum verification disabled
[    0.000881] ACPI: RSDP 0x000000003EFFA014 000024 (v02 VBOX  )
[    0.000885] ACPI: XSDT 0x000000003EFF90E8 000044 (v01 VBOX   VBOXFACP 00000001      01000013)
[    0.000891] ACPI: FACP 0x000000003EFF7000 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.000896] ACPI: DSDT 0x000000003EFF4000 002325 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.000899] ACPI: FACS 0x000000003EFFE000 000040
[    0.000902] ACPI: APIC 0x000000003EFF8000 00007C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.000905] ACPI: SSDT 0x000000003EFF3000 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.000909] ACPI: BGRT 0x000000003DE39000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.000911] ACPI: Reserving FACP table memory at [mem 0x3eff7000-0x3eff70f3]
[    0.000913] ACPI: Reserving DSDT table memory at [mem 0x3eff4000-0x3eff6324]
[    0.000914] ACPI: Reserving FACS table memory at [mem 0x3effe000-0x3effe03f]
[    0.000915] ACPI: Reserving APIC table memory at [mem 0x3eff8000-0x3eff807b]
[    0.000916] ACPI: Reserving SSDT table memory at [mem 0x3eff3000-0x3eff31cb]
[    0.000917] ACPI: Reserving BGRT table memory at [mem 0x3de39000-0x3de39037]
[    0.001135] No NUMA configuration found
[    0.001138] Faking a node at [mem 0x0000000000000000-0x000000003f46afff]
[    0.001142] NODE_DATA(0) allocated [mem 0x3f213000-0x3f217fff]
[... 716 lines cut ...]
[  148.370839] eth0: renamed from veth25381ad
[  148.378879] IPv6: ADDRCONF(NETDEV_CHANGE): veth977e3d7: link becomes ready
[  148.378909] hassio: port 5(veth977e3d7) entered blocking state
[  148.378911] hassio: port 5(veth977e3d7) entered forwarding state
[  298.556991] kauditd_printk_skb: 27 callbacks suppressed
[... cut ...]
[  553.596450] usb 2-1: New USB device found, idVendor=8087, idProduct=0026, bcdDevice= 0.02
[  553.596455] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  553.610331] Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
[  553.610405] Bluetooth: hci0: Boot Address: 0x24800
[  553.610406] Bluetooth: hci0: Firmware Version: 125-46.21
[  555.621149] Bluetooth: hci0: command 0xfc01 tx timeout
[  563.749308] Bluetooth: hci0: FW download error recovery failed (-110)

```
</details>

stalker1969, 2022-07-26:
Решилась ли данная проблема? Так же получаю ошибку (OS 8.2) 2022.7.5 Bluetooth в частности SkyKettle

stalker1969, 2022-07-26:
Has this problem been resolved? I also get an error (OS 8.2) 2022.7.5 Bluetooth in particular SkyKettle

mkanet, 2022-07-27:
> Has this problem been resolved? I also get an error (OS 8.2) 2022.7.5 Bluetooth in particular SkyKettle

I don't think so.  Still seeing the same error message after upgrading to:

Home Assistant 2022.7.7
Supervisor 2022.07.0
Operating System 8.4

**NOTE: I don't have a Bluetooth dongle installed on my Windows 10 PC (Host for Home Assistant VirtualBox VM).**  However, I started getting the same error as you recently.  I have VirtualBox running in "Bridged Mode".  Are you using Bridged Mode too or Host-Only mode?

I get the below error after each restart of Home Assistant Host:
```
22-07-26 16:14:01 WARNING (MainThread) [supervisor.host.network] Requested to update interface enp0s3 which does not exist or is disabled.
```

jask747, 2022-07-27:
Have same issue running KVM on Linux host Intel i5 NUC. Host host detects BT device and available to forward to KVM but not detected by HassOS

Home Assistant 2022.7.6
Supervisor 2022.07.0
Operating System 8.2

agners, 2022-07-27:
@mkanet 

> I am also started getting the below error after upgrading Home Assistant to Home Assistant 2022.7.5 (from an an old version from 6 months ago). I run Home Assistant in a VirtualBox VM on a Windows 10 host.
>
> 22-07-20 09:12:08 WARNING (MainThread) [supervisor.host.network] Requested to update interface enp0s3 which does not exist or is disabled.

This error is not related to Bluetooth, as it is about Ethernet interface. Also, the warning comes from the Supervisor. Besides the printed warning, are you having any issue with your system? If you have issues related to that warning open a issue with the Supervisor project: https://github.com/home-assistant/supervisor

@jask747 

Did Bluetooth work for you before? Any helpful information in dmesg (Host logs in the frontend)? How do you forward the device exactly (I assume USB?)

jask747, 2022-07-27:
> @mkanet
>
>
> This error is not related to Bluetooth, as it is about Ethernet interface. Also, the warning comes from the Supervisor. Besides the printed warning, are you having any issue with your [...]

@agners 

Hi

Onboard USB BT device  = ID 8087:0a2a Intel Corp. Bluetooth wireless interface.

This is seen by KVM which you can then select to be available for guest OS. 

Logs from HA Host logs as below (filtered on Bluetooth entries)

[    0.478339] Bluetooth: Core ver 2.22
[    0.478876] NET: Registered PF_BLUETOOTH protocol family
[    0.479216] Bluetooth: HCI device and connection manager initialized
[    0.480097] Bluetooth: HCI socket layer initialized
[    0.480217] Bluetooth: L2CAP socket layer initialized
[    0.480977] Bluetooth: SCO socket layer initialized
[    4.402356] Bluetooth: hci0: command 0x0c03 tx timeout
[   12.466244] Bluetooth: hci0: sending initial HCI reset failed (-110)

Regards

mcamou, 2022-08-29:
I am having the same problem with KVM on a NUC. The host OS is Ubuntu 22.04, and can see the Bluetooth controller. I've exported the Bluetooth controller (USB ID `8087:0a2b`) to the guest HAOS. `bluetoothctl devices` says `No default controller available`.

 Here's the output of `journalctl | grep -i blue` in the guest OS:

```
Aug 29 14:27:03 homeassistant kernel: Bluetooth: Core ver 2.22
Aug 29 14:27:03 homeassistant kernel: NET: Registered PF_BLUETOOTH protocol family
Aug 29 14:27:03 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
Aug 29 14:27:03 homeassistant kernel: Bluetooth: HCI socket layer initialized
Aug 29 14:27:03 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
Aug 29 14:27:03 homeassistant kernel: Bluetooth: SCO socket layer initialized
Aug 29 14:27:04 homeassistant systemd[1]: Mounting Bluetooth persistent state...
Aug 29 14:27:04 homeassistant systemd[1]: Mounted Bluetooth persistent state.
Aug 29 14:27:04 homeassistant systemd[1]: Starting Bluetooth service...
Aug 29 14:27:04 homeassistant bluetoothd[309]: Bluetooth daemon 5.63
Aug 29 14:27:04 homeassistant systemd[1]: Started Bluetooth service.
Aug 29 14:27:04 homeassistant systemd[1]: Reached target Bluetooth Support.
Aug 29 14:27:04 homeassistant bluetoothd[309]: Starting SDP server
Aug 29 14:27:04 homeassistant bluetoothd[309]: profiles/network/bnep.c:bnep_init() kernel lacks bnep-protocol support
Aug 29 14:27:04 homeassistant bluetoothd[309]: src/plugin.c:plugin_init() System does not support network plugin
Aug 29 14:27:04 homeassistant bluetoothd[309]: Bluetooth management interface 1.21 initialized
Aug 29 14:27:04 homeassistant dbus-daemon[117]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.7' (uid=0 pid=309 comm="/usr/libexec/bluetooth/bluetoothd ")
Aug 29 14:27:05 hass kernel: Bluetooth: hci0: command 0xfc05 tx timeout
Aug 29 14:27:05 hass kernel: Bluetooth: hci0: Reading Intel version command failed (-110)
Aug 29 14:27:44 hass ghcr.io/home-assistant/qemux86-64-homeassistant:2022.8.7/homeassistant[378]: 2022-08-29 16:27:44.285 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Bluetooth' for bluetooth integration not ready yet: Failed to start Bluetooth: adapter 'hci0' not found; Retrying in background
```

rubendediegomartinez, 2022-09-13:
It was a pitty! :-( 
I'm runnig HA OS 8.4, 8.5 (switching in the boot from one to other) in VirtualBox over Windows 11. I'm using the last versions of the rest of components.
Bluetooth doesn't run. 
I tried several adapter:
1rs- Hw computr integrated
2º On based in Realte BT 5.0
3rd. ASU BT500, one of recommended.

All the issues described in this theard hapenned to me.
It was a pitty, but in next days I will be impeled to use the EspHome BT Gw.
Best regards

closed 2023-01-11 by github-actions[bot]

cryptton2004, 2023-01-11:
/remove-lifecycle stale

ben0551, 2023-01-19:
This is still an issue

michaelwingender, 2023-03-25:
"To keep our backlog manageable we have to clean old issues, as many of them have already been resolved with the latest updates." Interesting strategy. Sound like "These are not the droids you're looking for" ;-)

The issue is not fixed.