home-assistant/operating-system #1936: Upgrade to OS 8.1 fails

csimonosian, 2022-05-21:
### Describe the issue you are experiencing

Tried to update to OS 8.1 but nothing happens, the update dialog remains on the screen. Errors appear in core and supervisor logs.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Click on Settings
2. Click on the notification Home Assistant Operating System Version 8.1 is available.
3. Clock in Install.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-05-21 10:23:55 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-8.1.raucb
22-05-21 10:23:56 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: signature verification failed: error:2E09A09E:CMS routines:CMS_SignerInfo_verify_content:verification failure
```

### Anything in the Host logs that might be useful for us?

```txt
No errors in the Host log, but this is in Core log:

2022-05-21 10:23:56 ERROR (MainThread) [homeassistant.core] Error executing service: <ServiceCall update.install (c:0180e5edcedb3acf78505bfe345143fa): entity_id=['update.home_assistant_operating_system_update'], version=8.1, backup=False>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 47, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1645, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1664, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 152, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 411, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: Unknown error, see supervisor
```

### System Health information

## System Health

version | core-2022.5.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/London

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.2
docker_version | 20.10.9
disk_total | 31.3 GB
disk_used | 2.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.6.1), Z-Wave JS (0.1.58)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 1
mode | storage

</details>

bobloadmire, 2022-05-21:
just pilling on to say same here on my rpi4. upgrade 7.6>8.0 failed, rolled back to 7.6, 7.6>8.1 also failed.

zagnuts, 2022-05-21:
...and same here - failed with upgrade from 7.6 to 8.0 and now 8.1. Standard rpi3 booting from good quality sd card with no errors. I think I might end up doing a fresh install and restoring from backup.

Host Operating System | Home Assistant OS 7.6
-- | --
Update Channel | stable
Supervisor Version | supervisor-2022.05.2
Docker Version | 20.10.9
Disk Total | 28.6 GB
Disk Used | 9.1 GB
Healthy | true
Supported | true
Board | rpi3
Supervisor API | ok
Version API | ok
Installed Add-ons | File editor (5.3.3), Samba share (9.6.1), Terminal & SSH (9.4.0), deCONZ (6.13.0), ESPHome (2022.5.0), Home Assistant Google Drive Backup (0.107.2)

brylee123, 2022-05-22:
I have the same issue. When the HA CLI loads it only outputs `Error returned from Supervisor: System is not ready with state: setup`

When I run `supervisor logs` i get: `{"result": "error", "message": "System is not ready with state: setup"}`

agners, 2022-05-23:
@bobloadmire

> just pilling on to say same here on my rpi4. upgrade 7.6>8.0 failed, rolled back to 7.6, 7.6>8.1 also failed.

@zagnuts

> ...and same here - failed with upgrade from 7.6 to 8.0 and now 8.1. Standard rpi3 booting from good quality sd card with no errors. 

Generally I tested upgrade on all machines, and it did work in my test setup, so there is no "generic issue". Hence the question: Do you have the same error message as the original poster when checking `ha su logs` (sepecifically `Home Assistant Operating System update failed with: signature verification failed`)? If so, please confirm, if you receive different errors, please check if that matches a different issue or create a new report.

@brylee123 you are having a different issue then. It seems the system did not completely boot yet? Can you wait for a while and try again? If the problem persists, please open a new issue.

agners, 2022-05-23:
@csimonosian that sounds like there was a problem during download of the update. Do you use a proxy or run in a otherwise firewalled/restricted environment? Is this reproducible (e.g if you try again, do you get the same error message?)

agners, 2022-05-23:
> I think download-type problems have occurred before when a new release has only just been announced, perhaps things haven't yet propagated!

Yeah that might be, we use GitHub releases and it certainly uses some type of CDN.

Thanks for retesting, closing the issue then.

closed 2022-05-23 by agners

daveholst, 2022-05-25:
Confirming that I had a similar problem upgrading to OS 8.1 running in a VM a few days ago. I have just run it again this morning and it is working fine! 👍

jaydhary14, 2022-05-25:
2022-05-25 15:25:46 ERROR (MainThread) [homeassistant.core] Error executing service: <ServiceCall update.install (c:0180fce43078c565a8f8dbd6269a3f79): entity_id=['update.home_assistant_operating_system_update'], version=8.1, backup=False>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 47, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/8.1/haos_rpi4-64-8.1.raucb: Response payload is not completed
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1645, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1664, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 152, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 411, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/8.1/haos_rpi4-64-8.1.raucb: Response payload is not completed