home-assistant/operating-system #1940: mdns-repeater causes high cpu usage

gomble, 2022-05-24:
### Describe the issue you are experiencing

Hello in my opinion mdns-repeater causes a high cpu usage:

`# ps -aux --sort=-pcpu
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root     1177405 20.9  1.0 888024 88104 ?        Sl   May23 122:19 frigate.detector.cpu1
root     1177406 20.9  1.0 888032 87684 ?        Sl   May23 122:13 frigate.detector.cpu2
root        2600 10.6  0.0    864     4 ?        Ss   May21 512:48 mdns-repeater -f hassio
root     1173260  9.4  8.2 2882532 683368 ?      Ssl  May23  56:19 python3 -m homeassistant --config /config
root     1177374  3.9  1.8 2478948 155072 ?      Sl   May23  23:06 python3 -u -m frigate
root     1177461  2.8  0.3 489660 31280 ?        Rsl  May23  16:33 ffmpeg -hide_banner -loglevel warning -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format yuv420p -avoid_negative_ts make_zero -fflags +genpts+disca
systemd+     338  2.3  0.1  15096 10092 ?        Ss   May21 111:15 /usr/lib/systemd/systemd-resolved
root     1177471  1.8  0.2 113568 24512 ?        Ssl  May23  10:48 ffmpeg -hide_banner -loglevel warning -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format yuv420p -avoid_negative_ts make_zero -fflags +genpts+disca
root        8026  1.6  1.1 337772 97748 ?        Sl   May21  77:49 node index.js
`

Is this normal ? 10% CPU Usage ? 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

After a a reboot and a while the cpu goes up
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:28:38 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Unknown option 'include_device_information' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:33:40 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:38:42 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-05-24 10:38:42 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:38:42 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-05-24 10:38:42 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt (45df7312_zigbee2mqtt)
[... 90 lines cut ...]
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] corefile.sh: executing... 
[cont-init.d] corefile.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
.:53
.:5553
CoreDNS-1.8.4
linux/amd64, go1.15.15, 053c4d5-dirty
[INFO] 172.30.32.1:37576 - 1246 "A IN ccab4aaf-frigate.local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000074656s
[INFO] 172.30.32.1:40416 - 13554 "A IN ccab4aaf-frigate.local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000111159s
[INFO] 127.0.0.1:49962 - 34253 "PTR IN 0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.ip6.arpa. udp 101 true 2048" NXDOMAIN qr,rd,ra 90 0.037337614s
[INFO] 172.30.33.5:55698 - 34253 "PTR IN 0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.ip6.arpa. udp 90 false 512" NXDOMAIN qr,rd,ra 90 0.060676757s
[INFO] 172.30.32.1:56466 - 29392 "A IN www.waze.com. udp 30 false 512" NOERROR - 0 12.001389949s
[ERROR] plugin/errors: 2 www.waze.com. A: read udp 127.0.0.1:36716->127.0.0.1:5553: i/o timeout
[INFO] 127.0.0.1:59887 - 29392 "A IN www.waze.com. udp 41 true 2048" NOERROR - 0 6.116264682s
[ERROR] plugin/errors: 2 www.waze.com. A: read tcp 172.30.32.3:49320->1.1.1.1:853: i/o timeout
[INFO] 127.0.0.1:42803 - 29555 "AAAA IN www.waze.com. udp 41 true 2048" NOERROR - 0 11.540143388s
[ERROR] plugin/errors: 2 www.waze.com. AAAA: read tcp 172.30.32.3:49334->1.1.1.1:853: read: connection reset by peer
[INFO] 127.0.0.1:59526 - 40291 "AAAA IN www.waze.com. udp 41 true 2048" NOERROR - 0 5.5146262539999995s
[ERROR] plugin/errors: 2 www.waze.com. AAAA: dial tcp 1.0.0.1:853: i/o timeout
[INFO] 127.0.0.1:50081 - 40144 "A IN www.waze.com. udp 41 true 2048" NOERROR - 0 11.469415155s
[ERROR] plugin/errors: 2 www.waze.com. A: dial tcp 1.1.1.1:853: i/o timeout`

gomble, 2022-05-24:
After Reboot of Hassos:

[image]

agners, 2022-05-24:
That seems like an excessive CPU load indeed.

Are you using a router with mDNS reflector and/or VLANs?

closed 2022-08-29 by github-actions[bot]