home-assistant/operating-system #1943: The "update.install" service does not update Home Assistant Operating system

sdschramm, 2022-05-26:
### Describe the issue you are experiencing

I created an automation to automatically install updates using the "update.install" service. It works for core updates but not the operating system updates.

Other discussion here: 
https://community.home-assistant.io/t/auto-update-feature-for-core-supervisor/412783/11

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Create a new automation 
2. Call the service "update.install" and select Home Assistant Operating system as the target
3. Trigger the automation 

### Anything in the Supervisor logs that might be useful for us?

```txt
I don't think so
```

### Anything in the Host logs that might be useful for us?

```txt
I don't think so
```

sdschramm, 2022-05-27:
I ended up figuring this out. Having the "Backup" option enabled causes it to fail. After disabling the option to take a backup before the update fixed the issue and caused it to run. This obviously isn't a good idea so a work around is to use the Google Drive addon and use that to schedule a backup before the update service is called. In my case I have a backup set to run every morning at 3 AM and then an automation to run the update service every day at 4 AM. If I wake up and find my system hosed, I can download the backup from Google Drive, format my Pi, and restore from the backup taken right before the upgrade. It would be nice if the built in backup option was supported but at least for now I found a workaround.

agners, 2022-05-27:
Hm, I remember we used to have a similar problem when triggering OS update from the frontend.

@ludeeus thoughts?

ludeeus, 2022-05-27:
The service call is supposed to fail if the entity does not support backup and backup is passed as an option.
There are no OS backup.

agners, 2022-05-27:
@sdschramm you can use the `hassio.backup_full` service to create a backup in your automation before triggering the update.

closed 2022-05-27 by agners