home-assistant/operating-system #1948: Multicast DNS stopped working after upgrading to HassOS 8.0 or 8.1

mpepping, 2022-05-29:
### Describe the issue you are experiencing

After upgrading to HassOS 8.1 (or even 8.0) Multicast DNS resolving stopped working for the HassOS host and all containers. Regular DNS resolving works as supposed, but resolving hosts like `host.local` or `host` or `host.local.` doesn't work. Resolving multicast DNS  doesn't work from the host or any container. 

No issues from any other hosts, or via `dig` on local nameservers.

The DNS log floods with:

```
[INFO] 172.30.32.1:53889 - 43511 "A IN airco-living.local.local.hass.io. udp 53 false 512" NXDOMAIN qr,aa,rd 53 0.000168785s
[INFO] 172.30.32.1:54266 - 25043 "A IN doorbell.local.local.hass.io. udp 46 false 512" NXDOMAIN qr,aa,rd 46 0.000244608s
[INFO] 172.30.32.1:33001 - 17575 "A IN descaler.local.local.hass.io. udp 47 false 512" NXDOMAIN qr,aa,rd 47 0.000273108s
[..]
```

The multicast service is online:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
mdns-repeater (6): dev hassio addr 172.30.32.1 mask 255.255.254.0 net 172.30.32.0
```

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade 7.x to 8.1
2. Run `ping somehost.local` and `ping somehost` from the HassOS console or SSH container.
3. No IP-address is returned

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System Health information

Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1
Kernel version 5.15.41
Agent version 1.2.1

### Additional information

[image]

agners, 2022-05-30:
Are you sure this is Home Assistant OS related and not caused by an update of another component? Can you try downgrade to HAOS 7.6 to see if that makes it indeed work again?

```
ha os update --version 7.6
```

agners, 2022-05-30:
Hm, can you check the RAUC logs?
```
journalctl -u rauc.service
```

mpepping, 2022-05-31:
@agners, downgrading from 8.1 to 8.0 went well. Downgrading 8.0 to 7.6 throws the same error. Here's the log from the 8.1 to 7.6 downgrade:

```text
May 31 18:51:20 homeassistant rauc[457]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
May 31 18:51:20 homeassistant rauc[457]: input bundle: /mnt/data/supervisor/tmp/hassos-7.6.raucb
May 31 18:51:20 homeassistant rauc[457]: Active slot bootname: B
May 31 18:51:20 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: started
May 31 18:51:20 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Checking and mounting bundle...
May 31 18:51:20 homeassistant rauc[457]: Reading bundle: /mnt/data/supervisor/tmp/hassos-7.6.raucb
May 31 18:51:20 homeassistant rauc[457]: Detected CRL but CRL checking is disabled!
May 31 18:51:20 homeassistant rauc[457]: Verifying bundle signature...
May 31 18:51:21 homeassistant rauc[457]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
May 31 18:51:21 homeassistant rauc[457]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-7.6.raucb' to '/run/rauc/bundle'
May 31 18:51:21 homeassistant rauc[457]: Configured loop device '/dev/loop0' for 186961920 bytes
May 31 18:51:21 homeassistant rauc[457]: Running bundle hook install-check
May 31 18:51:21 homeassistant rauc[457]: Marking target slot kernel.0 as non-bootable...
May 31 18:51:21 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Updating slots...
May 31 18:51:21 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Checking slot boot.0
May 31 18:51:21 homeassistant rauc[457]: Updating /dev/disk/by-partlabel/hassos-boot with 'install' slot hook
May 31 18:51:21 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Updating slot boot.0
May 31 18:51:21 homeassistant rauc[457]: Running slot hook 'slot-install' for boot.0
May 31 18:51:22 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Updating slot boot.0 status
May 31 18:51:22 homeassistant rauc[457]: Checking image type for slot type: raw
May 31 18:51:22 homeassistant rauc[457]: Image detected as type: *.img
May 31 18:51:22 homeassistant rauc[457]: Updating /dev/disk/by-partlabel/hassos-kernel0 with /run/rauc/bundle/kernel.img
May 31 18:51:22 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Updating slot boot.0 done
May 31 18:51:22 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Checking slot kernel.0
May 31 18:51:22 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Updating slot kernel.0
May 31 18:51:22 homeassistant rauc[457]: opening slot device /dev/disk/by-partlabel/hassos-kernel0
May 31 18:51:22 homeassistant rauc[457]: writing data to device /dev/disk/by-partlabel/hassos-kernel0
May 31 18:51:22 homeassistant rauc[457]: Running slot hook 'slot-post-install' for kernel.0
May 31 18:51:22 homeassistant rauc[48196]: blkdiscard: /dev/disk/by-partuuid/33236519-7f32-4dff-8002-3390b62c309d: BLKDISCARD ioctl failed: Operation not supported
May 31 18:51:22 homeassistant rauc[457]: Installation error: Failed updating slot kernel.0: failed to run slot hook: Child process exited with code 1
May 31 18:51:22 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: Installation error: Failed updating slot kernel.0: failed to run slot hook: Child process exited with code 1
May 31 18:51:22 homeassistant rauc[457]: installing /mnt/data/supervisor/tmp/hassos-7.6.raucb: finished
May 31 18:51:22 homeassistant rauc[457]: installing `/mnt/data/supervisor/tmp/hassos-7.6.raucb` failed: 1
```

Resolving `.local` addresses from HassOS or a container remains an issue after downgrading to 8.0.

agners, 2022-05-31:
> May 31 18:51:22 homeassistant rauc[48196]: blkdiscard: /dev/disk/by-partuuid/33236519-7f32-4dff-8002-3390b62c309d: BLKDISCARD ioctl failed: Operation not supported

Hm, it seems rauc uses blkdiscard but can't? Haven't seen that issue. What type of VM do you use and what kind of virtual disk?

mpepping, 2022-05-31:
It's a VMware ESXi VM, with a thick provisioned SATA disk. It indeed does not support TRIM:

```bash
 cat /sys/block/sda/queue/discard_granularity
0
```

agners, 2022-06-01:
It seems to fail when running our post install hook. The hook is not doing very much, my best guess is that trying to mount the boot partition fails for some reason. Can you try these commands manually?

```
systemctl start mnt-boot.mount
systemctl status mnt-boot.mount
```

mpepping, 2022-06-01:
No issue with that:

```
● mnt-boot.mount - HassOS boot partition
     Loaded: loaded (/usr/lib/systemd/system/mnt-boot.mount; enabled; vendor preset: enabled)
     Active: active (mounted) since Tue 2022-05-31 19:05:11 UTC; 24h ago
      Until: Tue 2022-05-31 19:05:11 UTC; 24h ago
      Where: /mnt/boot
       What: /dev/sda1
      Tasks: 0 (limit: 7119)
     Memory: 36.0K
     CGroup: /system.slice/mnt-boot.mount

May 31 19:05:11 homeassistant systemd[1]: Mounting HassOS boot partition...
May 31 19:05:11 homeassistant systemd[1]: Mounted HassOS boot partition.
```

Doesn't finalise the downgrade either after a reboot.

agners, 2022-06-01:
That is super weird. I am honestly out of idea. This is the script which rauc claims failing, but I see not a single line which remotely could use blkdiscard (other than the two posted, which we ruled out):

https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/ota/rauc-hook#L59-L74

mpepping, 2022-07-17:
OK, figured this out eventually. Apparently, upgrading my Unify USG router upgraded in the same time-frame, and broke it's mdns forwarding config. This service handles mdns multicasts between interfaces/vlan's. Reconfiguring that resolved the issue. 
The lesson learned is disabling automatic software updates on network components :) 

@agners thank you for troubleshooting

closed 2022-07-17 by mpepping