home-assistant/operating-system #1949: HassOS Hanging on Boot then no DHCP

neil1111, 2022-05-29:
### Describe the issue you are experiencing

When booting in RPI 4b 4GB, it hangs as described here: 
[](https://community.home-assistant.io/t/hassos-hanging-on-boot-stuck-at-wait-until-kernel-time-synchronized/207921)

Wasn’t able to reboot from the app or ssh to reboot, so I did a hard reset. Now during the boot process I’m stuck on a seemingly infinite wait for a start job to finish.

A start job is running for Wait Until Kernel Time Synchronized (1h 58min 55s / no limit)

Eventually did the solution described here: [](https://community.home-assistant.io/t/hassos-hanging-on-boot-stuck-at-wait-until-kernel-time-synchronized/207921/16?) to get DHCP assigned. Also needed to SAVE and then ACTIVATE.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Cold Start RPi (I haven't tested it via simple restart, but happy to test this if it would be helpful)
2. Monitor the launch with a connected screen & keyboard
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Not sure, but happy to provide. There are no Errors, but there are several warnings that state: 
22-05-29 07:58:50 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session
```

### Anything in the Host logs that might be useful for us?

```txt
[    5.093133] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
```

neil1111, 2022-05-29:
I'm running on an SSD.

zamog, 2022-06-01:
Some here - to reproduce it make sure the DHCP server is offline (and than cold start the PI)

agners, 2022-06-01:
So it seems that the network configuration got changed to disabled in all cases? The Supervisor handles the network configuration, so this is likely caused by the Supervisor.

neil1111, 2022-06-01:
> Some here - to reproduce it make sure the DHCP server is offline (and then cold start the PI)

In my case, each time it occurred following a cold start (power cycle).

> So it seems that the network configuration got changed to disabled in all cases? The Supervisor handles the network configuration, so this is likely caused by the Supervisor.
This highlights a good point. It is possible (and I will test), that one of the instances when this didn't work I had power cycled the power strip into which both the router (to which the RPi is connected and provides DHCP) and the RPi are plugged. That could mean that DHCP wasn't running when the HA looked for the Network, and then ran out of time (I think it was a 90 second count-down).

Perhaps the solution is, in the absence of an IP (DHCP, a disconnection, or otherwise)  for Supervisor to try again every (say) 10 seconds.

neil1111, 2022-06-07:
Hi everyone. I (unfortunately) encountered another use case for this problem. I encountered hardware problems with my router, and as a result had no ability for the RPi to get an IP address. Therefore, I couldn't even use the terminal to "force" it to look for it again. HA simply would not run.  

While in that situation, naturally there would be no Alexa, Google Home, or Nabu Casa connectivity, I VERY MUCH could have used ZWave to be up & running with the related Node-Red functionality running under HA.  Many of my lights only use logical switches (not connected as a "load"), so it is important for HA to run without DHCP, much less external IP access.

agners, 2022-06-08:
Just to be clear on how the system should behave in absence of network (e.g. DHCP server not reachable):

- The network will continue to try to wait for network to become online and try to synchronize network time up to 90s
- If that fails, the system will continue to boot regularly.

Any other behavior is unexpected and needs investigation.

> A start job is running for Wait Until Kernel Time Synchronized (1h 58min 55s / no limit)

It definitely should not wait for that long. HAOS in my virtual test environment has a ~90s timeout on that (`xx / 1min 29s`). Once that counted down it continues booting.

On the Raspberry Pi 4 installation the same service also has a timeout of 90s configured (from the OS shell):

```
# systemctl show systemd-time-wait-sync.service | grep Timeout
TimeoutStartUSec=1min 30s
```

What some might be seeing is a related issue where Supervisor causes network to remain disabled after a DHCP outage (see https://github.com/home-assistant/supervisor/issues/3674).

neil1111, 2022-06-09:
> home-assistant/supervisor#3674

Hi Stefan,

Thank you for the clear summary of expected behavior. I can confirm that
following the ~90 second timeout in the absence of DHCP, the system
continues to hang, and I needed to use the HA commands referenced in the
"workaround" linked in my OP.  I hope that clarifies.

Sesshoumaru-sama, 2022-06-20:
I had the same issue today. After reboot of my HA instance it did not get an IP address anymore.
Turned out that the entire ehternet connection had been dropped. 
As I run HA on Promox I was able to get to the CLI and add a new connection.

nmcli con edit
activate

Please investiagate. Got me a scare!

closed 2022-09-25 by github-actions[bot]

stuaxo, 2023-10-04:
Reproduced this, waited for 17m and turned off, will try re flashing this is just so I can evaluate the OS

c11umw, 2023-12-31:
> I had the same issue today. After reboot of my HA instance it did not get an IP address anymore. Turned out that the entire ehternet connection had been dropped. As I run HA on Promox I was able to [...]

Interesting that it still affected you even using Proxmox.
I've not seen the issue in a while - probably 'cos I now run multiple PiHoles across my network and they're all DHCP enabled (split the shared IPs between them to prevent conflics). One of them is in Proxmox next to HomeAssistant, so even if the local switch is offline then hopefully the Proxmox one should still be able to manage the DHCP request.

lakaassekoo, 2024-01-12:
Yes, same problem today with fresh install virtual box with "haos_ova-11.4.vdi"

TommiG1, 2024-02-03:
I had the same problem today after making several attempts to migrate a HomeAssistant vm to Unraid.
At some point I ran into the error that DHCP was no longer possible and the host didn't get an IP.
The reason was that there were several identical host names (homeassistant) with different MAC addresses on the DHCP (Fritz! Box). With the 6th doublet, the DHCP service on the router stopped.
The solution was to delete all hostname duplicates and generate a new MAC address for the vm.

Sim1Chin, 2024-02-17:
Experienced this same issue with HA running on VirtualBox VM on a NUC. I've configured my mesh (TPLink Deco) to reboot nightly and would notice that my HA instance becomes inaccessible over the network in the morning. I've also attempted to configure an automation that would trigger once my mesh has rebooted that would reboot the HA host. But this still doesn't allow HA to come back online and I can only access HA locally on the NUC the VM is running on. I have to resort to shutting down the VM running HA and starting it back up to be able to access across the network. As this can become tedious every day, I've now resorted to disabling the scheduled reboots on my mesh to maintain connectivity to HA.

Billybangleballs, 2024-04-02:
I would like to be able to install home-assistant, but because my network does not use dhcp (intentionally), and there is no facility for manually assigning a network configuration, it seems impossible to do.

Impact123, 2024-04-02:
Give this a read: https://github.com/home-assistant/operating-system/blob/dev/Documentation/network.md

Billybangleballs, 2024-04-02:
> Give this a read: https://github.com/home-assistant/operating-system/blob/dev/Documentation/network.md

That did the trick, it ought to be in the installation instructions so people can find it ;)