home-assistant/operating-system #1963: Failed to update from 8 to 8.1

AlRiachi, 2022-06-02:
### Describe the issue you are experiencing

I'm unable to update from 8 to 8.1 the supervisor log shows  this error

> 22-06-02 12:02:53 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Click on install OS 8.1
2. Update fails 
3. Supervisor log shows

> 22-06-02 12:02:53 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy

...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-06-02 12:02:53 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy
```

### Anything in the Host logs that might be useful for us?

```txt
NA
```

### System Health information

## System Health

version | core-2022.6.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | Asia/Amman

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.25.5
Stage | running
Available Repositories | 1115
Downloaded Repositories | 26

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | June 22, 2022, 3:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-1.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 237.7 GB
disk_used | 39.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Mosquitto broker (6.1.2), MariaDB (2.4.0), FTP (4.5.0), InfluxDB (4.4.1), Grafana (7.5.2), AppDaemon (0.8.2), Home Assistant Google Drive Backup (0.105.2), AirCast (3.3.0), Terminal & SSH (9.4.0), Studio Code Server (5.0.5), Duck DNS (1.15.0), deCONZ (6.14.1), Zigbee2mqtt (1.25.2-1), Node-RED (11.1.2), Tailscale (0.6.1), SSH & Web Terminal (10.1.3), Glances (0.15.0), Samba NAS (9.5.1-nas4), phpMyAdmin (0.7.1), Samba share (9.6.1), Filebrowser (2.21.1-2), Portainer (2.13.1), iPerf3 (0.1.0), Samba Backup (5.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 8
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 23, 2022, 8:47 AM
-- | --
current_recorder_run | June 2, 2022, 11:56 AM
estimated_db_size | 2112.80 MiB
database_engine | mysql
database_version | 10.4.19

</details>

agners, 2022-06-02:
Do you use SSD boot?

Can you check the rauc logs using

```
journalctl -u rauc.service
```

(either on the console or via ssh port 22222)

AlRiachi, 2022-06-02:
Yes, it is an SSD boot

> Jun 01 20:09:30 homeassistant systemd[1]: Starting Rauc Update Service...
Jun 01 20:09:30 homeassistant rauc[449]: Using central status file /mnt/data/rauc.db
Jun 01 20:09:30 homeassistant rauc[449]: Using central status file /mnt/data/rauc.db
Jun 01 20:09:30 homeassistant systemd[1]: Started Rauc Update Service.
Jun 01 20:11:36 homeassistant rauc[449]: rauc mark: marked slot kernel.1 as good
Jun 02 09:02:48 homeassistant rauc[449]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Jun 02 09:02:48 homeassistant rauc[449]: input bundle: /mnt/data/supervisor/tmp/hassos-8.1.raucb
Jun 02 09:02:48 homeassistant rauc[449]: Active slot bootname: B
Jun 02 09:02:48 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: started
Jun 02 09:02:48 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking and mounting bundle...
Jun 02 09:02:48 homeassistant rauc[449]: Reading bundle: /mnt/data/supervisor/tmp/hassos-8.1.raucb
Jun 02 09:02:48 homeassistant rauc[449]: Detected CRL but CRL checking is disabled!
Jun 02 09:02:48 homeassistant rauc[449]: Verifying bundle signature...
Jun 02 09:02:49 homeassistant rauc[449]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
Jun 02 09:02:49 homeassistant rauc[449]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-8.1.raucb' to '/run/rauc/bundle'
Jun 02 09:02:49 homeassistant rauc[449]: Configured loop device '/dev/loop0' for 120397824 bytes
Jun 02 09:02:49 homeassistant rauc[449]: Running bundle hook install-check
Jun 02 09:02:49 homeassistant rauc[449]: Marking target slot kernel.0 as non-bootable...
Jun 02 09:02:49 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slots...
Jun 02 09:02:49 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking slot boot.0
Jun 02 09:02:49 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot boot.0
Jun 02 09:02:49 homeassistant rauc[449]: Updating /dev/disk/by-partlabel/hassos-boot with 'install' slot hook
Jun 02 09:02:49 homeassistant rauc[449]: Running slot hook 'slot-install' for boot.0
Jun 02 09:02:53 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot boot.0 status
Jun 02 09:02:53 homeassistant rauc[449]: Checking image type for slot type: raw
Jun 02 09:02:53 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot boot.0 done
Jun 02 09:02:53 homeassistant rauc[449]: Image detected as type: *.img
Jun 02 09:02:53 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking slot kernel.0
Jun 02 09:02:53 homeassistant rauc[449]: Updating /dev/disk/by-partlabel/hassos-kernel0 with /run/rauc/bundle/kernel.img
Jun 02 09:02:53 homeassistant rauc[449]: opening slot device /dev/disk/by-partlabel/hassos-kernel0
Jun 02 09:02:53 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot kernel.0
Jun 02 09:02:53 homeassistant rauc[449]: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy
Jun 02 09:02:53 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy
Jun 02 09:02:53 homeassistant rauc[449]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: finished
Jun 02 09:02:53 homeassistant rauc[449]: installing `/mnt/data/supervisor/tmp/hassos-8.1.raucb` failed: 1

closed 2022-09-07 by github-actions[bot]