home-assistant/operating-system #1964: Increase UDP Receive Buffer Size

brenner-tobias, 2022-06-04:
### Describe the issue you are experiencing

Since the cloudflared tunnel connection default protocol changed from http2 to quic, an error is occuring that is linked to the UDP Receive Buffer Size: `failed to sufficiently increase receive buffer size (was: 208 kiB, wanted: 2048 kiB, got: 58 kiB).`

Now looking into quick, it is recommended to increase the UDP buffer size by running (see [here](https://github.com/lucas-clemente/quic-go/wiki/UDP-Receive-Buffer-Size)):

`sysctl -w net.core.rmem_max=2500000`

We can do that on our local installations, but the setting is not persisted after reboots. Therfor, I suggest to generally increase this buffer size.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install [Cloudflared add-on](https://github.com/brenner-tobias/addon-cloudflared)
2. Configure it with the Cloudflare managed Tunnel
3. Check the add-on logs

### Anything in the Supervisor logs that might be useful for us?

```txt
2022/04/29 18:30:50 failed to sufficiently increase receive buffer size (was: 208 kiB, wanted: 2048 kiB, got: 58 kiB). See https://github.com/lucas-clemente/quic-go/wiki/UDP-Receive-Buffer-Size for details.
```

### Anything in the Host logs that might be useful for us?

```txt
none
```

### Additional information

Issue is also addressed in detail in the [add-on repository](https://github.com/brenner-tobias/addon-cloudflared/issues/86).

michaelhart, 2022-06-07:
This is also impacting me with a fresh install of Home Assistant OS on macOS in Virtual Box. Running `sysctl -w net.core.rmem_max=2500000` produces no results, error: 
```
sysctl: error: 'net.core.rmem_max' is an unknown key
```
Any advice to unblock using [addon-cloudflared](https://github.com/brenner-tobias/addon-cloudflared)?

agners, 2022-06-08:
> Now looking into quick, it is recommended to increase the UDP buffer size by running (see [here](https://github.com/lucas-clemente/quic-go/wiki/UDP-Receive-Buffer-Size)):
> 
> sysctl -w net.core.rmem_max=2500000

Sounds fair. I wonder if we can apply this static value across all type of machines.

elcajon, 2022-06-08:
> Sounds fair. I wonder if we can apply this static value across all type of machines.

I would appreciate it if the value for other machine types could be adjusted as well. I have the same problem (because of the same add-on) with the .qcow2 image in Proxmox.

agners, 2022-06-08:
Hm, is that issue fatal? As in, the add-on won't start/run if it can't increase the buffer size? That seems rather harsh to me :cold_sweat:

agners, 2022-06-08:
FWIW,  should be rather safe to increase the maximum window size for all platforms, see #1968. I intend to merge that PR soon.

elcajon, 2022-06-08:
> Hm, is that issue fatal? As in, the add-on won't start/run if it can't increase the buffer size? That seems rather harsh to me 😰

In case of the cloudflare tunnel add-on the add-on produces only some annoying error messages but everything still works. So I would classify the change as nice to have. 😉

brenner-tobias, 2022-09-07:
@agners Any Updates on the implementation / PR?

agners, 2022-09-07:
See linked PR #1968, it should be increased with HAOS 9.0.

closed 2022-09-07 by agners

brenner-tobias, 2022-09-07:
Thanks a lot, looking forward to the release