home-assistant/operating-system #1965: Strange restart of container homeassistant

antoweb, 2022-06-05:
### Describe the issue you are experiencing

I expereinced some random restart of container. 

And docker logs
[06:57:10] INFO: Home Assistant Core finish process exit code 256
[06:57:10] INFO: Home Assistant Core finish process received signal 11

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

2022.6.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-06-05 08:57:32 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-06-05 08:57:32 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 08:57:32 WARNING (MainThread) [supervisor.addons.options] Option 'quiet_logs' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 08:58:37 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 08:58:37 WARNING (MainThread) [supervisor.addons.options] Option 'quiet_logs' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 09:00:32 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-05 09:02:52 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-06-05 09:02:52 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 09:02:52 WARNING (MainThread) [supervisor.addons.options] Option 'quiet_logs' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 09:07:55 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-06-05 09:07:55 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
22-06-05 09:07:55 WARNING (MainThread) [supervisor.addons.options] Option 'quiet_logs' does not exist in the schema for Mosquitto broker (core_mosquitto)
```

### Anything in the Host logs that might be useful for us?

```txt
[Sun Jun  5 06:57:20 2022] audit: type=1701 audit(1654412229.834:1711): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==unconfined pid=1124424 comm="python3" exe="/usr/local/bin/python3.9" sig=11 res=1
```

agners, 2022-06-08:
Signal 11 means `SIGSEGV`, which means memory access error. Than can happen due to software bugs (in C code) or hardware issues. Do you use custom components? If so, try disabling them to see if that helps. Otherwise using different hardware might help.

In any case, this is very unlikely an OS issue.

antoweb, 2022-06-08:
Ok thanks but if i have hardware issue the raspberry should be restart itself and not only the container. Right?

agners, 2022-06-08:
If the Core is the main work load, it can be that it happens mostly with the Core. But yeah, it probably would be a bit more random. If its always only the Core, I'd try to figure out what specific (custom) integration is causing it.

closed 2022-06-08 by github-actions[bot]