home-assistant/operating-system #1966: HA OS 8.1 High CPU speed

tomlut, 2022-06-05:
### Describe the issue you are experiencing

Since updating to HA OS 8,1 from OS 7.6 my CPU speed no longer idles:

[image]

This is not affecting the system in any way (temps are still good), just thought it was odd.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Observe CPU speed before upgrading to OS 8.1
2. Observe CPU speed after upgrading to OS 8.1
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-06-06 07:22:56 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-06-06 07:22:56 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-06-06 07:22:56 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-06-06 07:22:56 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-06-06 07:22:57 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-06 07:22:57 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-06-06 07:22:57 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-06-06 07:24:21 INFO (MainThread) [supervisor.backups.manager] Found 7 backup files
22-06-06 07:24:22 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-06-06 07:24:27 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-06 07:54:27 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-06 08:20:26 INFO (SyncWorker_0) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/generic-x86-64-homeassistant
22-06-06 08:20:35 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
22-06-06 08:20:37 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-06-06 08:20:37 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-06 08:20:37 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-06-06 08:20:37 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-06-06 08:20:37 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/hassio repository
22-06-06 08:20:37 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
22-06-06 08:20:37 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-06-06 08:20:37 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/alexbelgium/hassio-addons repository
22-06-06 08:20:38 INFO (MainThread) [supervisor.store] Loading add-ons from store: 134 all - 0 new - 0 remove
22-06-06 08:20:38 INFO (MainThread) [supervisor.store] Loading add-ons from store: 134 all - 0 new - 0 remove
22-06-06 08:21:41 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-06-06 08:22:57 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-06-06 08:24:28 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-06 08:24:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
22-06-06 08:24:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/alexbelgium/hassio-addons repository
22-06-06 08:24:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
[... 219 lines cut ...]
-- | --
resources | 24
views | 53
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 30 May 2022, 05:34
-- | --
current_recorder_run | 6 June 2022, 08:20
estimated_db_size | 1187.83 MiB
database_engine | mysql
database_version | 10.4.19

</details>

### Additional information

CPU use does not seem to have changed:

[image]

[image]

agners, 2022-06-08:
How do you measure CPU speed? It seems that you have a new baseline, but CPU speed quite often goes lower than that?

It seems a bit odd indeed, I am not aware of an intentional change from HAOS side. My best guess is that Linux' CPU frequency driver changed its default behavior between 5.10 and 5.15.

tomlut, 2022-06-08:
With this core integration: <https://www.home-assistant.io/integrations/cpuspeed>

agners, 2022-06-08:
Seems that the integration uses py-sysinfo, which in turn has a hole slew of mechanism how it determines CPU frequency (see https://github.com/workhorsy/py-cpuinfo/blob/master/cpuinfo/cpuinfo.py#L2637). It could be that the OS upgrade caused another method to be used. 

> This is not affecting the system in any way (temps are still good), just thought it was odd.

Since its not affecting the system negatively, I rather prefer to save the effort to figure out what exactly changed, and just assume its fine. :sweat_smile:

tomlut, 2022-06-08:
Fair enough.