home-assistant/operating-system #1969: [WARN] Home Assistant CLI not starting! Jump into emergency console

andreimoraru, 2022-06-09:
### Describe the issue you are experiencing

After reboot of Home Assistant virtual machine running on VMware Workstation, I get the messages at the console:

```
Waiting for the Home Assistant CLI to be ready...
[WARN] Home Assistant CLI not starting! Jump into emergency console...
```

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Shutdown or reboot gracefully the Home Assistant virtual machine
2. Power it up
3. Get to the console of the Home Assistant virtual machine
4. In the console see the message:
```console
Waiting for the Home Assistant CLI to be ready...
[WARN] Home Assistant CLI not starting! Jump into emergency console...
```
https://ibb.co/Qczsr1y

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't get into the user interface of Home Assistant.
https://ibb.co/D8gLztH
```

### Anything in the Host logs that might be useful for us?

```txt
Can't get into the user interface of Home Assistant.
https://ibb.co/D8gLztH
```

### System Health information

Can't get into the user interface of Home Assistant.
https://ibb.co/D8gLztH

Aleksandar-bit, 2022-06-16:
Hi buddy, I had the same problem, just enter "**login**" in the console and press **enter**, your problem will be solved. I found a solution in the comments of this post https://community.home-assistant.io/t/warn-home-assistant-cli-is-not-running-jump-into-emergency-console/303548/7

Aleksandar-bit, 2022-06-16:
enter in the browser **homeassistant.local:8123** or **homeassistant.local:4375**

andreimoraru, 2022-06-16:
> Hi buddy, I had the same problem, just enter "**login**" in the console and press **enter**, your problem will be solved. I found a solution in the comments of this post https://community.home-assistant.io/t/warn-home-assistant-cli-is-not-running-jump-into-emergency-console/303548/7

It may be the solution, I can't test anymore as I followed the workaround described here https://github.com/home-assistant/operating-system/issues/1970

leenemo, 2022-08-11:
I have the same bug.Although I can login through the post  http://homeassistant.local:8123 ,but don't know
how to repair the issue. can u tell the next works.

Aleksandar-bit, 2022-08-11:
Try to wait 20 minutes, if it won't shut down the virtual machine, start it again. If it doesn't work then try to change the IP address assigned to the machine from which you start the virtual machine and start it again. But it will definitely work on your local host as well.

bhuvaneshne, 2022-08-13:
I have similar issue. Had to reboot my VM running on virtualbox for system maintenance and home assistant did not come back up.
dmesg shows segfaults.

[image]

andreimoraru, 2022-08-17:
> Try to wait 20 minutes, if it won't shut down the virtual machine, start it again. If it doesn't work then try to change the IP address assigned to the machine from which you start the virtual machine and start it again. But it will definitely work on your local host as well.

I am afraid it doesn't make help, but thanks for your comment

andreimoraru, 2022-08-17:
> I have the same bug.Although I can login through the post http://homeassistant.local:8123 ,but don't know how to repair the issue. can u tell the next works.

I have restored from backup, there was no other way to fix the issue

nsde, 2022-10-27:
> Hi buddy, I had the same problem, just enter "**login**" in the console and press **enter**, your problem will be solved. I found a solution in the comments of this post https://community.home-assistant.io/t/warn-home-assistant-cli-is-not-running-jump-into-emergency-console/303548/7

Thanks, this solved the issue for me!

infoBrainSys, 2023-01-10:
i have the same issue，while i input "login"，then it doesn't worked, only reflash the screen again and show "[WARN] Home Assistant CLI not starting! Jump into emergency console" again. How can i fix this problem? my system was just break dowm when it's working fine. It's the auto update things?

almalino, 2023-01-12:
> i have the same issue，while i input "login"，then it doesn't worked, only reflash the screen again and show "[WARN] Home Assistant CLI not starting! Jump into emergency console" again. How can i fix this problem? my system was just break dowm when it's working fine. It's the auto update things?

Same problem when doing fresh install on Intel PC :( Login does not work, setting DNS does not work

Akinetopsia1, 2023-04-17:
Same problem. We lost power in Quebec for a while after an ice storm on the 5th of april and I noticed my HA install never came back online. None of the solutions suggested above work. 

Is there any way to restore an HA backup from the emergency console ? I made a full backup recently because of another issue.

agners, 2023-04-17:
Hm, the power outage probably lead to a corrupted file system of some sort.

What is `systemctl status docker` saying?

Wrt Backup: The files are stored on the data partition (8th partition on the boot storage, unless you are using the data disk feature, then it is on that partition). You'll find the backups under `supervisor/backup/`.

Akinetopsia1, 2023-04-17:
here's the output from systemctl status docker

[image]

This is running in a vm. Mmm, iirc last time my install crashed it also had something to do with a JSON call to that address but that was after an update. 

Issuing `ip addr` from the console shows an inet adapter (#3) named **hassio** with the 172.30.32.1/23 ip mask, so i assume that's an internal virtual network used by HA

as for the backups, they are on the same vm disk image. i assume i'd have to find a way to copy them elsewhere and reinstall them on a fresh HA install ? No other way to recover from the current state then ?

closed 2023-07-25 by github-actions[bot]

SkyLined, 2023-10-03:
I have this issue now and I am disappointed but not surprised to find this issue is closed. HA developers appear to focus on new features instead of stability and it is pushing me away from the platform. The time I save in automation is lost in dealing with these kind of inexplicable, undebuggable issues.

GoodHhh, 2024-01-09:
I also encountered the same problem. I installed it in Virtualbox version 6.1.32. Because the previous update caused system damage, I reinstalled it once, but it kept prompting this problem, even if I log in based on other people's input or modify the DNS, it doesn't work. This problem still occurs

Yonny24, 2024-05-06:
I've always installed HA on proxmox - never had any issues.
Had a spare intel atom with emmc. 
Loaded up ubuntu test from usb- downloaded the generic homeassistant zip and created it on my intel emmc disk. Thought this would be a 5 min install!!

Still get stuck at 'Waiting for HA CLI to be ready' 
Tried the login and enter..
Nothing, nada zilch
Reinstalled it - again same issue.

bit crappy really!

Going back to CasaOS on my intel atom and ditch this install as it is very buggy. Wanted to try Homeassistant OS but this is broken

agners, 2024-05-07:
> Still get stuck at 'Waiting for HA CLI to be ready'

How long did you wait?

What happened after you waited?

The Supervisor updates itself on first start. Depending on the connection speed you are using that might take a while. The CLI startup might then timeout. But on next start, the CLI should start properly.

@sairon we probably should fix that start script, and print a better message to users so they don't run out of patience and think it is a OS error :thinking:

markbrewster, 2024-06-30:
I am experiencing this error. 'Login' not working. 

EDIT: I can't believe this - the ethernet cable was not seated correctly in my laptop. Feel so stupid!

gitaarik, 2024-07-17:
> > Still get stuck at 'Waiting for HA CLI to be ready'
> 
> How long did you wait?
> 
> What happened after you waited?
> 
> The Supervisor updates itself on first start. Depending on the connection speed you are using that might take a while. The CLI startup might then timeout. But on next start, the CLI should start properly.
> 
> @sairon we probably should improve that start script, and print a better message so users don't run out of patience and think it is a OS error 🤔

Can someone clear up where exactly you should wait? At the message `[WARN] Home Assistant CLI not starting! Jump into emergency console...`? Or do you **have** to enter `login` and only then wait 20 minutes? And if after 20 minutes nothing still happens, should a reboot then do it? So what if all this doesn't work and you're still stuck at the same step, what are the ways of debugging this issue?

gitaarik, 2024-07-18:
I also found out that my issue was related to having no internet. Because I am installing it on a Raspberry Pi 2 Zero W, which has WiFi but no Ethernet connection. I did configure WiFi credentials in the [advanced options of Raspberry Pi Imager](https://www.tomshardware.com/news/raspberry-pi-imager-now-comes-with-advanced-options) when flashing the Home Assistant image, but I guess Home Assistant doesn't pick up on these settings. So I connected to the WiFi manually after the error message with `Jump into emergency console...`, I entered:

To see the available WiFi networks:

```sh
nmcli dev wifi list
```

To connect to network with name/SSID & password:

```sh
nmcli dev wifi connect MyNetworkName password "my-network-password"
```

Then it should show as being connected:

```sh
nmcli dev status
```

And only then enter `login`, and then it finally works.

Maybe the installer of Home Assistant should check if there's internet connection, and if there isn't, prompt the user to configure networking before proceeding.

arjun921, 2024-07-27:
> I also found out that my issue was related to having no internet. Because I am installing it on a Raspberry Pi Zero 2 W, which has WiFi but no Ethernet connection. I did configure WiFi credentials in the [advanced options of Raspberry Pi Imager](https://www.tomshardware.com/news/raspberry-pi-imager-now-comes-with-advanced-options) when flashing the Home Assistant image, but I guess Home Assistant doesn't pick up on these settings. So I connected to the WiFi manually after the error message where it says `Jump into emergency console...`:
> 
> To see the available WiFi networks:
> 
> ```shell
> nmcli dev wifi list
> ```
> 
> To connect to network with name/SSID & password:
> 
> ```shell
> nmcli dev wifi connect MyNetworkName password "my-network-password"
> ```
> 
> Then it should show as being connected:
> 
> ```shell
> nmcli dev status
> ```
> 
> And only then enter `login`, and then it finally works.
> 
> Maybe the installer of Home Assistant should check if there's internet connection, and if there isn't, prompt the user to configure networking before proceeding. Maybe giving a hint to using `nmcli` to accomplish this.
> 
> Edit: I did have to reboot again after this before Home Assistant actually came online on my network. And then I had to use `http://<ip-address>:8123`, because `http://homeassistant.local:8123` was not working for me, which was in the output of the Home Assistant CLI.

THIS WORKED FOR ME!

Dell Inspiron Installation without LAN, wireless connection only.

usedspirit, 2024-07-28:
I also found this issue, because I don't have lan. only 4g access with celular.  
I had to install a wifi bridge.
It would be good if the development team put more messages, as this wastes a lot of time.
Why is not possible to have an offline installation ?

it's really a problem that you can't set the wifi parameters when formatting haos in raspberry pi imager

Delentsperon, 2024-08-02:
> THIS WORKED FOR ME!
>
> Dell Inspiron Installation without LAN, wireless connection only.

This worked for me. Also Dell Inspiron without LAN wifi only oddly enough.

sarmaos, 2024-08-05:
> I also found out that my issue was related to having no internet. Because I am installing it on a Raspberry Pi Zero 2 W, which has WiFi but no Ethernet connection. I did configure WiFi credentials [...]

Thank you, this worked like a charm!

agners, 2024-08-06:
> I did configure WiFi credentials in the [advanced options of Raspberry Pi Imager](https://www.tomshardware.com/news/raspberry-pi-imager-now-comes-with-advanced-options) when flashing the Home Assistant image, but I guess Home Assistant doesn't pick up on these settings.

Currently, HAOS cannot process this settings. But it would actually be a nice feature. A quick glance [at the Raspberry Pi imager source code](https://github.com/raspberrypi/rpi-imager/blob/v1.8.5/src/OptionsPopup.qml), it might be possible to to do it. The imager adds a script to the boot directory and uses cmdline.txt to make it run on first boot. So we could process that as well, potentially :thinking: 

In general, with the upcoming HAOS 13.0 the CLI should be available even when the system is offline again (this was a bug in Supervisor fixed with https://github.com/home-assistant/supervisor/pull/5204). From that version onwards, the recommended way is to use the ha CLI, e.g.

```
ha network update wlan0 --ipv4-method auto --wifi-auth wpa-psk --wifi-mode infrastructure --wifi-ssid "MY-SSID" --wifi-psk MY_PASS
```

I am locking this issue as the original post is actually much older and likely a different cause anyways. If you still experience this with HAOS 13.0 and newer, please open a new issue.