home-assistant/operating-system #1970: dockerd level=error msg="failed to load container"

andreimoraru, 2022-06-09:
### Describe the issue you are experiencing

Home Assistant Operating System doesn't boot up and enters a boot loop.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Power on Home Assistant Operating System virtual machine
2. Wait for the virtual machine console
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't get supervisor logs as Home Assistant doesn't boot
```

### Anything in the Host logs that might be useful for us?

```txt
Jun 09 12:13:54 homeassistant dockerd[404]: time="2022-06-09T09:13:54.937513259Z" level=info msg="containerd successfully booted in 0.098714s"
Jun 09 12:13:55 homeassistant audit[421]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=421 comm="apparmor_parser"
Jun 09 12:13:55 homeassistant audit[421]: SYSCALL arch=c000003e syscall=1 success=yes exit=7945 a0=4 a1=5624d33534c0 a2=1f09 a3=0 items=0 ppid=420 pid=421 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="apparmor_parser" exe="/usr/sbin/apparmor_parser" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=61707061726D6F725F706172736572002D4B72002F6D6E742F646174612F646F636B65722F746D702F646F636B65722D64656661756C7431393937333033343235
Jun 09 12:13:55 homeassistant kernel: audit: type=1400 audit(1654766035.062:11): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=421 comm="apparmor_parser"
Jun 09 12:13:55 homeassistant kernel: audit: type=1300 audit(1654766035.062:11): arch=c000003e syscall=1 success=yes exit=7945 a0=4 a1=5624d33534c0 a2=1f09 a3=0 items=0 ppid=420 pid=421 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="apparmor_parser" exe="/usr/sbin/apparmor_parser" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant kernel: audit: type=1327 audit(1654766035.062:11): proctitle=61707061726D6F725F706172736572002D4B72002F6D6E742F646174612F646F636B65722F746D702F646F636B65722D64656661756C7431393937333033343235
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.069618724Z" level=info msg="parsed scheme: \"unix\"" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.069704556Z" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.069769564Z" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.069793659Z" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.072804679Z" level=info msg="parsed scheme: \"unix\"" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.072852521Z" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.072909422Z" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}] <nil> <nil>}" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.072932761Z" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.510719007Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.510790367Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.511137197Z" level=info msg="Loading containers: start."
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.520600103Z" level=error msg="failed to load container" container=93edabb4dd7472b9e229887f1127446d6bb375331811bdd4cf8207c630ed0fb0 error="Container 17f7ed101a68e40f1d345d3bb00b9ff81554071536ec1dea7693d9d2f88fa10f is stored at 93edabb4dd7472b9e229887f1127446d6bb375331811bdd4cf8207c630ed0fb0"
Jun 09 12:13:55 homeassistant dockerd[396]: time="2022-06-09T09:13:55.536707477Z" level=error msg="failed to load container" container=cdfd3091a149481cc16d19f18ade1efb643f1b02612c75b0048e66c4d2d1a2fe error="invalid character 'a' looking for beginning of value"
Jun 09 12:13:55 homeassistant audit[429]: NETFILTER_CFG table=nat family=2 entries=0 op=xt_register pid=429 subj==unconfined comm="iptables"
Jun 09 12:13:55 homeassistant audit[429]: SYSCALL arch=c000003e syscall=55 success=yes exit=0 a0=4 a1=0 a2=40 a3=7ffeb2eee900 items=0 ppid=396 pid=429 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4C002D6E
Jun 09 12:13:55 homeassistant audit[430]: NETFILTER_CFG table=filter family=2 entries=0 op=xt_register pid=430 subj==unconfined comm="iptables"
Jun 09 12:13:55 homeassistant audit[430]: SYSCALL arch=c000003e syscall=55 success=yes exit=0 a0=4 a1=0 a2=40 a3=7ffdf0814070 items=0 ppid=396 pid=430 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D4C002D6E
Jun 09 12:13:55 homeassistant audit[449]: NETFILTER_CFG table=nat family=2 entries=5 op=xt_replace pid=449 subj==unconfined comm="iptables"
Jun 09 12:13:55 homeassistant audit[449]: SYSCALL arch=c000003e syscall=54 success=yes exit=0 a0=4 a1=0 a2=40 a3=55ec8dbd1a60 items=0 ppid=396 pid=449 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4E00444F434B4552
Jun 09 12:13:55 homeassistant audit[451]: NETFILTER_CFG table=filter family=2 entries=4 op=xt_replace pid=451 subj==unconfined comm="iptables"
Jun 09 12:13:55 homeassistant audit[451]: SYSCALL arch=c000003e syscall=54 success=yes exit=0 a0=4 a1=0 a2=40 a3=563c4e5fd940 items=0 ppid=396 pid=451 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4E00444F434B4552
Jun 09 12:13:55 homeassistant audit[453]: NETFILTER_CFG table=filter family=2 entries=6 op=xt_replace pid=453 subj==unconfined comm="iptables"
Jun 09 12:13:55 homeassistant audit[453]: SYSCALL arch=c000003e syscall=54 success=yes exit=0 a0=4 a1=0 a2=40 a3=55f11bf0cb10 items=0 ppid=396 pid=453 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4E00444F434B45522D49534F4C4154494F4E2D53544147452D31
Jun 09 12:13:55 homeassistant audit[455]: NETFILTER_CFG table=filter family=2 entries=8 op=xt_replace pid=455 subj==unconfined comm="iptables"
Jun 09 12:13:55 homeassistant audit[455]: SYSCALL arch=c000003e syscall=54 success=yes exit=0 a0=4 a1=0 a2=40 a3=55f59b6e8cf0 items=0 ppid=396 pid=455 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
Jun 09 12:13:55 homeassistant audit: PROCTITLE proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4E00444F434B45522D49534F4C4154494F4E2D53544147452D32
[... 57 lines cut ...]
[... cut ...]
Jun 09 12:13:56 homeassistant systemd[1]: Dependency failed for HassOS supervisor.
Jun 09 12:13:56 homeassistant systemd[1]: hassos-supervisor.service: Job hassos-supervisor.service/start failed with result 'dependency'.
Jun 09 12:13:56 homeassistant systemd[1]: Dependency failed for Dropbear SSH daemon.
Jun 09 12:13:56 homeassistant systemd[1]: dropbear.service: Job dropbear.service/start failed with result 'dependency'.
```

### System Health information

Can't get to systems health as Home Assistant doesn't boot up

agners, 2022-06-09:
Hm, that looks similar to #1960.

Is that a new installation? Is that always reproducible? 

Can you test the latest release 8.2 (currenlty on the beta channel):
https://github.com/home-assistant/operating-system/releases/tag/8.2

andreimoraru, 2022-06-09:
> Hm, that looks similar to #1960.
> 
> Is that a new installation? Is that always reproducible?
> 
> Can you test the latest release 8.2 (currenlty on the beta channel): https://github.com/home-assistant/operating-system/releases/tag/8.2

This is not a new installation.

It is reproducible all the time. I can share the whole boot log file of the Home Assistant Operating System

I can not apply 8.2 beta because the Home Assistant Operating System never boots up but enters in a boot loop

andreimoraru, 2022-06-09:
```console
-- Journal begins at Tue 2022-05-17 12:46:16 EEST, ends at Thu 2022-06-09 12:14:16 EEST. --
Jun 09 12:13:18 homeassistant kernel: Linux version 5.15.41 (builder@58360eda4381) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g2eebdbad5) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Fri May 20 00:38:27 UTC 2022
Jun 09 12:13:18 homeassistant kernel: Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=a122ee3f8d7a4adea0882248aeb31ed0 cgroup_enable=memory fsck.repair=yes console=tty1 rauc.slot=A
Jun 09 12:13:18 homeassistant kernel: Disabled fast string operations
Jun 09 12:13:18 homeassistant kernel: x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
Jun 09 12:13:18 homeassistant kernel: x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
Jun 09 12:13:18 homeassistant kernel: x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
Jun 09 12:13:18 homeassistant kernel: x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
Jun 09 12:13:18 homeassistant kernel: x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
Jun 09 12:13:18 homeassistant kernel: signal: max sigframe size: 1776
Jun 09 12:13:18 homeassistant kernel: BIOS-provided physical RAM map:
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x0000000000100000-0x000000000ef24fff] usable
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef25000-0x000000000ef2dfff] type 20
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef2e000-0x000000000ef3cfff] usable
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef3d000-0x000000000ef42fff] type 20
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef43000-0x000000000ef47fff] ACPI NVS
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef48000-0x000000000ef60fff] usable
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef61000-0x000000000ef65fff] type 20
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ef66000-0x000000000fee6fff] usable
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000fee7000-0x000000000ff06fff] type 20
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ff07000-0x000000000ff56fff] reserved
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ff57000-0x000000000ff72fff] ACPI data
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ff73000-0x000000000ff76fff] ACPI NVS
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x000000000ff77000-0x00000000bfffffff] usable
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
Jun 09 12:13:18 homeassistant kernel: BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
Jun 09 12:13:18 homeassistant kernel: NX (Execute Disable) protection: active
Jun 09 12:13:18 homeassistant kernel: efi: EFI v2.31 by VMware, Inc.
Jun 09 12:13:18 homeassistant kernel: efi: SMBIOS=0xef43000 ACPI 2.0=0xff57000 MEMATTR=0xfcde698 
Jun 09 12:13:18 homeassistant kernel: SMBIOS 2.7 present.
Jun 09 12:13:18 homeassistant kernel: DMI: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.16221537.B64.2005150253 05/15/2020
Jun 09 12:13:18 homeassistant kernel: vmware: hypercall mode: 0x00
Jun 09 12:13:18 homeassistant kernel: Hypervisor detected: VMware
Jun 09 12:13:18 homeassistant kernel: vmware: TSC freq read from hypervisor : 1596.307 MHz
Jun 09 12:13:18 homeassistant kernel: vmware: Host bus clock speed read from hypervisor : 66000000 Hz
Jun 09 12:13:18 homeassistant kernel: vmware: using clock offset of 3180612198 ns
Jun 09 12:13:18 homeassistant kernel: tsc: Detected 1596.307 MHz processor
Jun 09 12:13:18 homeassistant kernel: e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
Jun 09 12:13:18 homeassistant kernel: e820: remove [mem 0x000a0000-0x000fffff] usable
Jun 09 12:13:18 homeassistant kernel: last_pfn = 0x140000 max_arch_pfn = 0x400000000
Jun 09 12:13:18 homeassistant kernel: x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
Jun 09 12:13:18 homeassistant kernel: e820: update [mem 0xc0000000-0xffffffff] usable ==> reserved
Jun 09 12:13:18 homeassistant kernel: last_pfn = 0xc0000 max_arch_pfn = 0x400000000
Jun 09 12:13:18 homeassistant kernel: Using GB pages for direct mapping
Jun 09 12:13:18 homeassistant kernel: Secure boot could not be determined
Jun 09 12:13:18 homeassistant kernel: ACPI: Early table checksum verification disabled
Jun 09 12:13:18 homeassistant kernel: ACPI: RSDP 0x000000000FF57000 000024 (v02 VMWARE)
Jun 09 12:13:18 homeassistant kernel: ACPI: XSDT 0x000000000FF57064 00005C (v01 INTEL  440BX    06040000 VMW  01324272)
Jun 09 12:13:18 homeassistant kernel: ACPI: SRAT 0x000000000FF570C0 0008D0 (v03 VMWARE EFISRAT  06040001 VMW  000007CE)
Jun 09 12:13:18 homeassistant kernel: ACPI: FACP 0x000000000FF7221A 0000F4 (v04 INTEL  440BX    06040000 PTL  000F4240)
Jun 09 12:13:18 homeassistant kernel: ACPI: DSDT 0x000000000FF57990 01A88A (v01 PTLTD  Custom   00000000 INTL 20130823)
[... cut ...]
Jun 09 12:13:56 homeassistant dockerd[396]:         github.com/docker/libnetwork/drivers/bridge/bridge_store.go:40 +0x1de
Jun 09 12:13:56 homeassistant dockerd[396]: github.com/docker/libnetwork/drivers/bridge.(*driver).configure(0xc000d52b00, 0x26eaf80)
Jun 09 12:13:56 homeassistant dockerd[396]:         github.com/docker/libnetwork/drivers/bridge/bridge.go:439 +0x66b
Jun 09 12:13:56 homeassistant dockerd[396]: github.com/docker/libnetwork/drivers/bridge.Init({0x2d78de8, 0xc000145b80}, 0x6)
Jun 09 12:13:56 homeassistant dockerd[396]:         github.

andreimoraru, 2022-06-16:
Issue couldn't be solved. The workaround was to download the latest version vmdk available on https://github.com/home-assistant/operating-system/releases/download/ then to attach the disk to virtual machine in VMware Workstation, detach the .vmdk file with issue described in the bug above.
Next I started the virtual machine, on setup screen I chose to restore from backup and pointed to the location of latest backup (it was a USB harddrive attached to the computer).

closed 2022-09-21 by github-actions[bot]