home-assistant/operating-system #1974: VM does not reboot after HAOS upgrade (Virtualbox)

Guyohms, 2022-06-14:
### Describe the issue you are experiencing

After an upgrade of HAOS, the host proceeds to reboot but hangs and never do so.
I have to power down the VM and restart it.

But, if I reboot using the home assistant interface under System/Hardware/Reboot host. Everything works fine.

Here's where the Virtualbox console stops:
[image]

I've waited 4-5 minutes there before decided to power cycle the VM.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade HAOS when offered through the UI
2. VM hangs at reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System Health information

## System Health

version | core-2022.6.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.15.45
arch | x86_64
timezone | America/Toronto

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4632
Installed Version | 1.25.5
Stage | running
Available Repositories | 1055
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 30.8 GB
disk_used | 14.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), File editor (5.3.3), Mosquitto broker (6.1.2), Terminal & SSH (9.4.0), Samba share (9.7.0), Home Assistant Google Drive Backup (0.108.2), Grafana (7.6.0), InfluxDB (4.5.0), Glances (0.16.0), MariaDB (2.4.0), Hyperion.NG (2.0.13), Nginx Proxy Manager (0.12.0), Vaultwarden (Bitwarden) (0.17.0), ESPHome (2022.5.1), AdGuard Home (4.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 8
views | 15
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 12, 2022, 17:36
-- | --
current_recorder_run | June 14, 2022, 17:15
estimated_db_size | 223.50 MiB
database_engine | mysql
database_version | 10.4.19

</details>

MasterCATZ, 2022-06-17:
same issues with virtualbox , nothing will upgrade while running v8 series images 
had no issues with things doing updates with haos_ova-7.6.vdi until it hit 8's 

and the freshly downloaded 8.2 VDI will not even boot up with a IP address to connect to

sjmotew, 2022-06-19:
Ongoing issues with VirtualBox owa installs.  I thought I final had some stability at 8.0, but still having issues (slow, erratic when rebooting).  Was able to get 8.2 installed and working (never able w 8.1), seemed to work well for a bit, then issues with the network adapter, and now seeing the same "Docker Application Container Engine" stalls for hours.  Trying to get back to 8.0.

agners, 2022-06-25:
> and the freshly downloaded 8.2 VDI will not even boot up with a IP address to connect to

This seems a different issue. Can you open a new issue with logs provided?

agners, 2022-06-25:
> Ongoing issues with VirtualBox ova installs. I thought I finally had some stability at 8.0, but still having issues (slow, erratic when rebooting). Was able to get 8.2 installed and working (never able w 8.1), seemed to work well for a bit, then issues with the network adapter, and now seeing the same "Docker Application Container Engine" stalls for hours. Trying to get back to 8.0.

Slow/erratic sounds like it's related to #1705. Can you try with a single virtual CPU only as suggested by https://github.com/home-assistant/operating-system/issues/1705#issuecomment-1144616267?

Guyohms, 2022-08-23:
For information, I just updated from 8.4 to 8.5 and it's still not rebooting and stays there until I power cycle the VM.
[image]

Guyohms, 2022-10-14:
Good news !
It seems like 9.2 fixed this issue. I just updated and the system restarted automatically.

closed 2023-01-19 by github-actions[bot]

MasterCATZ, 2023-01-19:
9.2 and working again